Koenkk/zigbee2mqtt #1478: Enddevices Xiomi Aqara stop working after a couple minutes

YamiFrankc, 2019-04-29:
# Bug Report
I noticed that my Xiomi Aqara motion and temperature sensor, and also the WXKG11LM stop working after a few minutes. Sometimes they still show a linkquality value over 0 (sometimes over 30) but they wont update anymore. The only way I have to get them working again is removing them from database.db and relink them. But again, they only work for some minutes.
This started about two weeks ago, but I didnt have time to mess with it until now.
All of them used to work properly. I did not update or change anything when they stopped working.

## What did you expect to happen
For my sensors to stay connected and work properly.

## How to reproduce it (minimal and precise)
I dont really know how to reproduce, but here are the logs from [today](https://pastebin.com/23dDgwM1) and [yesterday](https://pastebin.com/YNfnu5TY).

## Debug Info
zigbee2mqtt version: Updated developer branch today
CC253X firmware version: 20190223 (updated today, too)

Koenkk, 2019-05-01:
Maybe related to #1485, do you have a Sylvania SMART+ Smart Plug?

YamiFrankc, 2019-05-04:
Hello, I dont have a Sylvania SMART+ plug, but I do have other Sylvania devices(bulbs and rgb lights)
My motion sensors and one of the temperature sensors have been working ok for a few days now.
The WXKG11LM also hasnt disconnected yet, but doesnt seem to be working properly because it only sends empty "click" messages. 

I also cant join my 81809  AduroSmart ERIA bulb anymore (It was paired before), it doesn't produce message in my log when I try to pair it back,

Koenkk, 2019-05-05:
Can you post the log of the empty click messages?

YamiFrankc, 2019-05-06:
[LOG](https://pastebin.com/DKje91pp)

Taking a better look at it, it looks like sometimes it does register a "single" or "double" message (I cant get the triple or quadruplet one. But its always followed by a empty one?
When I tried to use it in automations, they randomly triggered sometimes.

Ton1965, 2019-05-06:
Just a silly question: did you try to change the batteries? I can see battery level 100 in the log but it seems to me that z2m is not always correct about battery level. My Xiaomi devices used to be connected to a Xiaomi gateway and it reported much lower battery levels than z2m after I moved them to it. They are with z2m for more than half a year but I still see 100 battery levels.
I can see this as one of possible explanations of why your devices stopped working correctly without any external changes.

Koenkk, 2019-05-06:
The empty click messages are expected, they are required for proper home assistant automation integration (and are only send if you set `homeassistant: true` in `configuration.yaml`). See automation example here: http://www.zigbee2mqtt.io/integration/home_assistant.html#via-home-assistant-entity

closed 2019-07-12 by stale[bot]