Koenkk/zigbee2mqtt #1479: Docker in Windows container base image

desertspider, 2019-04-29:
Would it be possible to make docker run inside a windows container so the command "docker inspect --format '{{.Os}}' koenkk/zigbee2mqtt" would show "windows".

This because in the next docker release, it is possible to mount the zigbee stick inside docker on a Windows host.
https://docs.microsoft.com/en-us/virtualization/windowscontainers/deploy-containers/hardware-devices-in-containers

Koenkk, 2019-04-29:
That would indeed be nice, could you make a `Dockerfile` based on the nano image?

desertspider, 2019-04-30:
I will give it a try.

Is it a problem I copy nodejs and git folder to the nanoserver instead
downloadingen the files and unzipping it in the buildfile? Because
nanoserver cannot unzip files without powershell.
Powershell nanoserver is 500 mb
Without powershell 100 mb

Koenkk, 2019-04-30:
Yes, just make it as small as possible.

desertspider, 2019-05-01:
Just made a quick and dirty dockerfile for testing purposes. For now i did use the powershell version, can change that later.

The dockerfile;
`FROM mcr.microsoft.com/powershell:nanoserver-1809

USER Administrator

ADD https://nodejs.org/dist/v10.15.3/node-v10.15.3-win-x64.zip C:\\build\\node-v10.15.3-win-x64.zip
RUN pwsh -Command Expand-Archive C:\build\node-v10.15.3-win-x64.zip C:\; Rename-Item C:\node-v10.15.3-win-x64 node

ADD https://github.com/git-for-windows/git/releases/download/v2.21.0.windows.1/MinGit-2.21.0-64-bit.zip C:\\build\\MinGit-2.21.0-64-bit.zip
RUN pwsh -Command Expand-Archive C:\build\MinGit-2.21.0-64-bit.zip C:\git

ENV PATH="$WindowsPATH;C:\node;C:\git\cmd"

RUN git clone https://github.com/Koenkk/zigbee2mqtt.git /app
WORKDIR /app
COPY run.sh /app
RUN npm install
RUN rmdir /Q /S C:\git

ENTRYPOINT cd c:\\app && npm start `

The run command;
`docker run -v c:\dockerdata\zigbee2mqtt:c:\app\data --isolation=process --device="class/86E0D1E0-8089-11D0-9CE4-08003E301F73" testapp:v1` 

The log
`zigbee2mqtt@1.3.1 start C:\app
> node index.js

  zigbee2mqtt:info 5/1/2019, 7:30:58 AM Logging to directory: 'C:\app\data\log\2019-05-01.07-30-58'
  zigbee2mqtt:info 5/1/2019, 7:30:59 AM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 5/1/2019, 7:30:59 AM Starting zigbee-shepherd
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
 1: 00007FF62F53F04A v8::internal::GCIdleTimeHandler::GCIdleTimeHandler+5114
 2: 00007FF62F51A0C6 node::MakeCallback+4518
 3: 00007FF62F51AA30 node_module_register+2032`

(log is longer)

if i occupy the com port by zigbee2mqtt on the host, i get the regular error message;
`  zigbee2mqtt:info 5/1/2019, 7:08:26 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 5/1/2019, 7:09:26 AM Starting zigbee-shepherd
  zigbee2mqtt:error 5/1/2019, 7:09:26 AM Error while starting zigbee-shepherd!` 

How can i debug the memory error?

Koenkk, 2019-05-02:
Did you try https://github.com/Chatie/wechaty/issues/1435#issuecomment-401686147 ?

desertspider, 2019-05-02:
> Did you try [Chatie/wechaty#1435 (comment)](https://github.com/Chatie/wechaty/issues/1435#issuecomment-401686147) ?

Yup, but dont thing the default nodejs 1500 mb memory limit is the problem.
Zigbee2mqtt hangs for 90 sec on init zigbee-shepherd booting. before hitting the memory limit. In this time the memory on the host system increases by 1500 mb.
```
  zigbee2mqtt:info 5/2/2019, 1:50:46 AM Starting zigbee-shepherd
2019-05-02T08:50:46.431Z zigbee-shepherd:init zigbee-shepherd booting...
2019-05-02T08:50:46.451Z zigbee-shepherd:request REQ --> SYS:osalNvRead
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
```

desertspider, 2019-05-02:
Got it working by using mcr.microsoft.com/windows/servercore:1809
So nanoserver is missing something......

desertspider, 2019-05-03:
Should i create a issue at https://github.com/zigbeer/zigbee-shepherd/ since they load a something that is missing in nanoserver?

Koenkk, 2019-05-03:
What exactly is it loading? zigbee-shepherd isn't maintained anymore so I forked it: https://github.com/koenkk/zigbee-shepherd

desertspider, 2019-05-03:
> What exactly is it loading? zigbee-shepherd isn't maintained anymore so I forked it: https://github.com/koenkk/zigbee-shepherd

Only thing i know for sure that the full server version aka servercore, zigbee-shepherd works just fine. With nanoserver it goes in some sort of memory loop. So shepherd is trying to use a library not available in nanoserver. Question is, what library.

desertspider, 2019-05-20:
Cannot get it to work on a nanoserver image. Servercore does work, but image is huge.
Closing this since a docker image of 4.67GB is to much i guess. 

The dockerfile:
```
FROM mcr.microsoft.com/windows/servercore:1809

USER ContainerAdministrator

COPY node-v10.15.3-win-x64\* /node

COPY git\* /git

ENV PATH="$WindowsPATH;C:\node;C:\git\cmd"

RUN git clone https://github.com/Koenkk/zigbee2mqtt.git /app
WORKDIR /app
RUN npm install
RUN rmdir /Q /S C:\git

ENTRYPOINT cd c:\\app && npm start
```

closed 2019-05-20 by desertspider

Koenkk, 2019-05-20:
@desertspider thanks for your investigation!