Koenkk/zigbee2mqtt #1481: SSL encryption

GForce2010, 2019-04-29:
Currently I have Zigbee2MQTT connected to a mosquito server within my own LAN.

For reasons not connected to Zigbee2MQTT at all I may be wanting to open up my mosquito server so I can connect to it from outside my LAN. If I do this I would want to enable SSL on my server and wanted to know if Zigbee2MQTT can connect to a MQTT server with SSL and if so how do I configure this?

Thanks.

Ton1965, 2019-04-30:
I currently do it a different way. I have a separate server in some cloud and it is running Mosquitto with SSL. External devices report to this server.
The server also runs OpenVPN and my HA raspberry connects to this VPN as a client.
HA raspberry runs Mosquitto server used for all home devices. 
The server Mosquitto also has a two-way bridge connection with Mosquitto on the raspberry.
Such configuration allows to use the internal server with full freedom and it is fully self-sufficient and at the same time have a protected connection from outside.
The same server with VPN serves as a proxy for external access to HA.

lolorc, 2019-04-30:
@GForce2010 it is supposed to work if you use mqtts:// instead of mqtt://
you can also have a look at stunnel

GForce2010, 2019-04-30:
Change of plan.

After doing some more research I found it was able to configure my Mosquitto broker to listen on two different ports. One port has no security for login details, used for in LAN only, and the second port I have configured for authentication with SSL/TSL encryption that can deal with items outside my LAN.

My goal is to have a system that has zero dependency on cloud servers. You never know when one might get shut down.

closed 2019-04-30 by GForce2010