Koenkk/zigbee2mqtt #1483: Xiaomi sensor (RTCGQ01LM) does not link to Ikea outlet router (E1603)

VictorvandenHoven, 2019-04-30:
# Bug Report:

## What happened
The Xiaomi sensor (RTCGQ01LM) does not link to an Ikea outlet router (E1603) that is very nearby (10cm). 
 Instead it only links directly to the coordinator two floors above with link-quality 1.  
OTHER end-devices (JTYJ-GD-01LM/BW) DO link via the Ikea outlet router.
....
**HAS ANYONE SUCCEEDED IN LINKING THE sensor RTCGQ01LM  VIA THE outlet E1603 ROUTER?**

- Coordinator == "CC2531 USB sniffer" with latest firmware
- Xiaomi sensor == "Xiaomi MiJia human body movement sensor (RTCGQ01LM)"
- Ikea outlet router == "IKEA TRADFRI control outlet (E1603)"
- Xiaomi smoke detector == "Xiaomi MiJia Honeywell smoke detector (JTYJ-GD-01LM/BW)"

## What did you expect to happen
Expected the Xiaomi sensor to link to the coordinator via the Ikea outlet.

## How to reproduce it (minimal and precise)
Build network with coordinator and two floors below an IKEA router and a Xiaomi sensor end-device (mentioned above) very close to each other and see that sensor device is not linked to outlet router.

## Debug Info
zigbee2mqtt version:        zigbee2mqtt@1.3.1
CC253X firmware version:        CC2531_MAX_STABILITY_20190315

> Apr 30 16:41:37 ZIGBEE npm[12409]:   zigbee2mqtt:debug 4/30/2019, 4:41:37 PM Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion' (0x00158d0001ae91ed) of endpoint 1
> Apr 30 16:41:37 ZIGBEE npm[12409]:   zigbee2mqtt:info 4/30/2019, 4:41:37 PM MQTT publish: topic 'zigbee2mqtt/BewegingBerging', payload '{"occupancy":true,"**linkquality":0**,"battery":100,"voltage":3015}'

Carstijn, 2019-04-30:
I have the same here. I have a sensor upstairs in the hallway, multiple lights and a ikea repeater are in the area of the sensor but it only links directly to the coordinator. I guess because of this I am missing some events of the movement sensors.
[image]

JariMartikainen, 2019-05-02:
Personally, I don't recommend Ikea E1603 with Xiaomi sensors.
Many of those sensors (it was water and door for me), will not work with Ikea router. I don't think that zigbee2mqtt is part of this problem.

I solved this problem by using CC2531 router and then everything is good.

VictorvandenHoven, 2019-05-02:
I have ordered some items for a CC2530 router in China
I hope the Xiaomi sensors can link to it.

closed 2019-07-08 by stale[bot]