Koenkk/zigbee2mqtt #1485: Aqara sensors are disconnected after some time when connected through router

imortkz, 2019-05-01:
# Bug Report

Aqara sensor connected to the router disconnects and became unavailable after some time. Other sensors, connected to the CC2531, are stable.

## What happened

Router: Sylvania SMART+ Smart Plug (72922-A)
Sensors: Xiaomi Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)

When connected to the router keeps online for hour or so, then didn't send any update.
No warning/error messages in the zigbee2mqtt logs.

## What did you expect to happen

Sensor keeps connected to the router.

## How to reproduce it (minimal and precise)

Connect sensor to the router and wait for a while.
Aqara sends 5 updates in 41 minutes after rejoin and connected to the router, and falls offline after.

## Debug Info
zigbee2mqtt version: zigbee2mqtt version 1.3.1 (commit #unknown)
CC253X firmware version: '20190223'

Koenkk, 2019-05-02:
I assume the same as @manup does (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1488#issuecomment-488245420). Keeping an eye on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1488

imortkz, 2019-05-14:
Installed a CC2530 router and connected two sensors through it.
Xiaomi MiJia temperature & humidity sensor (WSDCGQ01LM) survived few coordinator restarts and 10-60m downtime without any issues.

Turned off the coordinator for a whole night, both sensors remains active and reports when coordinator is back.

I'll give it a couple days more but it looks like it was a Sylvania SMART+ Smart Plug issue.

closed 2019-05-14 by imortkz

hawkefly, 2020-04-12:
It seems that 20200312 firmware in dev branch fix this issue @Koenkk

xraver, 2025-09-27:
Hello.
I want to share my experience.
I have aquara devices and legrand/bticino.

The only way to make it work is to switch off all routers and pair the aquara devices.
In this way aquara devices remain connected directly with the coordinator and connection is stable.

Hope this helps