Koenkk/zigbee2mqtt #1487: Missing "payload"-attributes - Gledopto Zigbee LED controlle

exetico, 2019-05-01:
Hi,

First a big **thanks** for the zigbee2mqtt solution. I'm very happy for it, and using for almost every smart-home elements i have.

The other day i installed a new `GL-C-008 | Gledopto Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer`, and added it to my network.

Every feature works as expected (so i can change colors, brightness, color temperature and so on). However, i noticed that the payload from zigbee2mqtt to my MQTT broker is limited to the `state`-attribute:
```
MQTT publish: topic 'zigbee2mqtt/ledstrip1', payload '{"state":"ON"}'
```

It looks like Home Assistant are messing up the state of the LED strip due to the missing attributes. So i can't check fx. brightness om HA, or create automation based on fx _add 10% more light to the LED-strip_..

I'm not 100% sure that this is a bug/limited support from zigbee2mqtt, but i would like to hear if i can do anything to fix it...? (Also posted in [the Home Assistants Community](https://community.home-assistant.io/t/brightness-stuck-at-255-zigbee2mqtt-led-strip/112891))

**Example with Gledopto Zigbee LED controller vs. Philips Hue light-bulb**

Here is the log from zigbee2mqtt, if i selecting a specific another lightlevel in HA (For the _Gledopto_-device)
```
2019-5-1 08:17:01 - info: Zigbee publish to device '0x00124b0019c796e8', genLevelCtrl - moveToLevelWithOnOff - {"level":145,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
2019-5-1 08:17:01 - info: MQTT publish: topic 'zigbee2mqtt/ledstrip1', payload '{"state":"ON"}'
```

And here is another with a Hue light-bulb:
```
2019-5-1 08:19:34 - info: Zigbee publish to device '0x0017880104e821a0', genLevelCtrl - moveToLevelWithOnOff - {"level":200,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-1 08:19:35 - info: MQTT publish: topic 'zigbee2mqtt/hue2', payload '{"state":"ON","brightness":254,"color_temp":355,"color":{"x":0.45,"y":0.408}}'
```

As you can see, the `payload` for `'zigbee2mqtt/ledstrip1` has nothing else than `state` included.

Is it possible to add the other state-attributes, fx. brightness, color and so, for my RGB CCT LED zigbee-powered controller?

<hr>

_Here is the schema sent to HA, from zigbee2mqtt:_
```
schema":"json","command_topic":"zigbee2mqtt/ledstrip1/set","state_topic":"zigbee2mqtt/ledstrip1","name":"ledstrip1_light","unique_id":"0x00124b0019c796e8_light_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00124b0019c796e8","name":"ledstrip1","sw_version":"Zigbee2mqtt 1.1.1","model":"Zigbee LED controller RGB + CCT / RGBW / WWCW / Dimmer (GL-C-008)","manufacturer":"Gledopto"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

_Here is the first lines of a "fresh start", i'm aware of the old firmware for my zigbee coordinator, and i'm going to update this soon, but i don't think that it's relevant for this.. is it? :-) _

```
2019-5-1 08:33:32 - info: Starting zigbee2mqtt version 1.1.1 (commit #ff02e54)
2019-5-1 08:33:32 - info: Starting zigbee-shepherd
2019-5-1 08:33:34 - info: zigbee-shepherd started
2019-5-1 08:33:34 - info: Coordinator firmware version: '20180815
```

_I'm running zigbee2mqtt in docker env. with image: `koenkk/zigbee2mqtt`_

<hr>

I actually think this is the same as #823  ...

Koenkk, 2019-05-01:
Please first update to the latest zigbee2mqtt version (you are currently on 1.1.1 and latest is 1.3.1).

exetico, 2019-05-01:
Damn, Koenkk. I'm feeling a bit stupid here... For some reason, i forgot `docker pull koenkk/zigbee2mqtt` in on of my docker-scripts (if i need to force updates to an container or two).

Thanks! :+1: 

And confirmed OK:
```
5/1/2019, 9:16:01 AM - info: Zigbee publish to device '0x00124b0019c796e8', genLevelCtrl - moveToLevelWithOnOff - {"level":129,"transtime":0} - {"manufSpec":0,"disDefaultRsp":0} - 11
5/1/2019, 9:16:01 AM - info: MQTT publish: topic 'zigbee2mqtt/ledstrip1', payload '{"state":"ON","color":{"x":0.701,"y":0.299},"color_temp":482,"brightness":129}'
```

closed 2019-05-01 by exetico