Koenkk/zigbee2mqtt #1489: [dev branch] Cannot read property 'registerOnStateChangeListener' of undefined

lolorc, 2019-05-02:
using latest dev
```
May 02 19:38:30 /home/pi/zigbee2mqtt/node_modules/q/q.js:155
May 02 19:38:30                 throw e;
May 02 19:38:30                 ^
May 02 19:38:30
May 02 19:38:30 TypeError: Cannot read property 'registerOnStateChangeListener' of undefined
May 02 19:38:30     at Groups.onZigbeeStarted (/home/pi/zigbee2mqtt/lib/extension/groups.js:89:20)
May 02 19:38:30     at extensions.filter.forEach (/home/pi/zigbee2mqtt/lib/controller.js:117:75)
May 02 19:38:30     at Array.forEach (<anonymous>)
May 02 19:38:30     at Controller.onZigbeeStarted (/home/pi/zigbee2mqtt/lib/controller.js:117:58)
May 02 19:38:30     at zigbee.start (/home/pi/zigbee2mqtt/lib/controller.js:174:26)
May 02 19:38:30     at shepherd.start (/home/pi/zigbee2mqtt/lib/zigbee.js:79:17)
May 02 19:38:30     at /home/pi/zigbee2mqtt/node_modules/q/q.js:2055:17
May 02 19:38:30     at runSingle (/home/pi/zigbee2mqtt/node_modules/q/q.js:137:13)
May 02 19:38:30     at flush (/home/pi/zigbee2mqtt/node_modules/q/q.js:125:13)
May 02 19:38:30     at process._tickCallback (internal/process/next_tick.js:61:11)
May 02 19:38:30 npm ERR! code ELIFECYCLE
May 02 19:38:30 npm ERR! errno 1
May 02 19:38:30 npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
May 02 19:38:30 npm ERR! Exit status 1
May 02 19:38:30 npm ERR!
May 02 19:38:30 npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
May 02 19:38:30 npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
May 02 19:38:30
May 02 19:38:30 npm ERR! A complete log of this run can be found in:
May 02 19:38:30 npm ERR!     /home/pi/.npm/_logs/2019-05-02T17_38_30_301Z-debug.log
```

``` info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.3.1~prestart: zigbee2mqtt@1.3.1
6 info lifecycle zigbee2mqtt@1.3.1~start: zigbee2mqtt@1.3.1
7 verbose lifecycle zigbee2mqtt@1.3.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.3.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/pi/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.3.1~start: CWD: /home/pi/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.3.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.3.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.3.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.3.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.3.1
15 verbose cwd /home/pi/zigbee2mqtt
16 verbose Linux 4.19.34-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.3.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.3.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
```

Koenkk, 2019-05-02:
Do you have `cache_state: false`?

lolorc, 2019-05-02:
yep :)

Koenkk, 2019-05-02:
Should be fixed with latest dev.

lolorc, 2019-05-02:
Thanks a lot !

closed 2019-05-02 by lolorc