Koenkk/zigbee2mqtt #1491: looks like the cc2531 stick is "dead" after flashing

dimaperi, 2019-05-04:
HI Guys!

im flashed the cc2531 usb stick with arduino uno, There were no problems.
but after that, the green light dont turn up when im plug it in and its seems like the 
stick is "dead"

closed 2019-07-10 by stale[bot]

fuomag9, 2020-07-26:
I have the same issue :(

martinszelcel, 2020-08-26:
Same problem :(

fuomag9, 2020-08-26:
@martinszelcel I ended up returning the cc2531 to amazon and bought a conbee II, nothing seemed to resurrect the usb interface (it wasn't picked up even after forcing modprobes)

martinszelcel, 2020-08-27:
@fuomag9 So I also try to return my module, I bought it on AliExpress so I'm not sure if it will be possible. Can you please tell me more about modprobes you mentioned?

fuomag9, 2020-08-27:
> @fuomag9 So I also try to return my module, I bought it on AliExpress so I'm not sure if it will be possible. Can you please tell me more about modprobes you mentioned?

Unfortunately I don't remember which modprobe I used since I was copy pasting them from the the internet and hoping the USB stick was recognized