Koenkk/zigbee2mqtt #1492: 'ccznp has not been initialized yet' upon start

vogler, 2019-05-04:
I always get this error upon starting. When I do `npm start` a second time immediately after, it works. I guess this is not the desired behavior. What's the reason for this?

~~~
/opt/zigbee2mqtt dev*
❯ npm start

> zigbee2mqtt@1.3.1 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 5/4/2019, 1:25:33 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-04.13-25-32'
  zigbee2mqtt:info 5/4/2019, 1:25:34 PM Starting zigbee2mqtt version 1.3.1 (commit #c6617fb)
  zigbee2mqtt:info 5/4/2019, 1:25:34 PM Starting zigbee-shepherd
/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:183
        throw new Error('ccznp has not been initialized yet');
        ^

Error: ccznp has not been initialized yet
    at CcZnp.request (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:183:15)
    at CcZnp.(anonymous function) [as zdoRequest] (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:225:21)
    at Zdo._sendZdoRequestViaZnp (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:39:12)
    at Zdo._genericRequest (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:65:10)
    at Zdo.request (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:24:21)
    at Controller.request (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:366:19)
    at Object.query.device (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/query.js:69:23)
    at Object.query.deviceWithEndpoints (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/query.js:101:11)
    at Controller.simpleDescReq (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:487:23)
    at Controller.endDeviceAnnceHdlr (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:704:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-05-04T11_25_35_120Z-debug.log
~~~

## Debug Info
zigbee2mqtt version: 1.3.1#dev c6617fb7d9af6ac03c603391a74d00cead458ec8
CC253X firmware version: Dec 2018?

Koenkk, 2019-05-05:
I haven't seen this before, what's your setup?

vogler, 2019-05-05:
Currently a CC2531 USB Stick on a RPi3. Flashed `CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex` on 19.12.2018.

~~~
❯ node --version
v10.15.3
❯ npm --version
6.9.0
~~~

`data/configuration.yaml`:
~~~
homeassistant: true
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0 # Location of CC2531 USB sniffer
  disable_led: true
# advanced:
  # log_level: debug
  # cache_state: false # home assistant requires all attributes in MQTT messages, disable cache to just get the original zigbee message with only the changed attributes
devices:
  '0x00158d00027419e4':
    friendly_name: Aqara_Temp_Bad
    retain: false
    debounce: 1
  '0x00158d000243567e':
    friendly_name: Aqara_Door_Bad
    retain: false
  '0x00158d0002b5306c':
    friendly_name: Aqara_Door_Apt
    retain: false
  '0x00158d0002c5bf49':
    friendly_name: Aqara_Door_WZ
    retain: false
  '0x00158d0002b127f4':
    friendly_name: Aqara_Vibration
    retain: false
  '0x000b57fffe321252':
    friendly_name: Tradfri_WW_Kueche
    retain: true
  '0x90fd9ffffede386b':
    friendly_name: Tradfri_WS_Bett
    retain: true
  '0x0017880103283554':
    friendly_name: Hue_Motion_WZ
    retain: false
  '0x001788010328d8bc':
    friendly_name: Hue_Motion_Bad
    retain: false
  '0x00158d0002ca364e':
    friendly_name: Aqara_Door_Balkon
    retain: false
  '0x00158d000312f723':
    friendly_name: Aqara_Door_Fridge
    retain: false
  '0x00158d00032b69d1':
    friendly_name: Aqara_Door_WC
    retain: false
    debounce: 1
~~~

Koenkk, 2019-05-05:
Where did you get this firmware from?

vogler, 2019-05-06:
I followed the [documentation](https://www.zigbee2mqtt.io/information/alternative_flashing_methods.html) for flashing with a Wemos D1 mini since I didn't have a CC debugger at the time. It still links to the same file. Can't this be updated automatically?
I'll try the new firmware.

httpedo13, 2019-05-11:
i have the same error..

vogler, 2019-05-15:
Using `CC2530_20190425` I no longer have the problem.
However, now the networkmap doesn't show any links anymore. I wanted to compare it with the CC2531 as a coordinator 😭

vogler, 2019-05-15:
Repaired everything; now the networkmap works again and has all edges as before.

closed 2019-05-15 by vogler