Koenkk/zigbee2mqtt #1493: Cant start Zigbee2mqtt as service

Wibias, 2019-05-04:
# Bug Report
Soo. I can start zigbee2mqtt with "npm start" from /opt/zigbee2mqtt. But when I try to start it as a service I get this Error:

`   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-05-04 14:57:45 CEST; 3s ago
  Process: 24824 ExecStart=/usr/bin/npm start (code=exited, status=203/EXEC)
 Main PID: 24824 (code=exited, status=203/EXEC)

May 04 14:57:45 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May 04 14:57:45 hassbian systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
May 04 14:57:45 hassbian systemd[1]: Stopped zigbee2mqtt.
May 04 14:57:45 hassbian systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
May 04 14:57:45 hassbian systemd[1]: Failed to start zigbee2mqtt.
May 04 14:57:45 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed state.
May 04 14:57:45 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
`
My service file looks like this:

`[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target`

I already deleted /opt/zigbee2mqtt and reinstalled it. Still the same error message.

Also tried as user "pi" and "root".

No clue where to go from here. Can anyone help me with this?

Koenkk, 2019-05-05:
Could you post the zigbee2mqtt log when starting as a service? Log can be found in the `data/log` folder.

Wibias, 2019-05-05:
There is no log being created when starting it as a service. Its just created when starting it with npm start from the folder.

Koenkk, 2019-05-06:
What is the output of `sudo journalctl -u zigbee2mqtt.service -f`?

Wibias, 2019-05-06:
Tried that already to see if it gives me more information but the same as the above error message unfortunately.

```
May 05 19:42:19 hassbian systemd[1]: Started zigbee2mqtt.
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=203/EXEC
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed state.
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
May 05 19:42:19 hassbian systemd[1]: Stopped zigbee2mqtt.
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
May 05 19:42:19 hassbian systemd[1]: Failed to start zigbee2mqtt.
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed state.
May 05 19:42:19 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

Koenkk, 2019-05-07:
Can you try running with:
```js
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=/tmp/zigbee2mqtt
StandardError=/tmp/zigbee2mqtt
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

and post the content of `/tmp/zigbee2mqtt` ?

Wibias, 2019-05-07:
There is nothing being created in /tmp when starting/restarting the service with that.

Koenkk, 2019-05-07:
Is there something logged to syslog when starting the service? (`tail -f /var/log/syslog`)

Wibias, 2019-05-07:
Actually, it does. Something odd, tbh.

```
May  7 18:13:12 hassbian systemd[1]: Started zigbee2mqtt.
May  7 18:13:12 hassbian systemd[11299]: zigbee2mqtt.service: Failed at step EXE                                                                                                                     C spawning /usr/bin/npm: No such file or directory
May  7 18:13:12 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, c                                                                                                                     ode=exited, status=203/EXEC
May  7 18:13:12 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed st                                                                                                                     ate.
May  7 18:13:12 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'ex                                                                                                                     it-code'.
May  7 18:13:12 hassbian systemd[1]: zigbee2mqtt.service: Service hold-off time                                                                                                                      over, scheduling restart.
May  7 18:13:12 hassbian systemd[1]: Stopped zigbee2mqtt.
```

That's pretty odd. Because:

[image]

The only thing that's not normal compared to the other files in there is that npm is red colored. Not sure what that means, though.

Koenkk, 2019-05-07:
```js
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=/tmp/zigbee2mqtt
StandardError=/tmp/zigbee2mqtt
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

Wibias, 2019-05-07:
That gives me:

`May 07 18:44:23 hassbian systemd[1]: [/etc/systemd/system/zigbee2mqtt.service:6] Executable path is not absolute, ignoring: npm start`

Koenkk, 2019-05-07:
What is the output of `which npm`

Wibias, 2019-05-07:
Yea I just tried that, as it came to my mind aswell. Its giving me:

`/home/pi/.nvm/versions/node/v10.15.3/bin/npm`

So I tried something like this:

```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/home/pi/.nvm/versions/node/v10.15.3/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

And that gives me in syslog:

```
May  7 18:54:11 hassbian systemd[1]: Started zigbee2mqtt.
May  7 18:54:11 hassbian npm[12213]: /usr/bin/env: ‘node’: No such file or directory
May  7 18:54:11 hassbian systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=127/n/a
May  7 18:54:11 hassbian systemd[1]: zigbee2mqtt.service: Unit entered failed state.
May  7 18:54:11 hassbian systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
May  7 18:54:12 hassbian systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
May  7 18:54:12 hassbian systemd[1]: Stopped zigbee2mqtt.
```

Koenkk, 2019-05-07:
Did you follow the installing section of http://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html ?

Wibias, 2019-05-07:
Yes.

Wibias, 2019-05-07:
Well I just fixed it myself. It turned out that my npm installation was broken somehow.

I completly uninstalled npm and nodejs, like did some google research where all the files are located. And then reinstalled it with:

```
sudo curl -sL https://deb.nodesource.com/setup_10.x | sudo -E bash -

sudo apt-get install -y nodejs git make g++ gcc
```

Now its working.

Thanks for the help!

closed 2019-05-07 by Wibias

pqpxo, 2019-12-23:
I had this issue and solved it by changing:

ExecStart=/usr/bin/npm start

To

ExecStart=/usr/local/bin/npm start

trikis1, 2020-01-03:
I have the same problem but y I change 
ExecStart=/usr/bin/npm start
to
ExecStart=/usr/local/bin/npm start
I change error 
systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-01-03 18:19:48 CET; 7s ago
  Process: 8946 ExecStart=/usr/bin/npm start (code=exited, status=1/FAILURE)
 Main PID: 8946 (code=exited, status=1/FAILURE)

to 
 zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-01-03 18:16:04 CET; 6s ago
  Process: 8197 ExecStart=/usr/local/bin/npm start (code=exited, status=203/EXEC)
 Main PID: 8197 (code=exited, status=203/EXEC)
somebody can help me?
Sorry

trikis1, 2020-01-03:
I solved my problem yet, I change user=pi to user=root
easy.

halicek, 2020-01-23:
I had very similar problem with similar solution (adding "local" to the path) so I will post my log from raspbian for somebody to find it here and to find the solution as well. 

-- Logs begin at Thu 2020-01-23 20:17:01 CET. --
Jan 23 21:16:07 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 23 21:16:07 raspberrypi systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Jan 23 21:16:07 raspberrypi systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Jan 23 21:16:07 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Jan 23 21:16:07 raspberrypi systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Jan 23 21:16:07 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Jan 23 21:16:07 raspberrypi systemd[1]: Failed to start zigbee2mqtt.

oyooyo, 2020-01-24:
I just had a similar problem, the zigbee2mqtt systemd service wouldn't start at bootup. In my case, I found some zigbee2mqtt:error lines stating `Error while opening serialport 'Error: Error: Device or resource busy, cannot open /dev/...` lines in the syslog. Close to these lines there were some other log messages which read `ModemManager: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'`.

I assumed that the problem with zigbee2mqtt not starting might have something to do with the zigbee2mqtt and ModemManager systemd services both trying to access /dev/ttyACM0 at the same time. I tried disabling the ModemManager service via `sudo systemctl disable ModemManager.service` (I assume I don't need that service anyway), and on the next boot the problem was indeed solved.

EpicLPer, 2020-06-01:
More info on this:

If you install it on a Pi Zero which requires you to use [THIS GUIDE](https://gist.github.com/Koenkk/11fe6d4845f5275a2a8791d04ea223cb) instead of the one linked on the Z2M Wiki Page you HAVE to replace the `ExecStart=/usr/bin/npm start` with `ExecStart=/usr/local/bin/npm start` since that guide install the npm binaries into another folder than the usual installer does.

Koenkk, 2020-06-02:
@EpicLPer updated, thanks!

armaurer88, 2020-12-09:
Hi Koenkk,

Following the guide, after creating the configuration file I had the same problem when checking the status. Additionally I ran:

sudo systemctl daemon-reload
sudo systemctl enable zigbee2mqtt.service

And then, started the service, checked the status and it was working like a charm (I didnt change the ExcecStart path nor the user) Using Raspberry pi 3+

Hope it is helpful!

rekcodocker, 2020-12-13:
Same happened to me. 
It is because I installed 'npm' by downloading and untarring the file, and then - as instructed - copied npm to /user/local/bin. 

The instructions of Zigbee2mqtt assume that npm is in /usr/bin.  

Very confusing because it works if you run it from the commandline (npm start) but it does not run as a service.

Suggested fix: Do not assume that npm is necessarily in /usr/bin.

EddieDSuza, 2021-03-26:
Does any one know the npm location for Homebridge.... Pretty much stuck at the ExecStar location. 

Any help would appreciate, On HOOBS it works with the default npm location

```
zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-03-26 14:49:47 -03; 1min 28s ago
  Process: 2737 ExecStart=/usr/bin/npm start (code=exited, status=203/EXEC)
 Main PID: 2737 (code=exited, status=203/EXEC)

Mar 26 14:49:47 homebridge systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Mar 26 14:49:47 homebridge systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Mar 26 14:49:47 homebridge systemd[1]: Stopped zigbee2mqtt.
Mar 26 14:49:47 homebridge systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Mar 26 14:49:47 homebridge systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 26 14:49:47 homebridge systemd[1]: Failed to start zigbee2mqtt.
```

EddieDSuza, 2021-04-16:
For those running homebridge use the below

```
Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```

vladimir9363, 2021-04-29:
Cant start Zigbee2mqtt as service
Help please.I use openhub. I installed zigbeetomqtt on my raspberry. I followed the instructions strictly, but I can't start the default service. when I write ( systemctl status zigbee2mqtt. service) linux outputs the following: openhabian@openhabian:/opt/zigbee2mqtt $ systemctl status zigbee2mqtt. service
● zigbee2mqtt.service - zigbee2mqtt
 Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
 Active: failed (Result: exit-code) since Thu 2021-04-29 12:30:10 +11; 44s ago
 Main PID: 11111 (code=exited, status=217/USER)

Apr 29 12:30:10 openhabian systemd[1]: zigbee2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Apr 29 12:30:10 openhabian systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Apr 29 12:30:10 openhabian systemd[1]: Stopped zigbee2mqtt.
Apr 29 12:30:10 openhabian systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Apr 29 12:30:10 openhabian
I tried all the options that were discussed in this chat. I have a slightly different error (code = exited, status = 217 / USER)

MarcinJakubczyk, 2021-05-22:
Try do that:
```
$ which npm
/home/pi/.nvm/versions/node/v14.16.1/bin/npm
$ ls -l /home/pi/.nvm/versions/node/v14.16.1/bin/npm
lrwxrwxrwx 1 pi pi 38 kwi  6 18:52 /home/pi/.nvm/versions/node/v14.16.1/bin/npm -> ../lib/node_modules/npm/bin/npm-cli.js
```
So I modified zigbee2mqtt.service ExecStart line to:
`ExecStart=/home/pi/.nvm/versions/node/v14.16.1/lib/node_modules/npm/bin/npm-cli.js start`

Now it works for me.

DennisJohnsen, 2021-11-29:
### Homebridge Pi Image installs
If you've used the Homebridge Pi Image installation method, you need to use a different value for your `ExecStart` parameter!
This is because the location of NPM is different.

Change the line to be `ExecStart=npm start`

Thanks to https://github.com/Koenkk/zigbee2mqtt/issues/1493#issuecomment-821085337 for his reply.
I've made this comment to highlight it more for others that come by with the same specific issue.

FreekBos, 2022-02-12:
Hi all, I have the same and haven't found a solution yet. This is my logging and I have no idea where to start
Running CentOS Stream 9 but the result is the same on 8 and 7.
Any idea? Please help

--
Feb 12 12:34:17 h13-z2m-03 systemd[1]: z2m-01.service: Main process exited, code=exited, status=1/FAILURE
Feb 12 12:34:17 h13-z2m-03 systemd[1]: z2m-01.service: Failed with result 'exit-code'.
Feb 12 12:34:17 h13-z2m-03 systemd[1]: z2m-01.service: Consumed 2.483s CPU time.
Feb 12 12:34:17 h13-z2m-03 systemd[1]: z2m-01.service: Scheduled restart job, restart counter is at 17.
Feb 12 12:34:17 h13-z2m-03 systemd[1]: Stopped z2m-01.
Feb 12 12:34:17 h13-z2m-03 systemd[1]: z2m-01.service: Consumed 2.483s CPU time.
Feb 12 12:34:17 h13-z2m-03 systemd[1]: Started z2m-01.
Feb 12 12:34:18 h13-z2m-03 npm[4678]: > zigbee2mqtt@1.23.0 start
Feb 12 12:34:18 h13-z2m-03 npm[4678]: > node index.js
**Feb 12 12:34:19 h13-z2m-03 npm[4689]: **Error: EACCES: permission denied, mkdir '/opt/z2m-01/data/log/2022-02-12.12-34-19'****
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at Object.mkdirSync (node:fs:1334:3)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at mkdirSyncRecursive (/opt/z2m-01/node_modules/mkdir-recursive/index.js:199:16)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at mkdirSyncRecursive (/opt/z2m-01/node_modules/mkdir-recursive/index.js:197:12)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at mkdirSyncRecursive (/opt/z2m-01/node_modules/mkdir-recursive/index.js:197:12)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at mkdirSyncRecursive (/opt/z2m-01/node_modules/mkdir-recursive/index.js:197:12)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at mkdirSyncRecursive (/opt/z2m-01/node_modules/mkdir-recursive/index.js:197:12)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at Object.mkdirSync (/opt/z2m-01/node_modules/mkdir-recursive/index.js:82:10)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at Object.<anonymous> (/opt/z2m-01/lib/util/logger.ts:22:8)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at Module._compile (node:internal/modules/cjs/loader:1101:14)
Feb 12 12:34:19 h13-z2m-03 npm[4689]:     at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
Feb 12 12:34:19 h13-z2m-03 systemd[1]: z2m-01.service: Main process exited, code=exited, status=1/FAILURE
Feb 12 12:34:19 h13-z2m-03 systemd[1]: z2m-01.service: Failed with result 'exit-code'.
Feb 12 12:34:19 h13-z2m-03 systemd[1]: z2m-01.service: Consumed 2.139s CPU time.
Feb 12 12:34:20 h13-z2m-03 systemd[1]: z2m-01.service: Scheduled restart job, restart counter is at 18.
Feb 12 12:34:20 h13-z2m-03 systemd[1]: Stopped z2m-01.
Feb 12 12:34:20 h13-z2m-03 systemd[1]: z2m-01.service: Consumed 2.139s CPU time.
Feb 12 12:34:20 h13-z2m-03 systemd[1]: Started z2m-01.
Feb 12 12:34:20 h13-z2m-03 npm[4699]: > zigbee2mqtt@1.23.0 start
--

csidirop, 2022-02-22:
Same here on [RockyOS ](https://rockylinux.org/) (former CentOS 8). I tried different npm/node versions. But still the same result. 
Running npm directly works, but not as a service. 

@Koenkk Btw it seams that npm v8 works too. Maybe you could test it and add it to the list of supported versions [here](https://www.zigbee2mqtt.io/guide/installation/01_linux.html#installing)

csidirop, 2022-02-25:
Here is a workaround working on Rocky Linux (and probably on CentOS and RHEL):

Instead of relying on systemctl to manage zigbee2mqtt we can use [pm2](https://www.npmjs.com/package/pm2) for that. A quick guide:

1. Install pm2:
    - install pm2 globally: ``npm install pm2 -g``
2. Setup pm2: 
    - create a startup script for pm2: `pm2 startup`
    - copy/paste the command created. This should be something like this `sudo env PATH=$PATH:/usr/bin /usr/local/lib/node_modules/pm2/bin/pm2 startup systemd -u USERNAME --hp /home/USERNAME`
 3. Let pm2 manage zigbee2mqtt:
    - run zigbee2mqtt through pm2: `pm2 start --name=zigbee2mqtt npm -- start` (assuming you are in the zigbee2mqtt directory)
    - check if zigbee2mqtt runs: `pm2 ps` or `pm2 list`
    - save process(es) for restart after reboot: `pm2 save`

Now you can run zigbee2mqtt in background managed by pm2 while pm2 is controlled by systemctl: `systemctl status pm2-USERNAME`

calderonix, 2022-06-20:
Thanks. pm2 worked for me under Debian 11.

csidirop, 2022-07-09:
Btw, this issue and more vanished after I switched from the [non recomended Texas Instruments CC2531](https://www.zigbee2mqtt.io/guide/adapters/#not-recommended) to the better and [recomended Slaesh's CC2652RB stick ](https://www.zigbee2mqtt.io/guide/adapters/#recommended). 

So with the new adapter the service worked like expected.

rmurken, 2024-01-02:
> =npm

I'm a homebridge user, and solved it by including the following two pieces in zigbee2mqtt.service:

`Environment=PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/opt/homebridge/bin:/opt/homebridge/bin
ExecStart=/opt/homebridge/bin/npm start`

donglj2023, 2025-01-06:
SELINUX=disable