Koenkk/zigbee2mqtt #1494: Ikea remote control not working

Ulli2k, 2019-05-04:
Hello,

I tried to pair the Ikea remote control E1524 without success.
I got the following debug output but after that I do not receive any messages by clicking the remote buttons.
(The remote was located very close to the Coordinator)

` zigbee2mqtt:info 5/4/2019, 2:01:28 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"unknown","coordinator":20181024,"log_level":"debug","permit_join":true}'

  zigbee2mqtt:debug 5/4/2019, 2:01:47 PM Saving state to file /app/data/state.json

  zigbee2mqtt:debug 5/4/2019, 2:01:54 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI remote control","powerSource":3}}' of device 'undefined' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:info 5/4/2019, 2:01:54 PM New device 'undefined' with address 0x000d6ffffe177cc9 connected!

  zigbee2mqtt:debug 5/4/2019, 2:01:54 PM Saving state to file /app/data/state.json

  zigbee2mqtt:info 5/4/2019, 2:01:54 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x000d6ffffe177cc9","meta":{}}'

  zigbee2mqtt:warn 5/4/2019, 2:01:54 PM Device with modelID 'undefined' is not supported.

  zigbee2mqtt:warn 5/4/2019, 2:01:54 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

  zigbee2mqtt:debug 5/4/2019, 2:01:56 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"dateCode":"20170302","swBuildId":"1.2.214"}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:info 5/4/2019, 2:02:00 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002e2e0f5', payload '{"illuminance":6,"linkquality":21,"occupancy":false,"battery":97,"voltage":2995}'

  zigbee2mqtt:debug 5/4/2019, 2:02:04 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden"}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:06 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"10":{"type":"Buffer","data":[69,49,53,50,52]},"modelId":"TRADFRI remote control","dateCode":"20170302","powerSource":3,"swBuildId":"1.2.214"}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:08 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"65533":1}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:08 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:09 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":4}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002e2e0f5) of endpoint 1

  zigbee2mqtt:info 5/4/2019, 2:02:09 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002e2e0f5', payload '{"illuminance":4,"linkquality":7,"occupancy":false,"battery":97,"voltage":2995}'

  zigbee2mqtt:debug 5/4/2019, 2:02:09 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":4}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002e2e0f5) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:09 PM Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002e2e0f5) of endpoint 1

  zigbee2mqtt:info 5/4/2019, 2:02:09 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002e2e0f5', payload '{"illuminance":4,"linkquality":7,"occupancy":true,"battery":97,"voltage":2995}'

  zigbee2mqtt:debug 5/4/2019, 2:02:12 PM Received zigbee message of type 'readRsp' with data '{"cid":"genPowerCfg","data":{"65533":1,"batteryVoltage":28,"batteryPercentageRemaining":60}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:info 5/4/2019, 2:02:12 PM MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe177cc9', payload '{"battery":60,"linkquality":141}'

  zigbee2mqtt:debug 5/4/2019, 2:02:12 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:16 PM Received zigbee message of type 'readRsp' with data '{"cid":"genIdentify","data":{"65533":1,"identifyTime":0}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:16 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:18 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:20 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:22 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:24 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:29 PM Received zigbee message of type 'readRsp' with data '{"cid":"genAlarms","data":{"65533":1}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:29 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:31 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:35 PM Received zigbee message of type 'readRsp' with data '{"cid":"haDiagnostic","data":{"65533":1}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:35 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:39 PM Received zigbee message of type 'readRsp' with data '{"cid":"lightLink","data":{"65533":1}}' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:debug 5/4/2019, 2:02:39 PM Received zigbee message of type 'devInterview' with data '"0x000d6ffffe177cc9"'

  zigbee2mqtt:debug 5/4/2019, 2:02:39 PM Accept device incoming with ieeeAddr '0x000d6ffffe177cc9' permit join is 'true'

  zigbee2mqtt:debug 5/4/2019, 2:02:39 PM Allowing device '0x000d6ffffe177cc9' to join

  zigbee2mqtt:debug 5/4/2019, 2:02:39 PM Received zigbee message of type 'devIncoming' with data '"0x000d6ffffe177cc9"' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1

  zigbee2mqtt:info 5/4/2019, 2:02:39 PM Device incoming...

  zigbee2mqtt:info 5/4/2019, 2:02:39 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'

  zigbee2mqtt:debug 5/4/2019, 2:02:39 PM Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI remote control' (0x000d6ffffe177cc9) of endpoint 1`

bertouttier, 2019-05-04:
I got the same issue. The remote is correctly paired to the cc2530 coordinator, but I do not see actions coming in. If I sniff the traffic, the remote seems to send out ZHA packets correctly...

Koenkk, 2019-05-05:
Make sure you are at least running the `20190223` firmware.

Damme, 2019-05-10:
I'm jumping into here instead of creating a new issue. 
I am really confused about the Ikea E1524. The documentation seams a bit thin on this area.
I updated my firmware to latest just the other day. 20190223. I am running latest dev from github.

Do I need to sniff traffic as it was said in another thread a while back? I was following the long discussion then trying to get this remote to work. Or is this issue resolved and not needed any more? I can pair mine but I don't see any messages then pressing buttons.

Koenkk, 2019-05-10:
With the latest zigbee2mqtt version and firmware, just pairing is enough.

capretto78, 2019-05-10:
i had the same problem but with the firmware 20190223 i can connect correctly remote tradfri.
Now my problem: how can i use it to do some automations?
What king of command can i use? thanks

adn77, 2019-05-12:
According to the docs (http://www.zigbee2mqtt.io/information/binding.html) the E1524 is not capable of binding. Hence I still had to sniff the traffic in order to find out the group the remote was sending its commands to.
Once I had the group id, I created the group with a friendly name (e.g. Remote_1) and added my lights to that group (http://www.zigbee2mqtt.io/information/groups.html).

Now the remote controls the lights directly and my MQTT bridge still can read the remote's commands.

Ulli2k, 2019-05-27:
Meanwhile, I fixed my issue by updating the firmware. Thanks.

marima84, 2019-05-27:
> Meanwhile, I fixed my issue by updating the firmware. Thanks.

@Ulli2k are you using a CC2531 coordinator? I've recently updated to the latest firmware `20190425` but when I try to add my Tradfri remote (E1524) I have no success:

`PM Device with modelID 'undefined' is not supported.`

crazyquark, 2019-06-28:
Anyway to make this work without updating the remote firmware? I do not have the Tradfri gateway...

Koenkk, 2019-06-28:
You have to update the sniffer firmware, not the remote control firmware.

crazyquark, 2019-06-30:
Awesome, updating the CC2531 stick fw fixed it for me.

closed 2019-06-30 by Koenkk