Koenkk/zigbee2mqtt #1496: IKEA E1524 ON/OFF button (in the middle) with issue if long time pressing

Maik-100, 2019-05-04:
I see in log:
zigbee2mqtt:info 2019-5-4 17:35:26 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":18,"duration":9.029,"battery":87,"action":"**toggle**"}'
if I press the ON/OFF button in the middle only short time
-> that is ok

If I press it for long time, I see in log:
zigbee2mqtt:info 2019-5-4 17:37:26 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":21,"duration":9.029,"battery":87,"action":"**toggle**"}'
zigbee2mqtt:info 2019-5-4 17:37:27 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":23,"duration":9.029,"battery":74}'
zigbee2mqtt:info 2019-5-4 17:37:30 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":28,"duration":9.029,"battery":74,"action":"**toggle_hold**"}'
zigbee2mqtt:info 2019-5-4 17:37:30 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":26,"duration":9.029,"battery":74,"action":"**arrow_right_click**"}'
-> that is not ok
I do not press the arrow_right button.

If I press the button "brightness_up" (or one of the other button) for a long time I got in the log:
zigbee2mqtt:info 2019-5-4 17:38:58 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":31,"duration":9.029,"battery":74}'
zigbee2mqtt:info 2019-5-4 17:38:58 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":31,"duration":9.029,"battery":74,"action":"**brightness_up_hold**"}'
zigbee2mqtt:info 2019-5-4 17:38:58 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":31,"duration":9.029,"battery":74,"action":"**brightness_up_release**"}'
-> that is ok

zigbee2mqtt:info 2019-5-4 16:55:44 Starting **zigbee2mqtt version 1.3.1** (commit #39d6a8e)
  zigbee2mqtt:info 2019-5-4 16:55:44 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-5-4 16:55:49 zigbee-shepherd started
  zigbee2mqtt:info 2019-5-4 16:55:49 **Coordinator firmware version: '20190223'**
  zigbee2mqtt:info 2019-5-4 16:55:49 Currently 4 devices are joined:
  zigbee2mqtt:info 2019-5-4 16:55:49 0x000d6ffffed15a64 (0x000d6ffffed15a64): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
  zigbee2mqtt:info 2019-5-4 16:55:49 0x000d6ffffee8053d (0x000d6ffffee8053d): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
  zigbee2mqtt:info 2019-5-4 16:55:49 0x000d6ffffe699d8b (0x000d6ffffe699d8b): E1525 - IKEA TRADFRI motion sensor (EndDevice)
  zigbee2mqtt:info 2019-5-4 16:55:49 0x000d6ffffe18e00a (0x000d6ffffe18e00a): **E1524 - IKEA TRADFRI remote control (EndDevice)**
  zigbee2mqtt:warn 2019-5-4 16:55:49 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-5-4 16:55:49 Allowing new devices to join.
  zigbee2mqtt:warn 2019-5-4 16:55:49 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-5-4 16:55:49 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-5-4 16:55:49 Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:info 2019-5-4 16:55:50 zigbee-shepherd ready
  zigbee2mqtt:info 2019-5-4 16:55:51 Connected to MQTT server
  zigbee2mqtt:info 2019-5-4 16:55:51 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-5-4 16:55:51 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffed15a64', payload '{"state":"ON","linkquality":0,"brightness":127}'
  zigbee2mqtt:info 2019-5-4 16:55:51 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffee8053d', payload '{"state":"ON","linkquality":0,"brightness":127}'
  zigbee2mqtt:info 2019-5-4 16:55:51 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe699d8b', payload '{"battery":87,"linkquality":5,"occupancy":false}'
  zigbee2mqtt:info 2019-5-4 16:55:51 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":23,"duration":0.144,"battery":87}'
  zigbee2mqtt:info 2019-5-4 16:55:52 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"39d6a8e","coordinator_firmware":20190223,"log_level":"info","permit_join":true}'

Koenkk, 2019-05-05:
Could you post the debug log? http://www.zigbee2mqtt.io/configuration/configuration.html (`log_level: debug`).

Maik-100, 2019-05-09:
First the configuration file:

pi@raspberrypi:/opt/zigbee2mqtt/data $ cat configuration.yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost:1883'
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  log_level: info/debug
  channel: 20
report: true
devices:
  '0x000d6ffffed15a64':
    friendly_name: '0x000d6ffffed15a64'
    retain: false
  '0x000d6ffffee8053d':
    friendly_name: '0x000d6ffffee8053d'
    retain: false
  '0x000d6ffffe18e00a':
    friendly_name: '0x000d6ffffe18e00a'
    retain: false
  '0x000d6ffffe699d8b':
    friendly_name: '0x000d6ffffe699d8b'
    retain: false
groups:
  '60068':
    friendly_name: group_E1524
        members:
  '0x000d6ffffe18e00a'
pi@raspberrypi:/opt/zigbee2mqtt/data $

Here it is.
I pressed all five buttons one after one. First time only short to get the click event, Second time for longer to get first die hold event, than the release event. The first 4 buttons works well, the last one (ON/OFF in the middle) has got the problems:

pi@raspberrypi:/opt/zigbee2mqtt/data/log $ cat 2019-05-09.08-30-47/log.txt
2019-5-9 08:30:47 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-09.08-30-47'
2019-5-9 08:30:47 - debug: Removing old log directory '/opt/zigbee2mqtt/data/log/2019-04-08.23-35-50'
2019-5-9 08:30:53 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[20],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2019-5-9 08:30:54 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-5-9 08:30:54 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-5-9 08:30:54 - info: Starting zigbee2mqtt version 1.3.1 (commit #39d6a8e)
2019-5-9 08:30:54 - info: Starting zigbee-shepherd
2019-5-9 08:30:59 - info: zigbee-shepherd started
2019-5-9 08:30:59 - info: Coordinator firmware version: '20190223'
2019-5-9 08:30:59 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":20,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b00193618c6","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190223},"startTime":1557383459,"joinTimeLeft":0}
2019-5-9 08:30:59 - info: Currently 4 devices are joined:
2019-5-9 08:30:59 - info: 0x000d6ffffed15a64 (0x000d6ffffed15a64): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
2019-5-9 08:30:59 - info: 0x000d6ffffee8053d (0x000d6ffffee8053d): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
2019-5-9 08:30:59 - info: 0x000d6ffffe699d8b (0x000d6ffffe699d8b): E1525 - IKEA TRADFRI motion sensor (EndDevice)
2019-5-9 08:30:59 - info: 0x000d6ffffe18e00a (0x000d6ffffe18e00a): E1524 - IKEA TRADFRI remote control (EndDevice)
2019-5-9 08:30:59 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-5-9 08:30:59 - warn: Allowing new devices to join.
2019-5-9 08:30:59 - warn: Set `permit_join` to `false` once you joined all devices.
2019-5-9 08:30:59 - info: Zigbee: allowing new devices to join.
2019-5-9 08:30:59 - info: Connecting to MQTT server at mqtt://localhost:1883
2019-5-9 08:31:00 - debug: Configuring 0x000d6ffffe699d8b (0x000d6ffffe699d8b) ...
2019-5-9 08:31:01 - info: zigbee-shepherd ready
2019-5-9 08:31:01 - info: Connected to MQTT server
2019-5-9 08:31:01 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-5-9 08:31:01 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffed15a64', payload '{"state":"ON","linkquality":10,"brightness":127}'
2019-5-9 08:31:01 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffee8053d', payload '{"state":"ON","linkquality":23,"brightness":35}'
2019-5-9 08:31:01 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe699d8b', payload '{"battery":87,"linkquality":10,"occupancy":false}'
2019-5-9 08:31:01 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":44,"duration":9.029,"battery":87}'
2019-5-9 08:31:03 - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"39d6a8e","coordinator_firmware":20190223,"log_level":"debug","permit_join":true}'
2019-5-9 08:31:03 - debug: Mounted the cieApp (epId 11)
2019-5-9 08:31:52 - warn: Failed to configure 0x000d6ffffe699d8b (0x000d6ffffe699d8b) ('Error: Timed out after 10000 ms') (attempt #1)
[... 50 lines cut ...]
2019-5-9 08:38:20 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":11914}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-9 08:38:22 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":11914}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-9 08:38:24 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":11914}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-9 08:38:34 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
[... cut ...]
2019-5-9 08:39:13 - debug: Received zigbee message of type 'cmdTradfriArrowSingle' with data '{"cid":"genScenes","data":{"value":2}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-9 08:39:13 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":13,"duration":11.914,"battery":74,"action":"arrow_right_click"}'
2019-5-9 08:40:46 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-5-9 08:40:46 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
2019-5-9 08:40:46 - info: Disconnecting from MQTT server
pi@raspberrypi:/opt/zigbee2mqtt/data/log $

Koenkk, 2019-05-10:
Fixed in the latest dev branch, thanks!

Maik-100, 2019-05-21:
I retest this issue, but the fix does not work.
A short pressing of the button works for me, but long pressing gives an error:

pi@raspberrypi:/opt/zigbee2mqtt/data $
pi@raspberrypi:/opt/zigbee2mqtt/data $ npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

  zigbee2mqtt:info 2019-5-21 21:57:00 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-21.21-57-00'
  zigbee2mqtt:debug 2019-5-21 21:57:01 Removing old log directory '/opt/zigbee2mqtt/data/log/2019-05-09.08-41-42'
  zigbee2mqtt:debug 2019-5-21 21:57:05 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[20],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-5-21 21:57:06 Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 2019-5-21 21:57:06 Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 2019-5-21 21:57:06 Starting zigbee2mqtt version **1.4.0 (commit #6b75465)**
  zigbee2mqtt:info 2019-5-21 21:57:06 Starting zigbee-shepherd

  zigbee2mqtt:info 2019-5-21 21:57:12 zigbee-shepherd started
  zigbee2mqtt:info 2019-5-21 21:57:12 **Coordinator firmware version: '20190223'**
  zigbee2mqtt:debug 2019-5-21 21:57:12 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"20","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b00193618c6","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20190223},"startTime":1558468632,"joinTimeLeft":0}
  zigbee2mqtt:info 2019-5-21 21:57:12 Currently 4 devices are joined:
  zigbee2mqtt:info 2019-5-21 21:57:12 0x000d6ffffed15a64 (0x000d6ffffed15a64): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
  zigbee2mqtt:info 2019-5-21 21:57:12 0x000d6ffffee8053d (0x000d6ffffee8053d): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
  zigbee2mqtt:info 2019-5-21 21:57:12 0x000d6ffffe699d8b (0x000d6ffffe699d8b): E1525 - IKEA TRADFRI motion sensor (EndDevice)
  zigbee2mqtt:info 2019-5-21 21:57:12 0x000d6ffffe18e00a (0x000d6ffffe18e00a): E1524 - IKEA TRADFRI remote control (EndDevice)
  zigbee2mqtt:warn 2019-5-21 21:57:12 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2019-5-21 21:57:12 Allowing new devices to join.
  zigbee2mqtt:warn 2019-5-21 21:57:12 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2019-5-21 21:57:12 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-5-21 21:57:12 Connecting to MQTT server at mqtt://localhost:1883
  zigbee2mqtt:debug 2019-5-21 21:57:13 Configuring 0x000d6ffffe699d8b (0x000d6ffffe699d8b) ...
  zigbee2mqtt:info 2019-5-21 21:57:14 zigbee-shepherd ready
  zigbee2mqtt:info 2019-5-21 21:57:14 Connected to MQTT server
  zigbee2mqtt:info 2019-5-21 21:57:14 MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 2019-5-21 21:57:14 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffed15a64', payload '{"state":"ON","linkquality":10,"brightness":127}'
  zigbee2mqtt:info 2019-5-21 21:57:14 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffee8053d', payload '{"state":"ON","linkquality":23,"brightness":35}'
  zigbee2mqtt:info 2019-5-21 21:57:14 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe699d8b', payload '{"battery":87,"linkquality":2,"occupancy":false}'
  zigbee2mqtt:info 2019-5-21 21:57:14 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":21,"battery":87}'
  zigbee2mqtt:debug 2019-5-21 21:57:15 Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data ''
  zigbee2mqtt:info 2019-5-21 21:57:15 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b00193618c6","type":"Coordinator"},{"ieeeAddr":"0x000d6ffffed15a64","type":"Router","model":"LED1623G12","friendly_name":"0x000d6ffffed15a64","nwkAddr":41749,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 W opal 1000lm","hwVersion":1,"dateCode":"20170302"},{"ieeeAddr":"0x000d6ffffee8053d","type":"Router","model":"LED1623G12","friendly_name":"0x000d6ffffee8053d","nwkAddr":48972,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 W opal 1000lm","hwVersion":1,"dateCode":"20170302"},{"ieeeAddr":"0x000d6ffffe699d8b","type":"EndDevice","model":"E1525","friendly_name":"0x000d6ffffe699d8b","nwkAddr":6645,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI motion sensor","hwVersion":1,"dateCode":"20170302"},{"ieeeAddr":"0x000d6ffffe18e00a","type":"EndDevice","model":"E1524","friendly_name":"0x000d6ffffe18e00a","nwkAddr":33894,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","hwVersion":1,"dateCode":"20170302"}]}'
  zigbee2mqtt:debug 2019-5-21 21:57:15 Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 2019-5-21 21:57:15 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2019-5-21 21:57:15 MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"6b75465","coordinator":20190223,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 2019-5-21 21:57:16 Mounted the cieApp (epId 11)
[... cut ...]
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1,"srcaddr":6645,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":0,"securityuse":0,"timestamp":13696990,"transseqnumber":0,"len":4,"data":{"0":8,"1":8,"2":7,"3":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":0},"manufCode":0,"seqNum":8,"cmdId":"configReportRsp","payload":[{"status":0}]}} +29ms
  zigbee-shepherd-converters:devices Configured '(cb) => device.foundation('genPowerCfg', 'configReport', [cfg], foundationCfg, cb)' with result 'OK' +2s
  zigbee2mqtt:info 2019-5-21 21:52:11 Successfully configured 0x000d6ffffe699d8b (0x000d6ffffe699d8b)

Koenkk, 2019-05-22:
Please don't post logs here but post them on pastebin.com, could you indicate what part is still unexpected?

Maik-100, 2019-05-22:
I see in log if I press the button a short time:

zigbee2mqtt:debug 2019-5-21 21:57:36 Received zigbee message of type '**cmdToggle**' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
zigbee2mqtt:info 2019-5-21 21:57:36 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":13,"battery":87,"action":"**toggle**"}'

-> still ok!

If I press it for long time, I see in log:

zigbee2mqtt:debug 2019-5-21 21:57:43 Received zigbee message of type '**cmdToggle**' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
zigbee2mqtt:info 2019-5-21 21:57:43 MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":0,"battery":87,"action":"**toggle**"}'
zigbee2mqtt:debug 2019-5-21 21:57:43 Received MQTT message on 'zigbee2mqtt/0x000d6ffffed15a64/set' with data '{"state": "**OFF**"}'
zigbee2mqtt:debug 2019-5-21 21:57:43 Received MQTT message on 'zigbee2mqtt/0x000d6ffffee8053d/set' with data '{"state": "**OFF**"}'
zigbee2mqtt:info 2019-5-21 21:57:43 Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - **null**
zigbee2mqtt:error 2019-5-21 21:57:43 Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - **null failed with error Error: AF data request fails, status code: 233. MAC no ack.**
zigbee2mqtt:info 2019-5-21 21:57:43 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"**Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffed15a64","type":"device","friendlyName":"0x000d6ffffed15a64"},"message":"{"state": "OFF"}"}}'**
zigbee2mqtt:info 2019-5-21 21:57:43 Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - **null**

-> the behavior changed (from zigbee2mqtt version 1.3.1) but it is still not ok!

Koenkk, 2019-05-23:
For how long do you hold it now? As the message reported by the button is the same as the single click.

Maik-100, 2019-05-23:
I pressed it for ca. 5 seconds.

Koenkk, 2019-05-24:
I'm not sure what happend but it seems that it's not reporting the messages anymore which it did in https://github.com/Koenkk/zigbee2mqtt/issues/1496#issue-440338355 (to distinguish between hold and click). Can you try to re-pair the device?

Maik-100, 2019-05-24:
OK, I did re-pairing: strange behaviour:

I press for a very short time (for ca. 100ms) the on/off button in the middle. some times I get in log:
2019-5-24 08:43:49 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":10,"battery":47,"action":"**toggle**"}'
-> ok

but most time I get (press very short time):
2019-5-24 08:43:57 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":5,"battery":47,"action":"**toggle**"}'
2019-5-24 08:43:57 - info: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - **null**
2019-5-24 08:43:57 - error: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - **null failed with error Error: AF d**ata request fails, status code: 233. MAC no ack.
2019-5-24 08:43:57 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"**Error: AF data request fails, status code: 233.** MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffed15a64","type":"device","friendlyName":"0x000d6ffffed15a64"},"message":"{\"state\": \"OFF\"}"}}'
2019-5-24 08:43:57 - info: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-24 08:43:57 - error: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 08:43:57 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffee8053d","type":"device","friendlyName":"0x000d6ffffee8053d"},"message":"{\"state\": **\"OFF\"}"}}'**
-> not ok

I press for a long time the on/off button in the middle. some times I get in log:

2019-5-24 08:44:16 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":10,"battery":47,"action":"**toggle**"}'
2019-5-24 08:44:16 - info: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-24 08:44:17 - error: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 08:44:17 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffed15a64","type":"device","friendlyName":"0x000d6ffffed15a64"},"message":"{\"state\": \"OFF\"}"}}'
2019-5-24 08:44:17 - info: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-24 08:44:17 - error: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 08:44:17 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffee8053d","type":"device","friendlyName":"0x000d6ffffee8053d"},"message":"{\"state\": \"OFF\"}"}}'
2019-5-24 08:44:20 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":10,"battery":47,"action":"**toggle_hold**"}'

-> not ok, but I see a "toogle_hold" after the error, I see no "toggle_release" at no time.

I play with the button. Here is the log:

2019-5-24 08:45:00 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":0,"battery":47,"action":"**toggle**"}'

2019-5-24 08:45:06 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":5,"battery":47,"action":"**toggle**"}'
2019-5-24 08:45:06 - info: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-24 08:45:06 - error: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 08:45:06 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffed15a64","type":"device","friendlyName":"0x000d6ffffed15a64"},"message":"{\"state\": \"OFF\"}"}}'
2019-5-24 08:45:06 - info: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-24 08:45:06 - error: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 08:45:06 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffee8053d","type":"device","friendlyName":"0x000d6ffffee8053d"},"message":"{\"state\": \"OFF\"}"}}'

2019-5-24 08:45:10 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":5,"battery":47,"action":"**toggle**"}'

[... cut ...]
2019-5-24 09:04:14 - error: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 09:04:14 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffee8053d","type":"device","friendlyName":"0x000d6ffffee8053d"},"message":"{\"state\": \"OFF\"}"}}'

Koenkk, 2019-05-24:
Could you post the debug logging when you release the button (and got a `toggle_hold`)

Maik-100, 2019-05-24:
with debug log:
I press for a short time  the on/off button in the middle. I get in log:

2019-5-24 14:49:12 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-24 14:49:12 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":26,"battery":47,"action":"**toggle**"}'
2019-5-24 14:49:12 - debug: Received MQTT message on 'zigbee2mqtt/0x000d6ffffed15a64/set' with data '{"state": "OFF"}'
2019-5-24 14:49:12 - debug: Received MQTT message on 'zigbee2mqtt/0x000d6ffffee8053d/set' with data '{"state": "OFF"}'
2019-5-24 14:49:12 - info: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2019-5-24 14:49:12 - error: Zigbee publish to device '0x000d6ffffed15a64', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
2019-5-24 14:49:12 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffed15a64","type":"device","friendlyName":"0x000d6ffffed15a64"},"message":"{\"state\": \"OFF\"}"}}'
2019-5-24 14:49:12 - info: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - **null**
2019-5-24 14:49:12 - **error: Zigbee publish to device '0x000d6ffffee8053d', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.**
2019-5-24 14:49:12 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 233. MAC no ack.","meta":{"entity":{"ID":"0x000d6ffffee8053d","type":"device","friendlyName":"0x000d6ffffee8053d"},"message":"{\"state\": \"OFF\"}"}}'
2019-5-24 14:49:15 - info: Successfully configured 0x000d6ffffe18e00a (0x000d6ffffe18e00a)

I press the on/off button:

2019-5-24 14:49:21 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-24 14:49:21 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":23,"battery":47,"action":"**toggle**"}'
2019-5-24 14:49:22 - debug: Received zigbee message of type 'attReport' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":74}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) of endpoint 1
2019-5-24 14:49:22 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":23,"battery":74}'
2019-5-24 14:49:22 - debug: Received zigbee message of type 'devChange' with data '{"cid":"genPowerCfg","data":{"batteryPercentageRemaining":74}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) of endpoint 1
2019-5-24 14:49:24 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-24 14:49:24 - debug: Received zigbee message of type 'cmdMoveToLevelWithOnOff' with data '{"cid":"genLevelCtrl","data":{"level":254,"transtime":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-24 14:49:24 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":21,"battery":74,"action":"**toggle_hold**"}'
2019-5-24 14:49:25 - debug: Received zigbee message of type 'cmdTradfriArrowSingle' with data '{"cid":"genScenes","data":{"value":2}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-5-24 14:49:26 - info: Zigbee publish to device '0x000d6ffffe18e00a', genBasic - readRsp - [] - {"direction":1,"seqNum":68,"disDefaultRsp":1} - 1

I release the button:
No log infos are generated!

Maik-100, 2019-05-31:
Hi Koenkk, how can I help?

I stopped zigbee2mqtt.
I stopped mosquito.
I started mosquito.
I started zigbee2mqtt.

If I press the "brighness_up" botton for a **short time**, I get a message from type 'cmd**Step**WithOnOff'.
{"cid":"genLevelCtrl","data":{"**stepmode**":0,"stepsize":43,"transtime":5}}'

If I press the "brighness_up" botton for a **long time**, I get a message from type 'cmd**Move**WithOnOff'
"cid":"genLevelCtrl","data":{"**movemode**":0,"rate":83}}'

If I **release** the "brighness_up" botton, I get a message from type 'cmd**Stop**WithOnOff'
"cid":"genLevelCtrl","data":{}}'
"action":"brightness_up_release"}'           _but no duration-time_

If I press the "OnOff" botton for a **short time**, I get a message from type 'cmdToggle'
{"cid":"genOnOff","data":{}}'
**Why is "data" empty?**
"action":"**toggle**"}'

If I press the "OnOff" botton for a **long time**, I get **immediately** the **same message** type then I press it only for a short time type 'cmdToggle'.
"cid":"genOnOff","data":{}}'
"action":"**toggle**"}'

I hold the "OnOff" botton **still pressed.**
**After 3 seconds** I get zigbee message of type 'cmdTradfriArrowRelease' 
"cid":"genScenes","data":{"value":0}}' 

I hold the "OnOff" botton **still pressed.**
**After 0 to 1 second** (t tried is some more times to get better results) I get zigbee message of type 'cmdMoveToLevelWithOnOff'
cid":"genLevelCtrl","data":{"level":254,"transtime":0}}'
After this I saw every time: "action":"**toggle_hold**"}'

I hold the "OnOff" botton **still pressed**.
**After 0 to 1 second** I get zigbee message of type '**cmdTradfriArrowSingle**' 
cid":"genScenes","data":{"value":2}}'

I pressed the "OnOff" botton **several times** for a **long time**, but I get **no** *release messages **nor** a *release action...

I pressed the "left arrow" botton it works:
message of type '**cmdTradfriArrowSingle**'
"cid":"genScenes","data":{"value":257}}'
action":"**arrow_left_click**"}'

message of type 'c**mdTradfriArrowHold'** 
"cid":"genScenes","data":{"value":3329}}' 
"action":"**arrow_left_hold**"}'

message of type '**cmdTradfriArrowRelease'**
"cid":"genScenes","data":{"value":7623}}' 
"action":"**arrow_left_release**","**duration":12.219**}'

Same with "right arrow" botton, it works.

Koenkk, 2019-06-03:
Can you share your `database.db`?

Maik-100, 2019-06-05:
I put it in a zip file.

[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/3256160/database.zip)

Koenkk, 2019-06-07:
I've checked again, but indeed the TRADFRI button does not report a hold release, this is because the button doesn't report anything when the toggle button is released (no zigbee messages). Meaning that we cannot know that the button has been released.

Maik-100, 2019-06-16:
I try to  check your assumption.
I have two sticks. I try to use one for sniffering and the other for the coordinator. But it is not working.
If I try to use both in my raspberry, I am not able to specify the stick to sniffer in whsniff.
If t try to sniffer in my WinPC with ZBOSS. But it founds no TI stick.
So I use only the sniffer stick in my raspberry to use whsniff. 
[sniff_2.zip](https://github.com/Koenkk/zigbee2mqtt/files/3294487/sniff_2.zip)

If there is no zigbee message -> no toggle hold release message can generated.
So it is a good idea to change the code in a way that no error comes up.

It is not importent for my to get the toggle hold release message.

Koenkk, 2019-06-17:
You can follow this guide to sniff traffic: http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Maik-100, 2019-06-19:
Ahh, I need to flash the ZBOSS sniffer firmware on the stick. If I use whsniff to sniff, I need a other firmware.
It is working, but I need more time to make it easier for you to read the trace.

[sniff_win.zip](https://github.com/Koenkk/zigbee2mqtt/files/3305811/sniff_win.zip)

Maik-100, 2019-06-24:
OK, I did an FirmwareUpdate to Coordinator firmware version: '20190425' (Starting zigbee2mqtt version 1.4.0 (commit #6b75465)). So I had to rePairing my devices.

I did two time and sniffer in two files (sniff_Win4.pcapng and sniff_Win5.pcapng):
I pressed on/off button only short time (two times).
I pressed on/off button for log time (only one time).

This is debug log:
2019-6-24 10:32:29 - debug: Accept device incoming with ieeeAddr '0x000d6ffffe18e00a' permit join is 'true'
2019-6-24 10:32:29 - debug: Allowing device '0x000d6ffffe18e00a' to join
2019-6-24 10:32:29 - debug: Received zigbee message of type 'devIncoming' with data '"0x000d6ffffe18e00a"' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) of endpoint 1
2019-6-24 10:32:29 - info: Device incoming...
2019-6-24 10:32:29 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-6-24 10:32:29 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) of endpoint 1

2019-6-24 10:34:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-24 10:34:09 - debug: Received zigbee message of type **'cmdToggle'** with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:34:09 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":21,"battery":21,"**action":"toggle**"}'
2019-6-24 10:34:17 - info: Succesfully reenabled joining

2019-6-24 10:36:47 - debug: Received zigbee message of type '**cmdToggle'** with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:36:47 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":10,"battery":21,"action":**"toggle"**}'
2019-6-24 10:37:00 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:00 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":0,"battery":21,"action":"toggle"}'

2019-6-24 10:37:15 - debug: Received zigbee message of type '**cmdToggle**' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:15 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":15,"battery":21,"**action":"toggle**"}'
2019-6-24 10:37:18 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:19 - debug: Received zigbee message of type '**cmdMoveToLevelWithOnOff**' with data '{"cid":"genLevelCtrl","data":{"level":254,"transtime":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:19 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":15,"battery":21,"action":"**toggle_hold**"}'
2019-6-24 10:37:19 - debug: Received zigbee message of type 'cmdTradfriArrowSingle' with data '{"cid":"genScenes","data":{"value":2}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:37 - info: Succesfully reenabled joining

Second try/sniffer file:
[sniff_Win4_5.zip](https://github.com/Koenkk/zigbee2mqtt/files/3320028/sniff_Win4_5.zip)

2019-6-24 10:38:28 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:38:28 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":7,"battery":21,"action":"toggle"}'
2019-6-24 10:38:43 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:38:43 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":18,"battery":21,"action":"toggle"}'
2019-6-24 10:38:56 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:38:56 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":28,"battery":21,"action":"toggle"}'
2019-6-24 10:38:59 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:39:00 - debug: Received zigbee message of type 'cmdMoveToLevelWithOnOff' with data '{"cid":"genLevelCtrl","data":{"level":254,"transtime":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:39:00 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":23,"battery":21,"action":"toggle_hold"}'
2019-6-24 10:39:00 - debug: Received zigbee message of type 'cmdTradfriArrowSingle' with data '{"cid":"genScenes","data":{"value":2}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:39:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-24 10:40:57 - info: Succesfully reenabled joining
2019-6-24 10:44:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-24 10:44:17 - info: Succesfully reenabled joining

Maik-100, 2019-06-24:
OK, I did an FirmwareUpdate to Coordinator firmware version: '20190425' (Starting zigbee2mqtt version 1.4.0 (commit #6b75465)). So I had to rePairing my devices.

I did two time and sniffer in two files (sniff_Win4.pcapng and sniff_Win5.pcapng):
I pressed on/off button only short time (two times).
I pressed on/off button for log time (only one time).

This is debug log:
2019-6-24 10:32:29 - debug: Accept device incoming with ieeeAddr '0x000d6ffffe18e00a' permit join is 'true'
2019-6-24 10:32:29 - debug: Allowing device '0x000d6ffffe18e00a' to join
2019-6-24 10:32:29 - debug: Received zigbee message of type 'devIncoming' with data '"0x000d6ffffe18e00a"' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) of endpoint 1
2019-6-24 10:32:29 - info: Device incoming...
2019-6-24 10:32:29 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
2019-6-24 10:32:29 - debug: Received zigbee message of type 'devStatus' with data '"online"' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) of endpoint 1

2019-6-24 10:34:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-24 10:34:09 - debug: Received zigbee message of type **'cmdToggle'** with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:34:09 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":21,"battery":21,"**action":"toggle**"}'
2019-6-24 10:34:17 - info: Succesfully reenabled joining

2019-6-24 10:36:47 - debug: Received zigbee message of type '**cmdToggle'** with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:36:47 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":10,"battery":21,"action":**"toggle"**}'
2019-6-24 10:37:00 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:00 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":0,"battery":21,"action":"toggle"}'

2019-6-24 10:37:15 - debug: Received zigbee message of type '**cmdToggle**' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:15 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":15,"battery":21,"**action":"toggle**"}'
2019-6-24 10:37:18 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:19 - debug: Received zigbee message of type '**cmdMoveToLevelWithOnOff**' with data '{"cid":"genLevelCtrl","data":{"level":254,"transtime":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:19 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":15,"battery":21,"action":"**toggle_hold**"}'
2019-6-24 10:37:19 - debug: Received zigbee message of type 'cmdTradfriArrowSingle' with data '{"cid":"genScenes","data":{"value":2}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:37:37 - info: Succesfully reenabled joining

Second try/sniffer file:

2019-6-24 10:38:28 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:38:28 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":7,"battery":21,"action":"toggle"}'
2019-6-24 10:38:43 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:38:43 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":18,"battery":21,"action":"toggle"}'
2019-6-24 10:38:56 - debug: Received zigbee message of type 'cmdToggle' with data '{"cid":"genOnOff","data":{}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:38:56 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":28,"battery":21,"action":"toggle"}'
2019-6-24 10:38:59 - debug: Received zigbee message of type 'cmdTradfriArrowRelease' with data '{"cid":"genScenes","data":{"value":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:39:00 - debug: Received zigbee message of type 'cmdMoveToLevelWithOnOff' with data '{"cid":"genLevelCtrl","data":{"level":254,"transtime":0}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:39:00 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":23,"battery":21,"action":"toggle_hold"}'
2019-6-24 10:39:00 - debug: Received zigbee message of type 'cmdTradfriArrowSingle' with data '{"cid":"genScenes","data":{"value":2}}' of device 'TRADFRI remote control' (0x000d6ffffe18e00a) with groupID undefined of endpoint 1
2019-6-24 10:39:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-24 10:40:57 - info: Succesfully reenabled joining
2019-6-24 10:44:07 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-6-24 10:44:17 - info: Succesfully reenabled joining

Koenkk, 2019-06-24:
Without annotation when what happend I'm not able to read this.

What you should do is to press the button and release it after a few seconds. Do you see a new zigbee message (`Received zigbee message`) on release? If so, please indicate which one.

Maik-100, 2019-06-28:
In the moment I release the button, I do not see any message in wireshark. I try it thee times.

[Downloads.zip](https://github.com/Koenkk/zigbee2mqtt/files/3340932/Downloads.zip)

Koenkk, 2019-06-28:
That is consistent with my findings in https://github.com/Koenkk/zigbee2mqtt/issues/1496#issuecomment-500049760

Conclusion: the remote doesnt send anything on hold release which means we cannot respond to it.

closed 2019-06-28 by Koenkk

Maik-100, 2019-06-29:
ok, but please  fix the problem with the 
Received zigbee message of type 'cmdTradfriArrowRelease' 
and
Received zigbee message of type 'cmdMoveToLevelWithOnOff'.

Look in the wireshark trace: If a press the button, after about 3 seconds (the button is still pressed) I got some zigbee messages (unknown command). I think zigbee2mqtt decode them to message of type 'cmdTradfriArrowRelease' and 'cmdMoveToLevelWithOnOff'.
I release the button after about 15 or 20 seconds. At the moment I release the button, no zigbee messages are sniffered.

I leave home for holidays for some weeks. So I am not able to test...

Koenkk, 2019-07-01:
But these message come from the actual remote, I don't understand what the problem is, could you explain this further?

Maik-100, 2019-07-08:
Yes, these messges come from the remote. But what is the intension of these messges? What will IKEA tell us? These messages make no sens at this state/time at the state-machine. 
The wireshark trace tells me, after about 3 seconds (the button is still pressed) the remote sends two zigbee messages from type "unknown command". I think zigbee2mqtt decodes them to message of type 'cmdTradfriArrowRelease' and 'cmdMoveToLevelWithOnOff'.
How is zigbee2mqtt able to decode them and wireshark not (I used the news version.)?
If I am back home, I am able to give you the exactly wireshark value of the command. Does this help? 
Have other user of zigbee2mqtt and this remote control problems or remaks on this topic (Press the On/Off button for a long time)?
I useed Domoticz with plugin and the plugin will be confused, if these zigbee2mqtt messages are received at this state/time.

Koenkk, 2019-07-08:
I understand your point know, they are shown as unknown because it are IKEA specific commands and therefore not part of the ZCL. We defined them here: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/zcl-id/definitions/cluster_defs.json#L80. 

Anyway, in my opinion, the naming of the commands doesn't really matter that much, eventually you care about the send MQTT message which I think has correct behaviour know.

Maik-100, 2019-07-14:
You change something in the code of zigbee2mqtt? So I have to use a new Version, which one?If I am back home, I can do the Test what Message MQTT ist now reseiving.

Maik-100, 2019-07-28:
Hi, I am back home, Irland is a great island with "normal" temperatures.
I made a update to the  current version:
info: Starting zigbee2mqtt version 1.5.1 (commit #ac3b924)
Coordinator firmware version: '20190425'
0x000d6ffffe18e00a (0x000d6ffffe18e00a): E1524 - IKEA TRADFRI remote control (EndDevice)

**If I press the ON/OFF button for a short time I get:**
2019-7-28 21:29:57 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":13,"battery":87,"action":"toggle"}'
**-> still ok**

**If I press the ON/OFF button for a long time I get 3 messages.**
The **first** one comes **immediately**:
2019-7-28 21:43:45 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":15,"battery":87,"action":"toggle"}'
The **second** and **third** comes **after approximately 3 seconds:**
2019-7-28 21:43:48 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"linkquality":13,"battery":87,"action":"toggle_hold"}'
2019-7-28 21:43:49 - info: Zigbee publish to device '0x000d6ffffe18e00a', genBasic - readRsp - [] - {"direction":1,"seqNum":33,"disDefaultRsp":1} - 1
**no other messages -> for me it looks ok**

On **mosquitto debug log**:
**If I press the ON/OFF button for a short time I get:**
1564344064: Sending PINGRESP to Domoticz_1564343924_1106
1564344069: Received PUBLISH from mqttjs_cde683ef (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6ffffe18e00a', ... (49 bytes))
1564344069: Sending PUBLISH to Domoticz_1564343924_1106 (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6ffffe18e00a', ... (49 bytes))
1564344074: Received PINGREQ from Domoticz_1564343924_1106

**If I press the ON/OFF button for a long time I get some messages:**
1564344244: Sending PINGRESP to Domoticz_1564343924_1106
1564344248: Received PUBLISH from mqttjs_cde683ef (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6ffffe18e00a', ... (48 bytes))
1564344248: Sending PUBLISH to Domoticz_1564343924_1106 (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6ffffe18e00a', ... (48 bytes))
1564344251: Received PUBLISH from mqttjs_cde683ef (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6ffffe18e00a', ... (53 bytes))
1564344251: Sending PUBLISH to Domoticz_1564343924_1106 (d0, q0, r0, m0, 'zigbee2mqtt/0x000d6ffffe18e00a', ... (53 bytes))
1564344254: Received PINGREQ from Domoticz_1564343924_1106
1564344254: Sending PINGRESP to Domoticz_1564343924_1106
1564344264: Received PINGREQ from Domoticz_1564343924_1106

Koenkk, 2019-07-29:
Thanks for investigating, however it's not clear to me what you want to accomplish/the problem is.