Koenkk/zigbee2mqtt #1498: Xiaomi Motion Sensor not reporting occupancy:true

zerofruchtshake, 2019-05-04:
# Bug Report
## What happened
One of my Xiaomi RTCGQ11LM Sensors does not report "occupancy":true most of the time
It updates the Illuminance/... but the occupancy is missing
`May 04 16:30:58 zigbee2mqttTest npm[13995]:   zigbee2mqtt:info 5/4/2019, 4:30:58 PM MQTT publish: topic 'zigbee2mqtt/xiaomimotionsz', payload '{"illuminance":17,"linkquality":65,"occupancy":false,"battery":100,"voltage":3015}'`

Working Sensor Log:
`May 04 20:05:53 zigbee2mqttTest npm[13995]:   zigbee2mqtt:info 5/4/2019, 8:05:53 PM MQTT publish: topic 'zigbee2mqtt/xiaomimotionkueche', payload '{"illuminance":0,"linkquality":55,"occupancy":false,"battery":100,"voltage":3005}'
May 04 20:05:53 zigbee2mqttTest npm[13995]:   zigbee2mqtt:info 5/4/2019, 8:05:53 PM MQTT publish: topic 'zigbee2mqtt/xiaomimotionkueche', payload '{"illuminance":0,"linkquality":55,"occupancy":true,"battery":100,"voltage":3005}'
`
## What did you expect to happen
See above
## How to reproduce it (minimal and precise)
No idea, other RTCGQ11LM sensors working as expected
## Debug Info
zigbee2mqtt version: 1.3.1 (commit #39d6a8e)
CC253X firmware version: Coordinator firmware version: '20190315'

zerofruchtshake, 2019-05-05:
And a second added sensor takes over 15 seconds to report occupancy despite changing illuminance immediately:
`May 05 10:09:24 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 10:09:24 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00029a59fb', payload '{"illuminance":71,"linkquality":63,"elapsed":1654849,"occupancy":false}'
May 05 10:09:29 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 10:09:29 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00029a59fb', payload '{"illuminance":160,"linkquality":63,"elapsed":5070,"occupancy":false}'
May 05 10:09:38 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 10:09:38 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00029a59fb', payload '{"illuminance":252,"linkquality":63,"elapsed":9338,"occupancy":false}'
May 05 10:09:39 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 10:09:39 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00029a59fb', payload '{"illuminance":252,"linkquality":60,"elapsed":35,"occupancy":true}'`

zerofruchtshake, 2019-05-05:
Second Sensor - Logs DEBUG level:
`May 05 10:21:22 zigbee2mqttTest npm[14745]:   zigbee2mqtt:debug 5/5/2019, 10:21:22 AM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":125}}' of device 'lumi.sensor_motion.aq2' (0x00158d00029a59fb) of endpoint 1

May 05 10:21:22 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 10:21:22 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00029a59fb', payload '{"illuminance":125,"linkquality":60,"elapsed":193735,"occupancy":false}'

May 05 10:21:22 zigbee2mqttTest npm[14745]:   zigbee2mqtt:debug 5/5/2019, 10:21:22 AM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":125}}' of device 'lumi.sensor_motion.aq2' (0x00158d00029a59fb) of endpoint 1

May 05 10:21:27 zigbee2mqttTest npm[14745]:   zigbee2mqtt:debug 5/5/2019, 10:21:27 AM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":733}}' of device 'lumi.sensor_motion.aq2' (0x00158d00029a59fb) of endpoint 1

May 05 10:21:27 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 10:21:27 AM MQTT publish: topic 'zigbee2mqtt/0x00158d00029a59fb', payload '{"illuminance":733,"linkquality":60,"elapsed":5193,"occupancy":false}'

May 05 10:21:27 zigbee2mqttTest npm[14745]:   zigbee2mqtt:debug 5/5/2019, 10:21:27 AM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":733}}' of device 'lumi.sensor_motion.aq2' (0x00158d00029a59fb) of endpoint 1`

Koenkk, 2019-05-05:
@zerofruchtshake could you do some zigbee sniffing (http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html) in order to verify that the sensor is indeed sending occupancy true? (to make sure this is not a malfunctioning device).

zerofruchtshake, 2019-05-05:
@Koenkk thanks for your reply! Unfortunately I have only macOS PCs and also only one cc2531. Is it possible to use a CC2530 for sniffing? If yes then I'll try to use a raspberry for sniffing.

The sensors were working fine for several weeks until i reinstalled zigbee2mqtt due to a memory card problem a few days ago

Koenkk, 2019-05-05:
@zerofruchtshake You could try to replace the battery and re-pair it.

zerofruchtshake, 2019-05-05:
New batteries should arrive the day after tomorrow. I'll try them.

I just discovered that zigbee2mqtt does publish the (correct) occupancy false when there is no more motion.. does this help?

May 05 18:49:20 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 6:49:20 PM MQTT publish: topic 'zigbee2mqtt/xiaomimotionsz', payload '{"illuminance":1,"linkquality":60,"occupancy":false,"battery":100,"voltage":3015,"elapsed":3041858}'

....

May 05 18:51:47 zigbee2mqttTest npm[14745]:   zigbee2mqtt:info 5/5/2019, 6:51:47 PM MQTT publish: topic 'zigbee2mqtt/xiaomimotionsz', payload '{"illuminance":1,"linkquality":57,"occupancy":false,"battery":100,"voltage":3015,"elapsed":146370}'

andreas-bell, 2019-05-09:
hi @Koenkk Koen,
first, thank you for the great software!
I have it running smoothly since one year with a CC2531 and around 25 Xiaomi Sensors.
Two of them are Xiaomi motion sensors (RTCGQ11LM) and since today I have a similar issue as @zerofruchtshake : 
The occupancy is always reported as "false"
{"illuminance":14,"linkquality":10,"occupancy":false,"battery":100,"voltage":3015}
I've gone into old logs and found that it was previously correctly reported:
 payload '{"illuminance":8,"linkquality":52,"occupancy":true,"battery":100,"voltage":3045}
This started after I upgraded today from Oct-2018 to version 1.3.1 and flashed the latest PROD firmware.
I've noticed as my rules for motion detection were no longer working after the upgrade.
As the rules were working fine based on the occupancy value =true I believe the device is still sending the correct zigbee message.
Do you have an idea how I could further troubleshoot ?
Cheers,
 Andy

Koenkk, 2019-05-10:
@andreas-bell could you also provide your debug log when receiving occupancy messages?

MarsWarrior, 2019-05-11:
I'm running version 1.3.1 with hass 0.91.3 and have no problems with the Xiaomi Motion sensors.

```
zigbee2mqtt:debug 5/11/2019, 4:40:50 PM Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 4:40:50 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":774}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 4:40:50 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":774}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 4:58:02 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":1000}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 4:58:02 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":1000}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 4:58:02 PM Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 5:03:23 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":1000}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1,
  zigbee2mqtt:debug 5/11/2019, 5:03:23 PM Received zigbee message of type 'attReport' with data '{"cid":"msOccupancySensing","data":{"occupancy":1}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b71fdf) of endpoint 1
```

closed 2019-07-17 by stale[bot]

e-t-h, 2019-11-04:
Hi,
is it possible to reopen this issue? Does anyone found a solution already?

I have the same issue: two RTCGQ11LM sensors, ran for a long time at an Aqara Hub, without any issues. Since I am working with this tool I've notice that they shows the same behavior zerofruchtshake have desribed: the occupancy value is not set properly. I am currently not able to get a message with "true".

My current zigbee2mqtt version is 1.6.0, Coordinator firmware version: '20190608'.  

Sensor 1 (friendly name Bew_unten) is hardware modified like discribed in the documentation to shorten the "on-time" to 5 secs. 
Sensor 2 (friendly name Bew_oben) is not modified.
Both have the same issue. 
I've collect some debug messages with the option DEBUG=zigbee-shepherd*-option, might be that helps?

Sensor 1, occupancy=false
> zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +8ms
 zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1024,"srcaddr":40372,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":31,"securityuse":0,"timestamp":8830101,"transseqnumber":0,"len":8,"data":{"0":24,"1":24,"2":10,"3":0,"4":0,"5":33,"6":44,"7":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":24,"cmdId":"report","payload":[{"attrId":0,"dataType":33,"attrData":44}]}} +9ms
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":44}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b42bd0) of endpoint 1
  zigbee2mqtt:info 11/4/2019, 12:27:34 PM MQTT publish: topic 'zigbee2mqtt/Bew_unten', payload '{"illuminance":44,"linkquality":31,"battery":100,"voltage":3035,"occupancy":false}'
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":44}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b42bd0) of endpoint 1

Sensor 2, occupancy=false
> zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg +114ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: {"groupid":0,"clusterid":1024,"srcaddr":33408,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":34,"securityuse":0,"timestamp":8829962,"transseqnumber":0,"len":8,"data":{"0":24,"1":47,"2":10,"3":0,"4":0,"5":33,"6":18,"7":0},"zclMsg":{"frameCntl":{"frameType":0,"manufSpec":0,"direction":1,"disDefaultRsp":1},"manufCode":0,"seqNum":47,"cmdId":"report","payload":[{"attrId":0,"dataType":33,"attrData":18}]}} +92ms
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'attReport' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":18}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b48ee2) of endpoint 1
  zigbee2mqtt:info 11/4/2019, 12:27:34 PM MQTT publish: topic 'zigbee2mqtt/Bew_oben', payload '{"illuminance":18,"linkquality":34,"occupancy":false,"battery":100,"voltage":3035}'
  zigbee2mqtt:debug 11/4/2019, 12:27:34 PM Received zigbee message of type 'devChange' with data '{"cid":"msIlluminanceMeasurement","data":{"measuredValue":18}}' of device 'lumi.sensor_motion.aq2' (0x00158d0002b48ee2) of endpoint 1
  zigbee-shepherd:msgHdlr IND <-- ZDO:srcRtgInd +31ms 

Any ideas? What can I additionaly provide? Does this debug output help to determine what the stick have received?

Sincerly..

Sepultura22, 2019-11-20:
> Hi,
> is it possible to reopen this issue? Does anyone found a solution already?
>
> I have the same issue: two RTCGQ11LM sensors, ran for a long time at an Aqara Hub, without any issues. Since I am [...]

I have the same problem