Koenkk/zigbee2mqtt #1499: CC2531 LED wont turn on after router flash

Brandon4466, 2019-05-05:
So I just flashed my CC2531 successfully with the router firmware and when I connect the v3.3 rail to the USB it starts up and the LED flashes like it is supposed to, but when I plug it into a wall charger it doesn't turn on at all which is weird.

Anybody have any idea what this could be?

Thanks.

mclaudiopt, 2019-05-05:
i use as usb stick on the pi and cannot disable also

Koenkk, 2019-05-05:
I think you can turn it on/off by pressing on of the buttons on it.

Brandon4466, 2019-05-09:
@Koenkk I pressed the button on it but it still wont turn on. It is not connecting to the main server either with the LED off, while other devices are working fine. I tried reflashing but same result, LED turns on when 3.3v is connected to the debug connectors but not when the USB is connected via USB port.

Thank you of helping, any ideas of how to fix this or what it could be?

Koenkk, 2019-05-10:
@ptvoinfo could you help?

ptvoinfo, 2019-05-11:
@Brandon4466 I think, it is a broken stick. It does not receive power from the USB connector. The USB port has a 5V voltage. Therefore, there is a DC converter from 5V to 3.3V.

Brandon4466, 2019-05-11:
@ptvoinfo Ah, Okay. I will get a new one. Thank you!

DCooper-nz, 2019-06-22:
I am having this same problem but I don't think its the stick. I have 3 and they are all having the same behaviour. They work fine when flashed with the coordinator firmware (at least the led lights up) but after flashing with the router firmware they are dead, the led doesn't work and they don't connect to the network.

ptvoinfo, 2019-06-22:
@DCooper-nz Are you using the standard CC2531 too? Where did you download the firmware? How did you flash the firmware?

closed 2019-08-28 by stale[bot]

CAPKLU, 2019-10-07:
Hi guy, I thought that I found the same problem. When I plug it into the USB socket of raspberry pi, the LED bright but turn off a few second later. Here are some information I have:
Flash tool : https://github.com/jmichault/flash_cc2531.git
firmware :  CC2531_DEFAULT_20190608.zip, CC2531ZNP-Prod.hex
after flash and plug into USB socket : 
```
pi@raspberrypi:~ $ dmesg
[  139.829484] usb 1-1.5: new full-speed USB device number 8 using dwc_otg
[  139.909483] usb 1-1.5: device descriptor read/64, error -32
[  140.099489] usb 1-1.5: device descriptor read/64, error -32
[  140.289488] usb 1-1.5: new full-speed USB device number 9 using dwc_otg
[  140.369484] usb 1-1.5: device descriptor read/64, error -32
[  140.559481] usb 1-1.5: device descriptor read/64, error -32
[  140.749495] usb 1-1.5: new full-speed USB device number 10 using dwc_otg
[  141.169494] usb 1-1.5: device not accepting address 10, error -32
[  141.249498] usb 1-1.5: new full-speed USB device number 11 using dwc_otg
[  141.669488] usb 1-1.5: device not accepting address 11, error -32
[  141.669560] usb 1-1-port5: unable to enumerate USB device
```
  
does not exist /dev/ttyACM0 
  
I once suspected that it was a hardware problem, but I can't verify it now.
I buy two of that CC2351 USB stick from taobao/aliexpress which claim "as same as TI official circuit". And I connected pi to debug pin via some MICRO-HOOK, I suspect that there may be a short circuit when in connect ( I found that I should do this when it is power on or I could not get ID ) and destroyed USB power ( but I can get the ID and flash successfully with flash_tool ).