Koenkk/zigbee2mqtt #1500: Adding support for sercomm SZ-ESW01-AU

brent0np, 2019-05-05:
Hi All,

Trying to add a Sercomm zigbee power plug (my first time adding a device) and am looking for some pointers.

I have the device connected however cannot figure out why I cannot see the state (or I am missing it).  Device Config and log of zigbee2mqtt here: https://pastebin.com/a8WMchaQ

If I use the switch on the device to turn the power on or off, there is nothing logged in zigbee2mqtt.  I have a xiaomi door sensor setup just to ensure everything is working properly.

I have tried adding the following fromzigbee converters into the device config but they have not helped at all: fz.generic_power, fz.state_change, fz.state_report, fz.ignore_diagnostic_change

Any help would be greatly appreciated.

Thanks,
Brenton.

Koenkk, 2019-05-05:
Can you try with:

```js
{
    zigbeeModel: ['SZ-ESW01-AU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SZ-ESW01-AU', // Vendor model number, look on the device for a model number
    vendor: 'Sercomm Corp.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Telstra Smart Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
},
```

brent0np, 2019-05-06:
Thanks, that's pretty much resolved everything except this one warning from the log:
`zigbee2mqtt:warn 5/6/2019, 9:59:11 PM No converter available for 'SZ-ESW01-AU' with cid 'genOnOff', type 'attReport' and data '{"cid":"genOnOff","data":{"onOff":0}}'`

Koenkk, 2019-05-06:
Can you try with:

```js
{
    zigbeeModel: ['SZ-ESW01-AU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SZ-ESW01-AU', // Vendor model number, look on the device for a model number
    vendor: 'Sercomm Corp.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Telstra Smart Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [fz.state], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
},
```

brent0np, 2019-05-07:
I still seem to be getting the same error.

5/7/2019, 8:30:47 PM - info: Successfully configured 0x000d6f000becca1d (0x000d6f000becca1d)
5/7/2019, 8:31:05 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000becca1d', payload '{"state":"OFF","linkquality":78}'
5/7/2019, 8:31:05 PM - warn: No converter available for 'SZ-ESW01-AU' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":0}}'
5/7/2019, 8:31:05 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
5/7/2019, 8:31:08 PM - info: MQTT publish: topic 'zigbee2mqtt/0x000d6f000becca1d', payload '{"state":"ON","linkquality":89}'
5/7/2019, 8:31:08 PM - warn: No converter available for 'SZ-ESW01-AU' with cid 'genOnOff', type 'devChange' and data '{"cid":"genOnOff","data":{"onOff":1}}'
5/7/2019, 8:31:08 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2019-05-09:
Please try with:

```js
{
    zigbeeModel: ['SZ-ESW01-AU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SZ-ESW01-AU', // Vendor model number, look on the device for a model number
    vendor: 'Sercomm Corp.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Telstra Smart Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [fz.state, fz.state_change], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
            ];

            execute(device, actions, callback);
        },
},
```

- You should be able to turn the switch on/off by publishing to `zigbee2mqtt/0x000d6f000becca1d/set` `ON`.
- For the energy monitoring, please share your `data/database.db`.

brent0np, 2019-05-10:
Thank you very much.  Publishing that allows me to switch on/off the device!

database.db is located here: https://pastebin.com/r6w14Pna

Koenkk, 2019-05-13:
Can you try if the metering also works with:

```js
{
    zigbeeModel: ['SZ-ESW01-AU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SZ-ESW01-AU', // Vendor model number, look on the device for a model number
    vendor: 'Sercomm Corp.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Telstra Smart Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [fz.state, fz.state_change, fz.generic_power, fz.ignore_metering_change], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                (cb) => device.report('seMetering', 'instantaneousDemand', 10, 60, 1, cb),
            ];

            execute(device, actions, callback);
        },
},
```

brent0np, 2019-05-15:
Thanks, that seems to have helped the power measurement in the log.  I have attached the current log, which seems to be throwing some new errors: https://pastebin.com/a7cXjNGN

Koenkk, 2019-05-16:
Can you post the log when running with `log_level: debug`? (http://www.zigbee2mqtt.io/configuration/configuration.html). Also, how much power does the device consume that you plugged into the SZ-ESW01-AU?

brent0np, 2019-05-17:
debug log located here: https://pastebin.com/PxCzeEhV

In the debug log I turned on the plug with nothing connected, then plugged in a samsung S10+ with the samsung adaptive fast charger.  It states its ouput is 9.0V 1.67A or 5.0V 2.0A

Koenkk, 2019-05-19:
Can you try with:

```js
{
    zigbeeModel: ['SZ-ESW01-AU'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SZ-ESW01-AU', // Vendor model number, look on the device for a model number
    vendor: 'Sercomm Corp.', // Vendor of the device (only used for documentation and startup logging)
    description: 'Telstra Smart Plug', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    supports: 'on/off', // Actions this device supports (only used for documentation)
    fromZigbee: [fz.state, fz.state_change, fz.SZ_ESW01_AU_power, fz.ignore_metering_change], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const cfg = {direction: 0, attrId: 0, dataType: 16, minRepIntval: 0, maxRepIntval: 1000, repChange: 0};
            const actions = [
                (cb) => device.bind('genOnOff', coordinator, cb),
                (cb) => device.foundation('genOnOff', 'configReport', [cfg], foundationCfg, cb),
                (cb) => device.report('seMetering', 'instantaneousDemand', 10, 60, 1, cb),
            ];

            execute(device, actions, callback);
        },
},
```

and add to `fromZigbee.js`

```js
    SZ_ESW01_AU_power: {
        cid: 'seMetering',
        type: ['attReport', 'readRsp'],
        convert: (model, msg, publish, options) => {
            if (msg.data.data.hasOwnProperty('instantaneousDemand')) {
                return {power: precisionRound(msg.data.data['instantaneousDemand'] / 1000, 2)}
            }
        }
    },
```

brent0np, 2019-05-20:
Thanks, that seems to be working.  Is there anything else I should be testing?

Koenkk, 2019-05-20:
Nope, device is supported now in the dev branch, thanks!

closed 2019-05-20 by Koenkk

matthewparkes, 2022-01-17:
Are you able to add this for "SZ-ESW02" It is the second iteration of this device.  I'm having trouble getting the power usage reporting to work, and I'm not sure how to modify "fromZigbee.js"