Koenkk/zigbee2mqtt #1502: OSRAM/Sylvania RGBW BR30 Not supported - will not pair?

vexter0944, 2019-05-05:
# Bug Report
## What happened
When trying to join an Osram/Sylvania 73739 RGBW BR30 - it appears to pair but docker container logs show 'zigbee2mqtt:warn 5/5/2019, 4:36:05 PM Device with modelID 'BR30 RGBW' is not supported.'
## What did you expect to happen
The light should pair and be recognized based on the documentation.

## How to reproduce it (minimal and precise)
pair model number73739 bulb to zigbee2mqtt and watch logs for above error

## Debug Info
zigbee2mqtt version:
1.3.1
CC253X firmware version:
20190223

vexter0944, 2019-05-05:
ok - just posted and changed this line in devices.js to 
        zigbeeModel: ['BR30 RGBW'], and it worked. for the 73739 model in the file.

Koenkk, 2019-05-05:
Will be supported in next zigbee2mqtt version, thanks!

closed 2019-05-05 by Koenkk

vexter0944, 2019-05-05:
ahh - great news - how long before that release?

Koenkk, 2019-05-05:
It should be in the dev branch somewhere in the end of next week.