Koenkk/zigbee2mqtt #1503: Adding support for Home Decorators WInk enabled ceiling fan (Hampton Bay controller)

YamiFrankc, 2019-05-06:
Hello,
I would like help to be able to properly control [this](https://www.homedepot.com/p/Home-Decorators-Collection-Gardinier-52-in-LED-Indoor-Brushed-Nickel-WINK-Enabled-Smart-Ceiling-Fan-with-Integrated-Light-Kit-with-Remote-Control-43260/206648825) ceiling fan.

I think the code would be almost the same as with the Hampton Bay Wink enabled fan controller because this fan actually uses it. I think the only difference is that the remote included with the fan has a button dedicated to toggling the fan direction. The stand alone fan controller doesnt have it.

So after pairing it HA properly autodiscovered it as a light and a fan, and it displays the actual current state of the device. But it cant really control it reliably. 

- Trying to set the fan speed from HA does absolutely nothing, Turning off works, and will always and only turn on on the highest speed.
- Turning off the light works sometimes. It doesnt work when I try to turn it on. Brightness has a delay of  like 30 seconds when it works.
- Lights turns off randomly sometimes
- Clicking  the "reverse direction" button on the remote doesnt produce any messages in the log.
 

[log](https://pastebin.com/t51VG56s)
[database.db](https://pastebin.com/VGM9Hr2m)

Koenkk, 2019-05-06:
Let's start with the fan speed. 

Can you publish to `hachi/0x0022a3000020ce19/set/fan_mode` payload `LOW`, `HIGH`. etc, please post the log if it doesn't change the actual speed.

YamiFrankc, 2019-05-06:
They seem to work.  I tried with all the options on the auto discovered fan in HA. Auto and SMART changed to MEDIUM speed. The controller is advertised has having 4 speeds, but currently cant get the faster one from HA.

[LOG](https://pastebin.com/eE0EHD8N)

Koenkk, 2019-05-06:
So that means turning the fan speed (point 1) works now?

YamiFrankc, 2019-05-07:
Seems like its working. Its kind of weird, but the drop down menu for the speed in HA doesnt always seem to work.  Sometimes it sends a "AF data request fails, status code 233. MAC no ack" error.
Also, fan_mode on is the higher speed. 

Last night something weird happened, I was adding back some other devices(unsuccessfully, for some reason they acted like they paired but didnt) and the fan left the network. It even appeared offline in the network map.

Unfortunate I made a mistake and deleted the entire Z2M folder before saving the logs.

[LOG](https://pastebin.com/ih3v6M4N) from today, after adding back the fan and testing setting the speed.
I noticed something weird, too. When I look in the log.txt files the times are all wrong (they show 7 am when it is 1 am for me), but my system time is set correctly and more importantly, it shows correctly when im monitoring the log live.
[IMAGE](https://imgur.com/a/DDdItsQ), the end of the pastebin log is the same as the one in this image.

Koenkk, 2019-05-07:
The 233 mac no ack happen when the device doesn't respond. Does the fan mode change when this occurs?

Also, can you post the log when changing the brightness?

YamiFrankc, 2019-05-08:
No the fan mode doesnt change.
Right now its not accepting any command from HA, they all show as "MAC no ack", but the changes I make from the remote where showing in the log until just now (From repairing the fan to having it disconnect, it took maybe about 15 min
So it keeps disconnecting at random, sometimes it will stay connected for hours and sometimes only for minutes, Nothing appears in the log when it disconnects. I havent been able to try the brightness again because all im getting is "MAC no ack"

[log-when disconnects.txt](https://github.com/Koenkk/zigbee2mqtt/files/3155917/log-when.disconnects.txt)
utes.)
[newlog.txt](https://github.com/Koenkk/zigbee2mqtt/files/3155919/newlog.txt)

Koenkk, 2019-05-09:
Can you try with the latest dev firmware? https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin

YamiFrankc, 2019-05-11:
I installed the "CC2531_20190425.zip" one, is that the one that you meant? The other oines do not seem to work with my stick
I finally got to pair it again, it has a really annoying pairing process because unlike other bulbs I have tried it wont instantly blink when pairing/paired. 

Right now its not accepting any command from HA(MAC no ack), but it shows changes in the log when I use the remote to change state,
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3168744/log.txt)
[database.db.txt](https://github.com/Koenkk/zigbee2mqtt/files/3168745/database.db.txt)

YamiFrankc, 2019-05-12:
[image]
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3169779/log.txt)
Been trying to keep the fan paired, but it always get disconnected. The first few times it was paired it lasted for hours but its only minutes now.
The weird thing is that it still shows as online in the map

YamiFrankc, 2019-05-12:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3169865/log.txt)
So I have found out that it will "connect again" after some time, it will accept state changes from HA, and the ones done with the remote show up in the log again.
Changes made from HA take seconds(maybe up to a minute) to take effect(show in the log and in the light itself); whereas changes made from the remote immediately reflect in HA.
This is the log of me trying to change the brightness, it works ok other than the long delay of them taking effect. 
And as randomly has it was working again, it stopped working again.

Koenkk, 2019-05-14:
This problems seems similar to #1055 (see https://github.com/Koenkk/zigbee2mqtt/issues/1055#issuecomment-492296006)

YamiFrankc, 2019-05-14:
Yes, seems similar.
So I found out a couple of things, the first is that it is listed as an end device even if it mains powered.
Second is that before now I only had this controller and other end device sensors paired. Last night I paired a router bulb and 
now the fan is working much more reliably now; it still has a delay sometimes.

YamiFrankc, 2019-05-19:
Okay, it has become much more stabler since I added the other routers back to the network.
It just needs to take a couple seconds to make a second state change right after another. Example:
Light has been off for a few hours, and I accidentally turn it on. It will take effect immediately.
Because it was an accident I proceed to turn it off. This time it will take a couple seconds(up to around 30s) to take effect.
This is true for light ON/OFF, light brightness and fan state.
Other than that it seems to work fine.

closed 2019-07-25 by stale[bot]