Koenkk/zigbee2mqtt #1504: Cannot get the Node Descriptor of the Device: {myDeviceId} (Error: Timed out after 10000 ms)

wengweng, 2019-05-06:
# Bug Report
## What happened
I have 3 Xiaomi devices:
- WXKG11LM (switch)
- WSDCGQ11LM (temp sensor)
- SJCGQ11LM (flood sensor)

I have paired these devices successfully. Then I decided to change the zigbee2mqtt network key in the config. The paired Xiaomi devices stopped working so I removed them using the 'zigbee2mqtt/bridge/config/remove' mqtt topic. Now I'm trying to re-pair but I'm getting the same error for all 3 devices: "Cannot get the Node Descriptor of the Device: {myDeviceId} (Error: Timed out after 10000 ms)"

## What did you expect to happen
I expect the devices to pair like they initially did.

## How to reproduce it (minimal and precise)
1. pair Xiaomi device
2. change network key and restart zigbee2mqtt
3. remove device using mqtt topic 'zigbee2mqtt/bridge/config/remove' (I even deleted 'database.db' at one point)
4. follow pairing guide

## Debug Info
zigbee2mqtt version: 1.3.1
CC253X firmware version: 20190223

Koenkk, 2019-05-06:
- Make sure to turn off all AC powered device which have not been re-paired after changing the network key
- Restart zigbee2mqtt
- Re-insert the battery of the Xiaomi devices
- Re-pair.

wengweng, 2019-05-06:
Thanks for your reply. I just followed your instructions but it doesn't work. The detailed steps:
1. Removed all the batteries and cut off AC power of each zigbee device
2. Stop zigbee2mqtt container (I'm running docker with data dir mounted)
3. Deleted `database.db` and `state.json`
4. Unplugged CC2531 stick and plugged it again 
5. Start zigbee2mqtt container
6. Enable `permit_join` using mqtt config topic
6. Insert battery to WSDCGQ11LM (temp sensor)
7. Re-pair

```  zigbee2mqtt:info 5/6/2019, 7:10:28 PM Logging to directory: '/app/data/log/2019-05-06.19-10-28'
  zigbee2mqtt:info 5/6/2019, 7:10:29 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 5/6/2019, 7:10:29 PM Starting zigbee-shepherd
  zigbee2mqtt:info 5/6/2019, 7:10:30 PM zigbee-shepherd started
  zigbee2mqtt:info 5/6/2019, 7:10:30 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:info 5/6/2019, 7:10:30 PM Currently 0 devices are joined:
  zigbee2mqtt:info 5/6/2019, 7:10:30 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 5/6/2019, 7:10:30 PM Connecting to MQTT server at mqtt://localhost
  zigbee2mqtt:info 5/6/2019, 7:10:31 PM zigbee-shepherd ready
  zigbee2mqtt:info 5/6/2019, 7:10:31 PM Connected to MQTT server
  zigbee2mqtt:info 5/6/2019, 7:10:31 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 5/6/2019, 7:10:31 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"unknown","coordinator_firmware":20190223,"log_level":"info","permit_join":false}'
  zigbee2mqtt:info 5/6/2019, 7:12:10 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 5/6/2019, 7:12:11 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"unknown","coordinator_firmware":20190223,"log_level":"info","permit_join":true}'
  zigbee2mqtt:error 5/6/2019, 7:13:41 PM Cannot get the Node Descriptor of the Device: 0x00158d0002d775ab (Error: Timed out after 10000 ms)```

Koenkk, 2019-05-07:
Those steps look good, could you repeat that steps but with step 4 also re-flash the firmware on the CC2531?

wengweng, 2019-05-07:
It works now! Re-flashing the firmware on the CC2531 seems to have done the job. Any idea why?

Koenkk, 2019-05-07:
Probably because the device was already in the device table on the CC2531.

wengweng, 2019-05-07:
Thanks for your help!

closed 2019-05-07 by wengweng

abmantis, 2019-08-31:
I also had this issue. The device stopped working and couldn't get it back. After pressing the button on the stick, unpluging and repluging it, I was able to finally repair it. Didn't need to flash it.

Isn't a way to fix this from Z2M side?

Lorenz84, 2019-09-06:
Which button of 2 did you press?

abmantis, 2019-09-06:
> Which button of 2 did you press?

Both :)

Lorenz84, 2019-09-06:
I have a Xiaomi door sensor that still gives me the error After pressing both buttons, un-plugging and re-plugging the stick. Do you suggest me te simply stop Zigbee2Mqtt addon, re-flash it the cc2531, and re-plug it?

alfem, 2019-10-09:
> I have a Xiaomi door sensor that still gives me the error After pressing both buttons, un-plugging and re-plugging the stick. Do you suggest me te simply stop Zigbee2Mqtt addon, re-flash it the cc2531, and re-plug it?

I also have got one new Xiaomi door sensor in the same situation. I already paired another two door sensors in July but this one does not pair.