Koenkk/zigbee2mqtt #1505: Battery info saying 100%, even after upgrade

fabricekrebs, 2019-05-06:
Hello,
I've a couple of same temperature sensor. Some of them are reporting payload differently, and even the battery information doesn't look normal, as staying always 100%.

Below is the configuration file
`
homeassistant: false
permit_join: true
mqtt:
  base_topic: xxxxxxx
  server: 'mqtt://x.x.x.x'
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d0001f403bf':
    friendly_name: Press Button 01
    retain: false
  '0x00158d0001f24c49':
    friendly_name: Living Room Power Switch 01
    retain: false
  '0x00158d0002366875':
    friendly_name: Dining Room Power Switch 01
    retain: false
  '0x00158d00022aeed7':
    friendly_name: Office Temperature 01
    retain: false
  '0x00158d0002227bef':
    friendly_name: Bathroom Temperature 01
    retain: false
  '0x00158d0002029233':
    friendly_name: Sleeping Room Temperature 01
    retain: false
  '0x00158d0002108257':
    friendly_name: Veranda Temperature 01
    retain: false
  '0x00158d0002c76dcc':
    friendly_name: Living Room Temperature 01
    retain: false
  '0x00158d00020d7d0d':
    friendly_name: Green House Temperature 01
    retain: false
  '0x00158d000222fbd9':
    friendly_name: Motion Sensor 01
    retain: false
  '0x00158d00025eaf7b':
    friendly_name: Garage Temperature 01
    retain: false
  '0x00158d0002bfd92c':
    friendly_name: Entrance Temperature 01
    retain: false
  '0x00158d0002b781b1':
    friendly_name: Vibration Sensor 01
    retain: false
advanced:
  log_level: debug

`

Below are the logs

`
cat /opt/zigbee2mqtt/data/log/2019-05-06.16-00-38/log.txt
2019-5-6 16:00:38 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-06.16-00-38'
2019-5-6 16:00:40 - debug: Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
2019-5-6 16:00:40 - debug: Loaded state from file /opt/zigbee2mqtt/data/state.json
2019-5-6 16:00:40 - debug: Saving state to file /opt/zigbee2mqtt/data/state.json
2019-5-6 16:00:40 - info: Starting zigbee2mqtt version 1.3.1 (commit #39d6a8e)
2019-5-6 16:00:40 - info: Starting zigbee-shepherd
2019-5-6 16:00:42 - info: zigbee-shepherd started
2019-5-6 16:00:42 - info: Coordinator firmware version: '20180815'
2019-5-6 16:00:42 - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":11,"panId":"0x1a62","extPanId":"0xdddddddddddddddd","ieeeAddr":"0x00124b0018ed1371","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20180815},"startTime":1557158442,"joinTimeLeft":0}
2019-5-6 16:00:42 - info: Currently 14 devices are joined:
2019-5-6 16:00:42 - info: Press Button 01 (0x00158d0001f403bf): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
2019-5-6 16:00:42 - info: Living Room Power Switch 01 (0x00158d0001f24c49): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
2019-5-6 16:00:42 - info: Dining Room Power Switch 01 (0x00158d0002366875): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)
2019-5-6 16:00:42 - info: Office Temperature 01 (0x00158d00022aeed7): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Bathroom Temperature 01 (0x00158d0002227bef): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Sleeping Room Temperature 01 (0x00158d0002029233): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: unknown (0x00158d00020d7fed): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Veranda Temperature 01 (0x00158d0002108257): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Green House Temperature 01 (0x00158d00020d7d0d): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Motion Sensor 01 (0x00158d000222fbd9): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
2019-5-6 16:00:42 - info: Living Room Temperature 01 (0x00158d0002c76dcc): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Garage Temperature 01 (0x00158d00025eaf7b): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Entrance Temperature 01 (0x00158d0002bfd92c): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2019-5-6 16:00:42 - info: Vibration Sensor 01 (0x00158d0002b781b1): DJT11LM - Xiaomi Aqara vibration sensor (EndDevice)
2019-5-6 16:00:42 - warn: `permit_join` set to  `true` in configuration.yaml.
2019-5-6 16:00:42 - warn: Allowing new devices to join.
2019-5-6 16:00:42 - warn: Set `permit_join` to `false` once you joined all devices.
2019-5-6 16:00:42 - info: Zigbee: allowing new devices to join.
2019-5-6 16:00:42 - info: Connecting to MQTT server at mqtt://x.x.x.x
2019-5-6 16:00:42 - info: zigbee-shepherd ready
2019-5-6 16:00:42 - info: Connected to MQTT server
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Press Button 01', payload '{"battery":100,"voltage":3002,"linkquality":76}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Living Room Power Switch 01', payload '{"state":"OFF","power":0,"voltage":null,"consumption":15.35,"temperature":30,"linkquality":134}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Dining Room Power Switch 01', payload '{"state":"OFF","power":0,"voltage":null,"consumption":3.63,"temperature":33,"linkquality":39}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Office Temperature 01', payload '{"temperature":20.64,"linkquality":134,"humidity":47.74,"battery":37,"voltage":2975}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Bathroom Temperature 01', payload '{"battery":100,"voltage":3005,"linkquality":134,"temperature":20.08,"humidity":36.68}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Sleeping Room Temperature 01', payload '{"battery":100,"voltage":3005,"linkquality":42,"temperature":16.66,"humidity":43.03}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00020d7fed', payload '{"temperature":21.49,"linkquality":63,"humidity":42.29,"battery":"99.00","voltage":3005}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Veranda Temperature 01', payload '{"temperature":31.49,"linkquality":39,"humidity":22.4,"battery":100,"voltage":3015}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Green House Temperature 01', payload '{"temperature":15.37,"linkquality":147,"humidity":97.43,"battery":99,"voltage":3005}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Motion Sensor 01', payload '{"occupancy":false,"linkquality":86,"battery":100,"voltage":3005}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Living Room Temperature 01', payload '{"temperature":21.62,"linkquality":134,"humidity":30.64,"battery":100,"voltage":3005}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Garage Temperature 01', payload '{"temperature":14.29,"linkquality":134,"humidity":59.06,"battery":86,"voltage":2975}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Entrance Temperature 01', payload '{"temperature":9.32,"linkquality":76,"humidity":40.82,"battery":57,"voltage":2925}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/Vibration Sensor 01', payload '{"angle_x":2,"angle_y":-1,"angle_z":88,"angle_x_absolute":88,"angle_y_absolute":91,"linkquality":63,"unknown_data":1638400,"battery":100,"voltage":3005}'
2019-5-6 16:00:42 - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.3.1","commit":"39d6a8e","coordinator_firmware":20180815,"log_level":"debug","permit_join":true}'
2019-5-6 16:00:43 - debug: Mounted the cieApp (epId 11)

`

Koenkk, 2019-05-07:
What do you mean with *Some of them are reporting payload differently*? Could you explain a bit more what the issue is here?

fabricekrebs, 2019-05-07:
Based on the provided logs you can see  the 2 following same sensors are providing a different payload format, and I'm wondering if it's normal? :
- Sleeping Room Temperature 01 -> {"battery":100,"voltage":3005,"linkquality":42,"temperature":16.66,"humidity":43.03}
- Veranda Temperature 01 -> {"temperature":31.49,"linkquality":39,"humidity":22.4,"battery":100,"voltage":3015}

But the main issue is that the battery is still showing 100%, but shouldn't as all battery are now more than 1 year old.

Koenkk, 2019-05-07:
- The payload (a JSON object) is an unordered collection, this is normal.
- The Xiaomi sensors contain a 3 volt battery, your sensors are reporting 3.015 and 3.005 volt meaning that they still seem to be full. Here you can find the used logic to determine the percentage: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L44

krfa1, 2019-05-07:
Ok, so I guess it's not a bug. Thank you for the informations

fabricekrebs, 2019-05-07:
Ok, so I guess it's not a bug. Thank you for the information

closed 2019-05-07 by Koenkk