Koenkk/zigbee2mqtt #1506: wiring diagram for flashing cc2530 cc2591

MubiTec, 2019-05-07:
Hi,

I tried to flash the cc2530 cc2591 and took a look at but had no success: http://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html
Does anyone have a concrete wiring diagram for flashing cc2530 cc2591?

Thanks a lot!

ssirkakriss, 2019-05-07:
There is a buried link within the Wiki that might be of help. its to another cc2530 info page (http://ptvo.info/how-to-select-and-flash-cc2530-144/)

Hope it helps.

MubiTec, 2019-05-07:
After a lot of trials it worked. I did it with this HowTo:
http://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html
I put CC Debugger Pin 1 against GND of an USB-TTL. I don't know, if this was the point but it worked :-)

closed 2019-05-07 by MubiTec