Koenkk/zigbee2mqtt #1507: Hue GO impossible to reset (and so to pair)

septeven, 2019-05-08:
Hi,

I have recently sold my Hue bridge (because I’m moving out of the country I’m currently living in) and have decided to use my zigbee2mqtt system to control my hue bulbs for the weeks before I move. I have perfectly paired all of them but the Hue Go which won’t want to reset at all.

I’m using the same method than for all others, which is holding ON and OFF button of a dimmer switch closeby the lamp, hue go power coor plugged in, but the lamp stays on and won’t flash.

Is there any of you who has experienced that?
Any help will be much appreciated.

Thanks

Koenkk, 2019-05-08:
The pairing procedure for the Hue go is a bit more complicated (I also own one). Instructions can be found here http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#philips-hue (third bullet).

septeven, 2019-05-14:
Thanks @Koenkk - draining the battery worked :)

closed 2019-05-14 by septeven

hay11mac, 2021-01-04:
Wondering if anyone can help, I sold my Philips hue go lamp and the buyer is unable to connect it to their hub even searching via serial number they’re still unable to find it. Are they able to reset it? Does it need to come back to me to reset? Before selling I removed it from my Philips hue I thought that would be enough, but obviously not.

I see the comment about draining the battery, so is that just turning the lamp on with no power lead until the light goes off?

would really appreciate the help with this

luchtgitaar, 2023-11-25:
If anyone comes here via Google. Holding the button on the device for about 1 minute also resets the device.

davidcartwright, 2024-08-06:
Except when it doesn't. Which is what brought me here. My second day with a used Hue go, it  came on fine this morning then suddenly flicked off. I have no idea if it's just suddenly otherwise dead or not.