Koenkk/zigbee2mqtt #1508: сс2531 coordinator does not enter the pairing mode

directman66, 2019-05-08:
The coordinator does not enter the pairing mode, although he notifies that the permit  join is enabled. Coordinator after some time stops receiving messages about pairing. There are several cc2531 sticks with stable firmware. One of the coordinators was connected with 5 devices. After some time, one of the devices stopped transmitting readings. An exception was made to the device using the mqtt request. The device from the database and the config is gone. But no means of pairing occurred. It was a 5-button wireless remote from Ikea. I tried to pair with other devices, but without success. By various manipulations I tried to make this process. Reboot, remove the device from the database and so on. within 2 days had no effect. I just replaced the stick with another one with the same firmware. Previously, paired devices were added automatically. New devices I could add. I believe that in the memory of the stick there is some kind of counter that can overflow when unsuccessful attempts to pair (for example, xiaomi devices that need to periodically press a button).

directman66, 2019-05-08:
my config files:
[123.zip](https://github.com/Koenkk/zigbee2mqtt/files/3155695/123.zip)

Koenkk, 2019-05-09:
Can you try if this still occurs with the latest dev branch?

directman66, 2019-05-12:
Problem сс2531 still lies on the shelf. I can update the dev branch and try to pair new devices. According to the result I will write here.

directman66, 2019-05-12:
git pull
remote: Enumerating objects: 17, done.
remote: Counting objects: 100% (17/17), done.
remote: Total 21 (delta 17), reused 17 (delta 17), pack-reused 4
Unpacking objects: 100% (21/21), done.
From https://github.com/Koenkk/zigbee2mqtt
   80e6826..ee95841  dev        -> origin/dev
Already up to date.

directman66, 2019-05-12:
I stuck the old problematic stick, the connection with the IKEA remote control did not restore itself. Removed through the publication of the MCT device from the config and base. I try to match again, the result is negative.

Koenkk, 2019-05-13:
So it works with stick A, but doesn't work with stick B? Could you explain the problem a bit more?

Koenkk, 2019-07-12:
Closing due to inactivity

closed 2019-07-12 by Koenkk