Koenkk/zigbee2mqtt #1512: Unknown zigbee publish cmdType undefined

davegreebe, 2019-05-09:
# Bug Report
## What happened
When I have my ikea-bulbs connect with MQTT, I want to send a command by HomeAssistant (has auto discovered enabled) to the lights but nothing is happening and I get errors in my zigbee2mqtt-docker container
## What did you expect to happen
Change brightness
## How to reproduce it (minimal and precise)
Have ikea-bulbs (those should be supported)
Enable MQTT Autodiscover in HomeAssistant
Make LIGHT-chard in Lovelace
Change brightness
## Debug Info
zigbee2mqtt version:
1.2.1
CC253X firmware version:
latest! (updated firmware last friday)

Here are my logs:
https://pastebin.com/9wVW207d

way2many, 2019-05-09:
Hi,
I couldn't load pastebin as it seems down for me so I couldn't check the logs but have you tried restarting Home Assistant?
I think that you have the same setup as me: Home Assistant and Zigbee2mqtt in their respective Docker containers. Whenever I have a new device autodiscovered by Home Assistant I need to restart 50% of the time for the new device to work.

davegreebe, 2019-05-09:
Hi, I did restart my HA couple of times, and indeed I have that configuration.
All docker;
- Zigbee2mqtt -> Mosquito -> Homeassistant.

I also have a button in HA to enable / disable permist_join, just to check if my HA is communicating with Mosquitto and with Zigbee2MQTT and that is but that's all.
Hope you can check the logs later this day. Otherwise I'm also online on discord (duck304)

Koenkk, 2019-05-10:
- Update to the latest zigbee2mqtt version first 1.3.1
- Make sure you don't have a zigbee-shepherd-converters docker volume mapped.

davegreebe, 2019-05-11:
Yes, this is working. I did use the lastest-version tag on my container but maybe something went wrong because its working now. 👍 
Can I use the zigbee-shepherd-converters or won't I needing them?

Koenkk, 2019-05-13:
This is only needed when you want to add support for new devices.

closed 2019-05-13 by Koenkk