Koenkk/zigbee2mqtt #1515: Xiaomi Wall switch group message support question

pkedvessy, 2019-05-09:
Hi All,

I'm really interested in the [zigbee groups](https://www.zigbee2mqtt.io/information/groups.html) feature to control multiple Xiaomi Wall Switches by a single MQTT command.

I can see that how we can send messages to a group, but sending control messages to these switches is a bit different because they are using separated topics for left and right relays.

Is it possible to apply the group functionality for these switches? If so, how we can set up them?

Thanks,
Peter

Koenkk, 2019-05-10:
First we have to check if this device support groups. Could you try to add it to a group the way it is now? Then either the left or right button should be in a group.

pkedvessy, 2019-05-28:
Sorry for the late response.
Finally I was able to have a try with a "Xiaomi Aqara Neutral Line Type Remote Control Switch": it got added to a really new group (lets say 'test'), but when I have tried to send '{"state": "ON"}' message to '/bridge/group/test/set' I got the following error: '{"type":"entity_not_found","message":"bridge/group/test"}'

Koenkk, 2019-05-29:
You should send the message to `zigbee2mqtt/test/set` `{"state": "ON"}`

pkedvessy, 2019-05-29:
Ahh, thanks, it is working now: left is in the group. Can we add right as well or can we configure somehow which one is in the group?

Koenkk, 2019-05-29:
This was not implemented yet but can now be done in the dev branch. (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Documentation: https://github.com/Koenkk/zigbee2mqtt.io/blob/dev/information/groups.md#adding-a-specific-endpoint

pkedvessy, 2019-05-29:
I'm experiencing the following kind of issue with DEV branch:

Error: EACCES: permission denied, mkdir '/opt/zigbee2mqtt/node_modules/cc-znp/node_modules/@serialport/bindings/build'

Should I update my coordinator's firmware? I have no issue with the master branch...

pkedvessy, 2019-05-29:
I missed to say thank you for the quick turnaround on having support on my request! It's going to be more than brilliant to controll both channels by group messages!

pkedvessy, 2019-06-01:
I have updated my coordinator to Z-Stack_Home_1.2 version 20190425 and dev branch is working as expected. But when I'm trying to add a specific endpoint to a group, I get this error:

`Jun 01 14:08:22 peetserver npm[17732]:   zigbee2mqtt:debug 6/1/2019, 2:08:22 PM Received MQTT message on 'zigbee2mqtt/ground_floor/bridge/group/wall_switches/add' with data 'kitchen/wall_switch/2/right'
Jun 01 14:08:22 peetserver npm[17732]:   zigbee2mqtt:error 6/1/2019, 2:08:22 PM Failed to find device with entity ID 'kitchen/wall_switch/2'
Jun 01 14:08:22 peetserver npm[17732]: /opt/zigbee2mqtt/lib/extension/groups.js:150
Jun 01 14:08:22 peetserver npm[17732]:         const {entityID, endpointID} = this.parseID(ID);
Jun 01 14:08:22 peetserver npm[17732]:                ^
Jun 01 14:08:22 peetserver npm[17732]: TypeError: Cannot destructure property `entityID` of 'undefined' or 'null'.
Jun 01 14:08:22 peetserver npm[17732]:     at Groups.updateDeviceGroup (/opt/zigbee2mqtt/lib/extension/groups.js:150:45)
Jun 01 14:08:22 peetserver npm[17732]:     at Groups.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/groups.js:221:14)
Jun 01 14:08:22 peetserver npm[17732]:     at results.extensions.filter.map (/opt/zigbee2mqtt/lib/controller.js:152:27)
Jun 01 14:08:22 peetserver npm[17732]:     at Array.map (<anonymous>)
Jun 01 14:08:22 peetserver npm[17732]:     at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:152:14)
Jun 01 14:08:22 peetserver npm[17732]:     at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:81:18)
Jun 01 14:08:22 peetserver npm[17732]:     at MqttClient.emit (events.js:189:13)
Jun 01 14:08:22 peetserver npm[17732]:     at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1162:12)
Jun 01 14:08:22 peetserver npm[17732]:     at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:351:12)
Jun 01 14:08:22 peetserver npm[17732]:     at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:283:12)
Jun 01 14:08:23 peetserver npm[17732]: npm ERR! code ELIFECYCLE
Jun 01 14:08:23 peetserver npm[17732]: npm ERR! errno 1
Jun 01 14:08:23 peetserver npm[17732]: npm ERR! zigbee2mqtt@1.4.0 start: `node index.js`
Jun 01 14:08:23 peetserver npm[17732]: npm ERR! Exit status 1
`
and Z2M is terminating.

Koenkk, 2019-06-03:
Did you configure a `base_topic` in `configuration.yaml`?

pkedvessy, 2019-06-03:
Yes I did: zigbee2mqtt/ground_floor

Koenkk, 2019-06-03:
Can you post your `configuration.yaml`?

closed 2019-08-09 by stale[bot]