Koenkk/zigbee2mqtt #1516: All bulbs of one type suddenly offline

hgelpke, 2019-05-10:
Version 1.3.1
CC2531 with max stability firmware

I have a bunch of these:

53170161
Commercial Electric Matte White Recessed Retrofit Smart Led Downlight - 4 Inch (on/off, brightness, color temperature)

They worked in the past but recently I made 2 changes that somehow impacted them and I don't know how/why.

Using a docker image I pulled the most recent zigbee-shepherd-converters to a local mount that which I was previously not doing.  

I added a second repeater which shows a strong signal so I don't think that has mattered. 

Basically the lights all show offline while all the rest of my system still works. It's very specific to this one type of light so it's a common problem. I just dont know what it is.

hgelpke, 2019-05-10:
Here is my network map

[https://i.imgur.com/3sQMThm.png](url)

Koenkk, 2019-05-13:
Can you still control them? It is known that the online/offline state is not very reliable.

hgelpke, 2019-05-13:
I can't. I'm going to try and reflash because I was on max stability and
I'm going to go back to stock and see if that helps.

closed 2019-07-19 by stale[bot]