Koenkk/zigbee2mqtt #1517: Downgrade

GerSant, 2019-05-10:
I installed the 1.3.1 version when was available, for that point several issues begins...how can downgrade to 1.2.

Thanks!

brent0np, 2019-05-14:
If you are using hassio, hopefully you have snapshots so you can restore to a point before the upgrade

GerSant, 2019-05-16:
No, i don't use hassio, i solved selecting the release 1.2.1 instead 1.3.1 when make the "git clone" command. I will try the new 1.4 soon.

Regards

closed 2019-05-16 by GerSant

monotonus, 2020-05-29:
> 
> 
> No, i don't use hassio, i solved selecting the release 1.2.1 instead 1.3.1 when make the "git clone" command. I will try the new 1.4 soon.
> 
> Regards

How did you do this in detail?

GerSant, 2020-05-29:
`git clone [remote_address_here] my_repo
cd my_repo
git reset --hard [ENTER HERE THE COMMIT HASH YOU WANT]`

monotonus, 2020-05-29:
Thanks for your quick replay. It was very helpful. The single line command did'nt work because of "Too many arguments." So I splitted it in three lines and this is it for version 1.12.2.

sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt
cp -R data data-backup
git checkout HEAD -- npm-shrinkwrap.json
git clone https://github.com/Koenkk/zigbee2mqtt
my_repo cd my_repo
git reset --hard 7d27a54
npm ci
cp -R data-backup/* data
rm -rf data-backup
sudo systemctl start zigbee2mqtt

This was the result:

root@homeassistant:/opt/zigbee2mqtt# git clone https://github.com/Koenkk/zigbee2mqtt
Cloning into 'zigbee2mqtt'...
remote: Enumerating objects: 176, done.
remote: Counting objects: 100% (176/176), done.
remote: Compressing objects: 100% (125/125), done.
remote: Total 10641 (delta 123), reused 87 (delta 51), pack-reused 10465
Receiving objects: 100% (10641/10641), 13.62 MiB | 5.31 MiB/s, done.
Resolving deltas: 100% (7774/7774), done.
**root@homeassistant:/opt/zigbee2mqtt# **my_repo cd my_repo**
-bash: my_repo: command not found**
root@homeassistant:/opt/zigbee2mqtt# git reset --hard 7d27a54
HEAD is now at 7d27a54 1.12.2

After that the installation was successfull and I'm now running 1.12.2 again. But do I have to worry about the marked error concerning the change directory?

jeppper, 2020-08-31:
Is it possible to install a specific version when installing the first time?
What should I change here?
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm ci

Koenkk, 2020-08-31:
After git clone, do `git checkout tags/1.14.1`

jeppper, 2020-08-31:
So it gets: ?
sudo git clone git checkout tags/1.14.1 /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
npm ci

Koenkk, 2020-08-31:
```
sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
sudo chown -R pi:pi /opt/zigbee2mqtt
cd /opt/zigbee2mqtt
git checkout tags/1.14.1
npm ci
```

jeppper, 2020-08-31:
Thanks. Is it also possible to update to the dev branch for lets say 1.14.1?

Koenkk, 2020-08-31:
no (because its not a single commit, these are all commits between 1.14.1 and 1.14.2)

jeppper, 2020-08-31:
okay, so switching to dev branch is only for the latest master release?

Koenkk, 2020-08-31:
yes

jeppper, 2020-08-31:
Thanks for clearing this up :-)

rbswift, 2022-09-01:
Is this the same thing?

```
npm ERR! code 1
npm ERR! path /opt/zigbee2mqttnew/node_modules/@serialport/bindings
npm ERR! command failed
npm ERR! command sh /tmp/install-6244ae6a.sh
npm ERR! make: Entering directory '/opt/zigbee2mqttnew/node_modules/@serialport/bindings/build'
npm ERR!   CXX(target) Release/obj.target/bindings/src/serialport.o
npm ERR! make: Leaving directory '/opt/zigbee2mqttnew/node_modules/@serialport/bindings/build'
npm ERR! prebuild-install WARN install No prebuilt binaries found (target=16.17.0 runtime=node arch=arm libc= platform=linux)
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.0.0
npm ERR! gyp info using node@16.17.0 | linux | arm
npm ERR! gyp info find Python using Python version 3.7.3 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/opt/zigbee2mqttnew/node_modules/@serialport/bindings/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/pi/.cache/node-gyp/16.17.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/home/pi/.cache/node-gyp/16.17.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/home/pi/.cache/node-gyp/16.17.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/opt/zigbee2mqttnew/node_modules/@serialport/bindings',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:78:69: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!    uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);
npm ERR!                                                                      ^~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:135:71: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!    uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);
npm ERR!                                                                        ^~~~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:175:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!    uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);
npm ERR!                                                                       ^~~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:215:70: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!    uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);
npm ERR!                                                                       ^~~~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:271:68: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!    uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);
npm ERR!                                                                     ^~~~~~~~~~~~
npm ERR! ../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Get(Nan::NAN_METHOD_ARGS_TYPE)’:
npm ERR! ../src/serialport.cpp:316:68: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
npm ERR!    uv_queue_work(uv_default_loop(), req, EIO_Get, (uv_after_work_cb)EIO_AfterGet);
[... cut ...]
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 5.10.103-v7+
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /opt/zigbee2mqttnew/node_modules/@serialport/bindings
npm ERR! gyp ERR! node -v v16.17.0
npm ERR! gyp ERR! node-gyp -v v9.0.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2022-09-01T08_14_46_349Z-debug-0.log
pi
```