Koenkk/zigbee2mqtt #1520: Zigbee2mqtt V1.3.1 - AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)

c3reeman, 2019-05-12:
Hi, Hoping some one can provide some help on the errors shown below, My zigbee network was stable and i had no problems for months but just recently i just cant get things up and running.

My setup

Raspbery Pi - Home Assistant 
zigbee2mqtt V1.3.1 ( tried edge version, same error)
CC2531 as co ordinator, latest firmware used "CC2531_DEFAULT_20190223" ( also tried Max stability still same error  )
x 2 - CC2530+CC2591 setup as Routers
x 4 Aqara Bulbs
x 5 Aqara Door sensors
x 2 Aqara Motion sensors
x 1 Aqara Moisture sensor
x 1 Aqara Cube switch
x 2 Aqara single button switch 

`[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-05-12T08:12:40: PM2 log: Launching in no daemon mode
2019-05-12T08:12:40: PM2 log: App [npm:0] starting in -fork mode-
2019-05-12T08:12:40: PM2 log: App [npm:0] online
> zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
> node index.js
  zigbee2mqtt:info 5/12/2019, 8:12:46 AM Logging to directory: '/share/zigbee2mqtt/log/2019-05-12.08-12-46'
  zigbee2mqtt:info 5/12/2019, 8:12:48 AM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 5/12/2019, 8:12:48 AM Starting zigbee-shepherd
  zigbee2mqtt:info 5/12/2019, 8:12:48 AM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 5/12/2019, 8:13:48 AM Starting zigbee-shepherd
  zigbee2mqtt:error 5/12/2019, 8:13:48 AM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 5/12/2019, 8:13:48 AM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 5/12/2019, 8:13:48 AM Failed to start
	{"message":"Error: No such file or directory, cannot open /dev/ttyACM0","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0"}
  zigbee2mqtt:error 5/12/2019, 8:13:48 AM Exiting...
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-12T07_13_49_023Z-debug.log
2019-05-12T08:13:51: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-05-12T08:13:51: PM2 log: App [npm:0] starting in -fork mode-
2019-05-12T08:13:51: PM2 log: App [npm:0] online
> zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
> node index.js`

ssirkakriss, 2019-05-12:
> zigbee2mqtt:error 5/12/2019, 8:13:48 AM Failed to start
	{"message":"Error: No such file or directory, cannot open /dev/ttyACM0","stack":"Error: Error: No such file or directory, cannot open /dev/ttyACM0"}

Looks like the the coordinator was not found on the specific port. 
from command prompt type:
`ls /dev/ttyACM*`  
To list all possible ports the coordinator could be on. 

Hope it helps.

c3reeman, 2019-05-12:
Thanks so much for that it was on ACM1 and not ACM0 now it works perfectly, but i don't understand why the port changed the usb stick has always been in the same port.

c3reeman, 2019-05-12:
So i just unplugged it and plugged it back in (same usb port) and it reverted back to ACM0 Is there a way i can lock this some way to one or the other ? disabling the other 3 usb ports ?

c3reeman, 2019-05-12:
Im getting another error now ?? it did seem to fix it i got my network up and running again but after a home assistant start i got an error again, This time round ive verified the port is good....

```
  zigbee2mqtt:info 5/12/2019, 10:55:22 PM Logging to directory: '/share/zigbee2mqtt/log/2019-05-12.22-55-22'
  zigbee2mqtt:info 5/12/2019, 10:55:23 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 5/12/2019, 10:55:23 PM Starting zigbee-shepherd
  zigbee2mqtt:info 5/12/2019, 10:55:25 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 5/12/2019, 10:56:25 PM Starting zigbee-shepherd
  zigbee2mqtt:error 5/12/2019, 10:56:27 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 5/12/2019, 10:56:27 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 5/12/2019, 10:56:27 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.3.1/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.3.1/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 5/12/2019, 10:56:27 PM Exiting...
npm
 ERR! code ELIFECYCLE
npm 
ERR! errno 1
npm
 ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR!
 Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-12T21_56_27_254Z-debug.log
2019-05-12T22:56:29: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-05-12T22:56:29: PM2 log: App [npm:0] starting in -fork mode-
2019-05-12T22:56:29: PM2 log: App [npm:0] online
> zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
> node index.js
```

shadow-fight, 2019-05-13:
Same error happened to me today, have re-install the add-on and the home assistant but can't solve the problem, maybe it because from a update?

Koenkk, 2019-05-13:
http://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html could help

olanga, 2019-05-26:
I had the same issue. Solved by flashing again the CC2531. Nothing else helped.

HaNgocLong, 2019-05-27:
> http://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html could help

I have the same issue with the CC2530 with the guide in [http://www.zigbee2mqtt.io/information/connecting_cc2530.html](url). 
When performing the diagnostic on the link [ http://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html ](url).  

`
pi@hassbian:/opt/zigbee2mqtt $ ls /dev/ttyAMA0
/dev/ttyAMA0
`
`
pi@hassbian:/opt/zigbee2mqtt $ test -w /dev/ttyAMA0 && echo success || echo failure
success
`
Please give some suggestion!

closed 2019-08-02 by stale[bot]