Koenkk/zigbee2mqtt #1521: Cannot pair Ikea Tradfri Bulb - "Cannot get the Node Descriptor of the Device"

bhopkinson, 2019-05-12:
# Bug Report
## What happened

I have successfully paired 9 Ikea Tradfri GU10 400 bulbs, however I cannot get the 10th to pair.

## What did you expect to happen

Bulb to pair successfully.

## How to reproduce it (minimal and precise)

I turn the bulb on/off 6 times, and the bulb flashes to indicate it's reset. Usually I see nothing in the logs, and sometimes when I try I see the following:
`error: Cannot get the Node Descriptor of the Device: 0xd0cf5efffeda849d (Error: Timed out after 10000 ms)
`

- I have restarted zigbee2mqtt docker container.
- I have checked database.db and the device does not appear in there.
- I have reflashed the CC2531 firmware.
- I have turned on DEBUG=* and the logs during pairing appear as follows: [https://pastebin.com/PsHz8qPP](https://pastebin.com/PsHz8qPP)

## Debug Info
zigbee2mqtt version: version 1.3.1 (commit #unknown)
CC253X firmware version: 20190223

Thank you.

andreasbrett, 2019-05-13:
Is the device you're trying to pair to (=the device you're holding the bulb close to in order to pair both) saturated? There is a max number of devices that can be paired to a device. Maybe try pairing to a different device and see if the error persists.

bhopkinson, 2019-05-18:
I was trying to pair to the coordinator CC2531. This probably had all of the existing devices paired to it, which was 13 previously. I thought the standard firmware allowed 15? Is there anyway to tell how many devices are paired to a particular device at anytime?

Anyway, today I managed to get some time to try again, and I tried pairing in a different room close to the bulbs it will be installed next to. It worked first time in there.

Thank you.

closed 2019-05-18 by bhopkinson