Koenkk/zigbee2mqtt #1523: service restarts in a loop due to error

sparky-Ol, 2019-05-12:
# Bug Report
## What happened
I did install zigbee2mqtt as a deamon on a pi 3 according to manual. Checking the journalctl after installation showed an error which forces restart in a loop.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: current
CC253X firmware version:

Log 1: https://pastebin.com/xZAQMDW8
Journalctl: https://pastebin.com/F640pRD4

andreasbrett, 2019-05-13:
Logs say zigbee2mqtt can't connect to your mqtt server. Have you checked the credentials? Is your mqtt server up and running?

sparky-Ol, 2019-05-13:
Hi Andreas,
you were right. Mosquitto did not get the right PW-File. When I changed to anonymous login it worked again. Now i fixed the pw-file and it works with credentials login.
So i am closing here.

closed 2019-05-13 by sparky-Ol