Koenkk/zigbee2mqtt #1525: ZigBee Router unable to update its binary sensor state in real time

jerrychong25, 2019-05-13:
# Bug Report
## What happened
I have one ZigBee Router (CC2530 + CC2591) that connected to two ZigBee Coordinators (CC2531) without any issue.

However, the state of both ZigBee Router (binary_sensor.0x00XXXXXXXXXX_router) will only updates in more than 1 minutes or later, as per screenshot below:
[image]

It is suppose to connect at this time, but not reflect in real time. This have confused me for a while.

Similar issue: https://github.com/Koenkk/zigbee2mqtt/issues/1227

## What did you expect to happen
The state of ZigBee Router should reflect to Connect or Disconnected in real time (Around 1 seconds is acceptable).

## How to reproduce it (minimal and precise)
1) Install firmware on ZigBee Coordinator and ZigBee Router.
2) Turn on all devices.
3) Connect them by pairing them.
4) State of router not update real time.

## Debug Info
zigbee2mqtt version: 1.3.1 (Hass.io add on)
CC2531 firmware version: 20190223 as per https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/default/CC2531
CC2530 firmware version: 201902 as per https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2530_CC2591/bin

Could advise on this issue?

Koenkk, 2019-05-14:
This state reflects the state of the led, not the disconnected state. 

Should be fixed in the dev branch (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

jerrychong25, 2019-05-15:
> This state reflects the state of the led, not the disconnected state.
> 
> Should be fixed in the dev branch (http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Noted on this is fixed in dev branch.

When will this fix release on master branch?

Thanks.

Koenkk, 2019-05-15:
With the next zigbee2mqtt release (1.5), thanks for confirming.

closed 2019-05-15 by Koenkk

jerrychong25, 2019-05-15:
> With the next zigbee2mqtt release (1.5), thanks for confirming.

Noted and thanks