Koenkk/zigbee2mqtt #1526: Double topic For Philips hue Dimmer

Bastiencc, 2019-05-13:
Hi 

I just update to the last version (1.4.0 and last cc2531 firmware) great works guys thanks !

Something seem to be not like before, when i press my Philips hue dimmer switch, now I get 2 topics (only get 1 before.
First topic is ok, but the second is strange, `"action":""` is empty and there is no duration :
```
May 13 19:58:52 raspberrypi npm[2952]:   zigbee2mqtt:info 5/13/2019, 7:58:52 PM MQTT publish: topic 'zigbee2mqtt/salon_switch_mur', payload '{"battery":100,"linkquality":39,"counter":1,"brightness":205,"action":"on-press","duration":0}'
May 13 19:58:52 raspberrypi npm[2952]:   zigbee2mqtt:info 5/13/2019, 7:58:52 PM MQTT publish: topic 'zigbee2mqtt/salon_switch_mur', payload '{"battery":100,"linkquality":39,"counter":1,"brightness":205,"action":""}'
```

Same issue with my xiaomi aqara cube:
```
May 13 20:04:15 raspberrypi npm[2952]:   zigbee2mqtt:info 5/13/2019, 8:04:15 PM MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"side":3,"linkquality":39,"battery":91,"voltage":2985,"angle":54.26,"from_side":4,"to_side":0,"action":"wakeup"}'
May 13 20:04:15 raspberrypi npm[2952]:   zigbee2mqtt:info 5/13/2019, 8:04:15 PM MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"side":3,"linkquality":39,"battery":91,"voltage":2985,"angle":54.26,"from_side":4,"to_side":0,"action":""}'
May 13 20:04:15 raspberrypi npm[2952]:   zigbee2mqtt:info 5/13/2019, 8:04:15 PM MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"side":3,"linkquality":36,"battery":91,"voltage":2985,"angle":54.26,"from_side":4,"to_side":0,"action":"slide"}'
May 13 20:04:15 raspberrypi npm[2952]:   zigbee2mqtt:info 5/13/2019, 8:04:15 PM MQTT publish: topic 'zigbee2mqtt/aqara_cube', payload '{"side":3,"linkquality":36,"battery":91,"voltage":2985,"angle":54.26,"from_side":4,"to_side":0,"action":""}'
```

Koenkk, 2019-05-14:
This is expected and will only be send when `homeassistant: true` is in `configuration.yaml`. It allows you to use the native Home Assistant entities in your automations (https://github.com/Koenkk/zigbee2mqtt/issues/959#issuecomment-480605053)

Bastiencc, 2019-05-14:
Ok, I missreading the doc... sorry !

closed 2019-05-14 by Bastiencc