Koenkk/zigbee2mqtt #1527: Error when starting the zigbee2mqtt add-on in hassio

shadow-fight, 2019-05-13:
The CC2531 Device was working well for over 1 month until yesterday morning. The LED on sniffer display green so I reboot the hassio and the add-on, the following warning message appeared.

> [Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-05-13T23:50:40: PM2 log: Launching in no daemon mode
2019-05-13T23:50:40: PM2 log: App [npm:0] starting in -fork mode-
2019-05-13T23:50:40: PM2 log: App [npm:0] online

    zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
    node index.js
    zigbee2mqtt:info 5/13/2019, 11:50:44 PM Logging to directory: '/share/zigbee2mqtt/log/2019-05-13.23-50-44'
    zigbee2mqtt:info 5/13/2019, 11:50:45 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
    zigbee2mqtt:info 5/13/2019, 11:50:45 PM Starting zigbee-shepherd
    zigbee2mqtt:info 5/13/2019, 11:50:53 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
    zigbee2mqtt:info 5/13/2019, 11:51:53 PM Starting zigbee-shepherd
    zigbee2mqtt:error 5/13/2019, 11:51:53 PM Error while starting zigbee-shepherd!
    zigbee2mqtt:error 5/13/2019, 11:51:53 PM Press the reset button on the stick (the one closest to the USB) and start again
    zigbee2mqtt:error 5/13/2019, 11:51:53 PM Failed to start
    {"message":"request unsuccess: 4","stack":"Error: request unsuccess: 4\n at /zigbee2mqtt-1.3.1/node_modules/zigbee-shepherd/lib/components/zdo.js:45:21\n at CcZnp. (/zigbee2mqtt-1.3.1/node_modules/cc-znp/lib/ccznp.js:264:13)\n at Object.onceWrapper (events.js:273:13)\n at CcZnp.emit (events.js:182:13)\n at CcZnp._mtIncomingDataHdlr (/zigbee2mqtt-1.3.1/node_modules/cc-znp/lib/ccznp.js:366:14)\n at Immediate._onImmediate (/zigbee2mqtt-1.3.1/node_modules/cc-znp/lib/ccznp.js:342:22)\n at runCallback (timers.js:705:18)\n at tryOnImmediate (timers.js:676:5)\n at processImmediate (timers.js:658:5)"}
    zigbee2mqtt:error 5/13/2019, 11:51:53 PM Exiting...
    npm
    ERR! code ELIFECYCLE
    npm ERR! errno 1
    npm
    ERR! zigbee2mqtt@1.3.1 start: node index.js
    npm ERR! Exit status 1
    npm ERR!
    npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
    npm ERR!
    This is probably not a problem with npm. There is likely additional logging output above.
    npm ERR! A complete log of this run can be found in:
    npm ERR! /root/.npm/_logs/2019-05-13T22_51_53_618Z-debug.log
    2019-05-13T23:51:56: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
    2019-05-13T23:51:56: PM2 log: App [npm:0] starting in -fork mode-
    2019-05-13T23:51:56: PM2 log: App [npm:0] online
    zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
    node index.js
    zigbee2mqtt:info 5/13/2019, 11:51:59 PM Logging to directory: '/share/zigbee2mqtt/log/2019-05-13.23-51-59'
    zigbee2mqtt:info 5/13/2019, 11:52:00 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
    zigbee2mqtt:info 5/13/2019, 11:52:00 PM Starting zigbee-shepherd
    zigbee2mqtt:info 5/13/2019, 11:52:01 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)`

I have already tried to re-install the hassio and the add-on, tried on another board but the problem still there.

Because I don't have the debugger so it is difficult to re-flash the firmware or check it.

Anyone could help me? Thanks in advance

Koenkk, 2019-05-14:
Can you check http://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html#verify-that-you-put-the-correct-port-in-configurationyaml

gewoonik, 2019-05-17:
you can check this site https://medium.com/@fedorets.alex/hass-io-zigbee2mqtt-io-the-simplest-way-b7375c9fba22

shadow-fight, 2019-05-31:
Thanks For all the help. Have configured in the raspberry pi and all setting are correct. Finally it restored after I flash the latest version firmware in. 

Seems it just not in the working mode so that zigbee2mqtt could not communicate with the stick even if it connected in the right port.

Thanks again for all the respond.

closed 2019-05-31 by shadow-fight