Koenkk/zigbee2mqtt #1528: WXKG02LM stopped responding

b4dpxl, 2019-05-14:
At the weekend, my oldest Xiaomi WXKG02LM switch stopped sending events. At first I thought it might be the battery, which was reporting at 74%. I replaced the battery to no effect. After a couple of reset/re-pair attempts it reconnected and started working (despite saying 'Device with modelID 'undefined' is not supported'; I guess that's related to #1378 and #1401 ?). 

Is this normal? I need to update from 1.2.1, but it's been stable for months. Looking through the fixes I'm not seeing anything that looks related to this? This particular switch is also the closest one to the router.

Thanks

Bastiencc, 2019-05-14:
Hi
#1401 is fix with the 1.4.0 version, I think you should try to update

b4dpxl, 2019-05-14:
I saw that one, would that explain why it just dropped off? Also, does the firmware _have_ to be updated?

Bastiencc, 2019-05-14:
Not sure you have to update, but it will be a good idea.
When I get this pairing error with my device, I remove them from Zigbee2mqtt, stop my rpi, remove the cc2531 for one minute, plug it back, restart the rpi and then pairing works better.
you could try

b4dpxl, 2019-05-14:
I'll look at the update. I'm more concerned about why it seemed to drop off the network in the first place.

Bastiencc, 2019-05-14:
is your device still listed on zigbee2mqtt (you can see the device list on startup)  and not working? or really removed ?

b4dpxl, 2019-05-14:
It was still listed AFAIK.

closed 2019-07-20 by stale[bot]