Koenkk/zigbee2mqtt #1529: Can't pair more than 6 Xiaomi Devices

BenKuch, 2019-05-15:
I have issues to add more than 6 Xiaomi WSDCGQ11LM Device, after trying to add the 7th Sensor joining is refused, the blue LED on the Sensor is blinking just one time not 3 times short. There is no Debug information, nothing in the log. It seems to that the Coordinator is refusing to add the seventh Sensor. I run in this issue the second time with a second coordinator.

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190315

Koenkk, 2019-05-16:
Can you try with the latest firmware?

boojew, 2019-05-16:
I had a similar issue. 20190425 seems to have solved it.

closed 2019-07-15 by Koenkk