Koenkk/zigbee2mqtt #1530: Power CC2530 through ESP32?

opl-git, 2019-05-15:
Hi,
I know this is a bit off topic:
I am using an ESP32 as BLE Gateway. So I already put two EPS32 around my house and they are positioned quite well. Thought I use the 3.3 and GND Pins of the ESP32 to power my CC2530.
At first it was looking well, but after some hours the ESP32 stops working and seems to lose connection to my wifi. The CC2530 on the other hand keeps working fine. 

I guess the the CC2530 is drawing to much current, or what might be the problem?

Thank you for your help and ideas!

Kind regards

vogler, 2019-05-15:
Which board do you use?
Have you tried debugging via serial?
I triggered the ESP32's brownout detection upon wifi connection for some combinations of serial adapters/batteries/cables. Maybe try a bigger power supply or power the CC2530 from 5V via a separate LDO.

opl-git, 2019-05-16:
Thank you for your answer!
I am using a Geekcreit ESP32 module (DOIT DEV Kit V1) from Bangood.
The power supply is already at 2.1A, so I just guess the problem is that the CC2530 is sucking to much current from the ESP32 board.

vogler, 2019-05-16:
[This](https://docs.zerynth.com/latest/official/board.zerynth.doit_esp32/docs/index.html) claims the input voltage should be 7-12 V, but IMO 5V should be fine.
The [CC2530 datasheet](http://www.ti.com/lit/ds/symlink/cc2530.pdf) says <40 mA current draw during TX - that shouldn't pull the voltage down enough to be a problem.
According to [this schematic](https://dl.espressif.com/dl/schematics/ESP32-DevKitJ-v1_sch.pdf) the [LDO](https://www.onsemi.com/pub/Collateral/NCP1117-D.PDF) should also be fine (the ESP32 should draw at most [260 mA during wifi](https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/)):
[image]
But best to measure what the actual values are and add some capacitors / try different power supplies.

vogler, 2019-05-16:
Seems like a common problem with just wifi:
https://github.com/nkolban/esp32-snippets/issues/168
https://github.com/espressif/arduino-esp32/issues/863
https://www.reddit.com/r/esp32/comments/9wsip1/having_a_really_tough_time_with_brownouts_when/

closed 2019-07-22 by stale[bot]