Koenkk/zigbee2mqtt #1531: Many devices have become disconnected, shown via Network map any sugestions

c3reeman, 2019-05-15:
[image]

As you can see by my network many of my devices have become disconnected pressing the reset button on the door sensors don't help.

Is there away to establish com's again without having to rebuild the network from scratch and having to re-flash my Diy routers, Interestingly these DIY routers show as connected in home assistant, but as the map shows clearly there not

Thanks in advance

mpuff, 2019-05-16:
i have the same3 problem on all router devices.
but the devices are working.

[image]

mrosseel, 2019-05-18:
possibly related do #1536 ?

closed 2019-07-24 by stale[bot]