Koenkk/zigbee2mqtt #1532: CC2530PA module compatible?

ssirkakriss, 2019-05-16:
I picked up a few CC2530PA module.  The modules where under $5 each and had through hole solder points, a RF frontend and external antenna hookup. Many things that appealed to me. Although I've not read of Z2M being compatible or not with that  module. I felt like a gambler that day I guess.( Link to info https://www.aliexpress.com/wholesale?catId=0&initiative_id=SB_20190515163643&SearchText=CC2530PA+ )

The modules flash ok with the CC2530 + CC2591 coordinator firmware. And Z2M boots up seeing the module and the associated network. However device control is extremely sporadic. Close to one in fifty attempts work at best. 

Is this module compatible? Or did I just become the sad guinea pig.

Short log clip attached for the interested, with one error showing a attempt to switch a light and a mapping call : 
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/3185075/z2m.log)

ssirkakriss, 2019-05-16:
I noticed a FW update landed today. After posting =)

I tried the FW files from Z-Stack_Home_1.2 . CC2530_CC2591_20190515 and CC2530_CC2592_20190515 as well. The same behavior as before.

Koenkk, 2019-05-16:
@kirovilya have you every tried this device?

kirovilya, 2019-05-16:
I have not tried this module. By the picture and the description there is an RFX2401 amplifier.
I tried a similar module with RFX2401C (in "zigbee onoff controller"). Possible problem in used pins to control the amplifier.

By default, pins are recommended:
PAEN P1_1 - TX
EN P1_4 - RX
and they are used in the firmware for сс2591/сс2590.
http://processors.wiki.ti.com/index.php/Enabling_the_Support_of_CC259x_PA/LNA_with_Z-Stack-Home
[image]

And in my case, the pins were different:
P1_2 - TX RFX2401C
P1_4 - RX RFX2401C

Accordingly, you need to change the firmware and rebuild.

It is necessary to watch which pins come from the cc2530 chip to the amplifier:
[image]

ssirkakriss, 2019-05-16:
I took a photo of the backside to show pin connections to the rf fontend.
[image]

As I'm a self taught hobbyist. I'll need to do a bit of reading before I can comprehend and implement the suggestion. 

Many thanks for the tips.

ssirkakriss, 2019-05-16:
> By default, pins are recommended:
PAEN P1_1 - TX
EN P1_4 - RX
and they are used in the firmware for сс2591/сс2590.

Going by that . I think this will work. 

I'll see if I can cut the trace between the RX line and rewire to P1_4, Something like this:
[image]

That's because I better at hardware then software.

Opinions welcome. =)

kirovilya, 2019-05-17:
@ssirkakriss yes you can try :)

ssirkakriss, 2019-05-17:
Well after modding the board still no luck.

I look like their more wiring differences that I'm missing.

Going to close this, for now. I need a known working module to compare with.

Thanks for the tips.

closed 2019-05-17 by ssirkakriss

nistvan86, 2021-11-30:
Yikes, I've ordered a similar module [from here](https://www.banggood.com/CC2530-UART-Wireless-Core-Development-Board-CC2530F256-Serial-Port-Wireless-Module-2_4GHz-For-Zigbee-p-1445025.html?rmmds=myorder&cur_warehouse=CN) without realizing it's different than the standard CC2530 module and has the extra amplifier IC. Though a review on the site saying it's working with zigbee2mqtt. I'm really curious now if it's going to work for me or not, it's too late to cancel the order. Hopefully it's wired correctly.