Koenkk/zigbee2mqtt #1534: Cannot get the Node Descriptor

jesperldk, 2019-05-16:
I get "Cannot get the Node Descriptor" when trying to add a second Aqara Cube. I added the first one more than a month ago with no problems.

I am aware that there are several issues logged here with the same error message, but reading these it is still unclear for me if the fixes apply for my situation. Sorry if I am missing something obvious.

I was on 1.3.0 when I first tried adding. I have now upgraded to 1.4.0 but it does still not work. My 2531 coordinator is 20190223.

I read, that one of the other issues with same error was resolved by reflashing. Should I do that? But which version? And do I keep my bindings if I reflash - I have more than 20 devices around the house and would hate to need to rebind ;-)

Thanks in advance!

Koenkk, 2019-05-17:
When a device joins, zigbee2mqtt does a node descriptor request to the device (to gather basic information about it). This could fail for several reasons (e.g. device doesn't respond).

Most of the times you should be able to get around this by:
- Restarting zigbee2mqtt
- Re-pair the device (be sure to follow http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#xiaomi)
- Sometimes re-pairing close to the coordinator also helps.

jesperldk, 2019-05-17:
Thanks, I have however tried all three approaches :-(

Koenkk, 2019-05-17:
Do you perhaps have another cc2531? If so, could you sniff the traffic to see whats happening? http://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

jesperldk, 2019-05-18:
I will order a new cube and s 2531 for sniffing; will take some weeks for them to arrive, so I close this as everything else is working. 
Thanks.

closed 2019-05-18 by jesperldk