Koenkk/zigbee2mqtt #1535: State.json incomplete

FRFrancisqueti, 2019-05-16:
Hi Koen.
I already had zigbee2mqtt installed on an OrangePi running perfectly where the state.json brought me this info:

{
    "0x000d6f000e1e407c": {
        "contact": false,
        "tamper": false,
        "battery_low": false,
        "linkquality": 128,
        "device": {
            "ieeeAddr": "0x000d6f000e1e407c",
            "friendlyName": "0x000d6f000e1e407c",
            "type": "EndDevice",
            "nwkAddr": 14885,
            "manufId": 4619,
            "manufName": "HEIMAN",
            "powerSource": "Battery",
            "modelId": "DoorSensor-EM",
            "status": "offline"
        }}
}

I had to format and generate a new image for the card. I installed zigbee2mqtt again but now the state.json is incomplete, it does not show me the "device" data:

{
    "0x000d6f000e1e407c": {
        "contact": false,
        "tamper": false,
        "battery_low": false,
        "linkquality": 128,
}}

The sensors work perfectly. Do you know what could be happening?

Thank you.

Koenkk, 2019-05-17:
You need to set `include_device_information: true` (http://www.zigbee2mqtt.io/configuration/configuration.html). But note that this file is not meant to be used externally.

FRFrancisqueti, 2019-05-17:
I have already set the include_device_information: true, but the problem continues.
Since you commented, why is not it conditioned to use this file?
Currently I read this json periodically to check device data through another application ...

Koenkk, 2019-05-17:
It's an internal file, for integration with another application, subscribe to the corresponding MQTT device topics.

closed 2019-07-16 by Koenkk