Koenkk/zigbee2mqtt #1536: Route lose with new firmware

jasimancas, 2019-05-16:
Hi!
I have Z2M 1.4.0, i update my CC2531 coordinator with the new firmware, all right the first hours, but as the day progresses the devices start to stop working, the log tells me that there are no routes, I enter the map and I find this:
[image]

Reflash the old firmware (max_stability) and in the same time 0 problems:
[image]

Any idea?
Reflash 2 coordinator and same result.

Bojan023, 2019-05-16:
I encounter the same problems. No mesh network can be created somehow

Imperial-Guard, 2019-05-16:
Any idea where I can download the older stable version? Got the same issue here

chris-jennings, 2019-05-17:
1.4 is causing me problems as well. I can't form a mesh and when I re-pair devices it knocks out other devices because only the coordinator has active routes. I will re-flash with the older firmware to see if I get back normal functionality tonight.

Koenkk, 2019-05-17:
Can you post the log when having this 205 network errors?

Koenkk, 2019-05-17:
For users wanting to downgrade, older firmwares can be found here: https://github.com/Koenkk/Z-Stack-firmware/tree/c36ba37ebb10690245ba29d1360a21b5abfa9c43/coordinator/Z-Stack_Home_1.2/max_stability

jasimancas, 2019-05-17:
This is part of this log, i have all in a same txt file (good firm and bad firm):
> 5/15/2019, 10:36:50 PM - error: Failed to setup reporting for 0xd0cf5efffe1d7098 - genOnOff - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:50 PM - error: Failed to setup reporting for 0x90fd9ffffeee1239 - genLevelCtrl - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:51 PM - error: Failed to setup reporting for 0x90fd9ffffeee1239 - lightingColorCtrl - 3 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:51 PM - error: Failed to setup reporting for 0x90fd9ffffed9bc1e - genOnOff - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:51 PM - error: Failed to setup reporting for 0xd0cf5efffe1d7098 - genLevelCtrl - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:52 PM - warn: Failed to configure Mando Salon (0xd0cf5efffe18c94b) ('Error: Timed out after 10000 ms') (attempt #1)
5/15/2019, 10:36:52 PM - warn: This can be ignored if the device is working properly
5/15/2019, 10:36:52 PM - error: Failed to setup reporting for 0xd0cf5efffe1d7098 - lightingColorCtrl - 3 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:52 PM - error: Failed to setup reporting for 0x90fd9ffffed9bc1e - genLevelCtrl - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:53 PM - error: Failed to setup reporting for 0x90fd9ffffed9bc1e - lightingColorCtrl - 3 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:53 PM - error: Failed to setup reporting for 0x90fd9ffffef807de - genOnOff - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:53 PM - error: Failed to setup reporting for 0x90fd9ffffef807de - genLevelCtrl - 1 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)
5/15/2019, 10:36:54 PM - error: Failed to setup reporting for 0x90fd9ffffef807de - lightingColorCtrl - 3 - (Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.)

At the moment 0 problems with the older firm.
Another thing that happened to me with the new firmware is that the answers were longer, pressed and took almost 1 second to answer the bulb.

Imperial-Guard, 2019-05-17:
[code]
  zigbee2mqtt:error 5/16/2019, 6:54:30 PM Zigbee publish to device '0xd0cf5efffe                                                                                                             2ed08a', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed wi                                                                                                             th error Error: AF data request fails, status code: 205. No network route. Pleas                                                                                                             e confirm that the device has (re)joined the network.
  zigbee2mqtt:error 5/16/2019, 8:21:44 PM Zigbee publish to device '0xd0cf5efffe                                                                                                             cb89ad', genLevelCtrl - moveToLevelWithOnOff - {"level":254,"transtime":0} - {"m                                                                                                             anufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request f                                                                                                             ails, status code: 233. MAC no ack.
  zigbee2mqtt:error 5/16/2019, 9:15:02 PM Zigbee publish to device '0x90fd9ffffe                                                                                                             010c68', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed w                                                                                                             ith error Error: AF data request fails, status code: 233. MAC no ack.
  zigbee2mqtt:error 5/16/2019, 9:15:02 PM Zigbee publish to device '0x00158d0001                                                                                                             922ada', hvacThermostat - write - [{"attrId":16387,"dataType":41,"attrData":1600                                                                                                             }] - {"manufSpec":1,"manufCode":4151} - null failed with error Error: AF data re                                                                                                             quest fails, status code: 205. No network route. Please confirm that the device                                                                                                              has (re)joined the network.
  zigbee2mqtt:error 5/17/2019, 4:05:30 AM Zigbee publish to device '0xd0cf5efffe                                                                                                             2ed08a', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed w                                                                                                             ith error Error: AF data request fails, status code: 205. No network route. Plea                                                                                                             se confirm that the device has (re)joined the network.
  zigbee2mqtt:error 5/17/2019, 4:53:04 AM Zigbee publish to device '0x90fd9ffffe                                                                                                             010c68', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed wi                                                                                                             th error Error: AF data request fails, status code: 205. No network route. Pleas                                                                                                             e confirm that the device has (re)joined the network.
  zigbee2mqtt:error 5/17/2019, 4:53:31 AM Zigbee publish to device '0xd0cf5efffe                                                                                                             2ed08a', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed w                                                                                                             ith error Error: AF data request fails, status code: 205. No network route. Plea                                                                                                             se confirm that the device has (re)joined the network.
  zigbee2mqtt:error 5/17/2019, 5:14:33 AM Zigbee publish to device '0xd0cf5efffe                                                                                                             cb89ad', genLevelCtrl - moveToLevelWithOnOff - {"level":254,"transtime":0} - {"m                                                                                                             anufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request f                                                                                                             ails, status code: 205. No network route. Please confirm that the device has (re                                                                                                             )joined the network.
  zigbee2mqtt:error 5/17/2019, 5:20:39 AM Zigbee publish to device '0xd0cf5efffe                                                                                                             2ed08a', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed w                                                                                                             ith error Error:  [... cut ...]  (re)joined the network.
  zigbee2mqtt:error 5/17/2019, 5:27:43 AM Zigbee publish to device '0x90fd9ffffe                                                                                                             010c68', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed w                                                                                                             ith error Error: AF data request fails, status code: 233. MAC no ack.
rancher@RancherOS:~$
[code]

Koenkk, 2019-05-17:
Does this happen when sending a burst of commands?

jasimancas, 2019-05-17:
Press button -> Turn On light (fail)
Press button -> Turn Off light (fail)
Open door -> Turn On light (fail)
Nothing more, only this, simple and basic orders.

Imperial-Guard, 2019-05-17:
> Does this happen when sending a burst of commands?

No max 2 / 3 devices.

Koenkk, 2019-05-17:
So it even fails when just controlling a single bulb?

Imperial-Guard, 2019-05-17:
> * So it even fails when just controlling a single bulb?
> * Did commands to the same bulb succeed before (with new firmware)?

Yes and Yes :)

jasimancas, 2019-05-17:
> * So it even fails when just controlling a single bulb?
> * Did commands to the same bulb succeed before (with new firmware)?

Yes and yes.
I dont touch anything in HomeAssistant, only update Z2M and Firmware.

Koenkk, 2019-05-17:
Can you try to re-pair the specific bulbs?

**Background information**
This probably happens because the source routing has been disabled in order to fix #1408. Meaning that we are now in a deadlock situation (we either have one of the issues). Therefore I would to check if we can workaround this.

jasimancas, 2019-05-17:
I have 15 bulbs, all with this problem, its random, I repair one of this, and the problem continue.

chris-jennings, 2019-05-17:
I find that Hue bulbs seem to have a big impact on the mesh routing, but this always seems to have been an issue (even with the old firmware). Obviously with the new firmware and source routing disabled if the bulbs are not routing correctly then the mesh stops working. I have enough other devices (DIY router, Xiaomi plugs) to not let the bulbs act as routers. Is there any way to disable the Hue bulbs from being routers?

Smiggel, 2019-05-19:
Same issue here. I upgrades to Z2M 1.4.0 tonight and i started to notice that my nodes are dropping from the network. Did not flash my nodes. I run the latest stable already. 

I use Domoticz. Zigbee network was really stable.

Koenkk, 2019-05-20:
@Smiggel what firmware version are you running?

Smiggel, 2019-05-20:
@Koenkk My coordinator is running version 20190223 and my routers are running version 2018_09 (CC2531 sticks). That is stable with Z2M 1.1.1. When I upgrade to 1.4.0, the routers drop of the network after less then 30 minutes.

Koenkk, 2019-05-20:
@Smiggel what do you see in the log? How do you know they drop out of the network?

Smiggel, 2019-05-20:
@Koenkk When I watch the logs, I see the link quality drop from 50+ to zero. And after a while the nodes don’t show up in the logs. Also the sensors downstairs (coordinator is on de second floor) are big longer visible in the logs. I need the routers to get the signal on the top floor.

glentakahashi, 2019-05-23:
After upgrading to the latest firmware i'm seeing the same issue. They're still in the network, as if I add them to a group and use that to power my bulbs, I can still control them even though trying to ping the bulbs directly throws the 205 like others see.

Koenkk, 2019-05-23:
Seeing that the disabling source routing does more harm than good, I've enabled it again. CC2531 firmware can be found here: https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/CC2531_20190523.zip (it still allows for 25 direct children). Can you guys confirm it's as stable as the max stability firmware?

Smiggel, 2019-05-24:
> Seeing that the disabling source routing does more harm than good, I've enabled it again. CC2531 firmware can be found here: [...]

I flashed a CC2531 stick with this firmware. At first I see my closest router show up with a link quality of 47. Then it drops to 45 and the next time it shows up in the logs it is at 0. I have a Xiaomi sensor about 60 cm away from my coordinator and even that sensor shows up with a link quality of 0. Matter of fact, all sensors, except for two Xiaomi PIR sensors) have a link quality of 0 with the new firmware.

With my older firmware (CC2531ZNP-Prod_20190223 the link quality of my closest router (one floor down) mostly about 40.

I Run Zigbee2mqtt version version 1.1.1 (commit #e40a3ba).

MrLight, 2019-05-24:
Same for me. Actually no log available. But after flashing the coordinator with the latest firmware (and with the one out of this conversation) The network-graph is showing good routes (twoway + mesh) for some minutes after restart and drops all routes after time. It shows only oneway connection from the routers to the coordinator but no connection from the coordinator to the routers...

Koenkk, 2019-05-24:
@MrLight @Smiggel besides the networkmap, is controlling devices working reliable?

Smiggel, 2019-05-24:
> @MrLight @Smiggel besides the networkmap, is controlling devices working reliable?

In my case not all sensors respond. My routers are not visible. That could be the reason why not all pir and contact sensors respond, because of the distance.

james-fry, 2019-05-24:
I seem to be having this issue and its not related to the latest firmware AFAIK. Pretty sure this was happening before 1.4.0, too.

james-fry, 2019-05-24:
- Confirming that restarting Z2M resolves this issue for me (not that this is a long term fix)
- Still investigating what is the root cause but its happened twice today :(
- Im not sure what versions of coordinator (GBAN) and router (CC2531) I am using but all were flashed in march/april this year. I dont think its the very latest...

Koenkk, 2019-05-24:
@james-fry in that case you are not experiencing this issue but #652

james-fry, 2019-05-24:
My issue is not just a problem with map. Because the routers become non functioning a number of my sensors become unavailable. I think the map is accurately reflecting that the routers are dropped from the network and some endpoints are disconnected

Fabiancrg, 2019-05-24:
I am also running Z2M v1.4.0 (commit #0074c32) with coordinator firmware version '20190425'
It's running without any issue so why will this be linked to source routing ?

Koenkk, 2019-05-26:
@Fabiancrg it depends on the setup I guess

abmantis, 2019-05-26:
I'm also having the same behavior. Some devices stopped working so I checked the network map and all the indirect routes to the coordinator were gone. Only direct connections to the coordinator remained.
Restarting z2m restored normal behavior again.
Coordinator firmware version: '20190223'

MrLight, 2019-05-27:
Update from my side: I'm using the actual coordinator cc2531 firmware and I have reflashed my two routers once again (cc2530 - same firmware as before | not sure if this is important) . I'm using the latest master branch.
Yesterday evening all my routers were connected to the mesh and communication was up and running.
In the morning it looked like the mesh was broken and all devices were assigned to the coordinator. 
After that I moved one device into a physical area were the coordinator was out of range but one router was reachable. After forcing some transmit from the device the complete mesh was rebuilt or visible again in the networkmap. So I'm not 100% sure if I really have a problem or if it is only the map... My actual feeling is that the direction from device to coordinator is working. But the direction from coordinator to device is not working when the mesh is broken... But I have to double check that.

Koenkk, 2019-05-27:
Note that this issue is not about things not showing in the network map. It's about errors after flashing `20190425` firmware (like https://github.com/Koenkk/zigbee2mqtt/issues/1536#issuecomment-493335557). For issues in the networkmap please refer to https://github.com/Koenkk/zigbee2mqtt/issues/652 to avoid cluttering of this issue.

Smiggel, 2019-05-27:
Ok, i just did another try. I stopped Zigbee2mqtt and switched sticks. I then started Zigbee2mqtt and after it started, I did a restart. My nodes are now still visible and have a link quality of 50+. My routers are still on the 3 month old version. I did upgrade Zigbee2mqtt to 1.4. That runs fine now too, as before it would not start.

I will keep monitoring the status of my network for the rest of the evening and night. Hopefully the restart fixed it.

MrLight, 2019-05-29:
Update from my side: System is stable with latest coordinator firmware and latest dev branch.
Thank you very much!

Koenkk, 2019-05-29:
@MrLight what version number of the firmware?

MrLight, 2019-05-29:
20190523

miroslavpetrov, 2019-05-30:
I also updated zigbee2mqtt to 1.4 and flashed my cc2531 coordinator with 3.0 firmware 20190425. The link quality of my sensors went from around 80 to 30 and no traffic was forwarded from the cc2530 routers that I have. I went back to 1.2 firmware 20190425 and the routers started to work again but the problem with the link quality is still here. 2 of the xiaomi temp and hum sensors are unable to send data to the coordinator. If I press the button on the sensor few times to send data manualy it works after the 5th or 6th press.

Smiggel, 2019-05-30:
@miroslavpetrov Did you try restarting Zigbee2mqtt? Helped most of us, including me.

miroslavpetrov, 2019-05-30:
@Smiggel I also did complete reinstall and added the nodes again.

Ton1965, 2019-05-30:
@miroslavpetrov,

Could it be your zigbee channel changed after re-installation? I sometimes have intermittent problems with Xiaomi sensors; link quality just drops significantly. This can last for a few hours then everything is good again. I attribute this to some external interference. My guess is that you might have inadvertently changed the channel and got higher interference with WiFi or some other source. It's just a guess though.

UnrealKazu, 2019-05-30:
After upgrading to 1.4 and firmware 20190425, I encountered the same 205 errors as jasimancas had. My sensor end devices were still operating, but the routers were not. When I flashed the 20190523 firmware from this thread, the routers still did not function, but as soon as soon as I restarted `zigbee2mqtt` with pairing enabled, the routers repaired themselves and started working again.

I've been running a stable network for two days now. Not sure if my problem is directly related to jasimancas', but maybe this piece of information has some value.

enomam, 2019-05-30:
Is everyone just upgrading their coordinator firmwares, or router firmware too?

Smiggel, 2019-05-31:
> Is everyone just upgrading their coordinator firmwares, or router firmware too?

The problem for me happened after upgrading only the coordinator firmware. The router firmware is working fine and does not need an upgrade.

UnrealKazu, 2019-05-31:
> 
> 
> Is everyone just upgrading their coordinator firmwares, or router firmware too?

I do not have custom routers, only third party routers (TRADFRI stuff). So I only upgraded the coordinator.

vogler, 2019-05-31:
@Koenkk Could you please also update the CC2530 coordinator firmware? Can it only be built using Windows?

Smiggel, 2019-05-31:
Perhaps useful. I paired all my zigbee devices to a Home Assistant installation. Routers show up just fine. Not with 0 and don’t have to restart Zigbee2mqtt.

Smiggel, 2019-05-31:
Ok, I was busy with switching all my Zigbee devices to Hass.io, coming from Domoticz. I setup the Zigbee map and noticed that all my sensors did no not communicate with my two routers. So, I thought I would connect my 4 sensors on the ground floor to the router on the ground floor by repairing them close to the stick.

I noticed after doing that, my sensors connected to the router. I could see it on the map. However, I also noticed that 1 of my door sensors and PIR sensors signal did not come through to the coordinator at all. The link quality of the router is 50+. The link quality to my sensors was also above 20. But the signal of opening and closing a door does not come through at all. I had to repair it close the the coordinator in order to make it work.

I use the 2018-09 router firmware. My Zigbee2Mqtt version on Hass.io is 1.4.0. And my router firmware is CC2531_20190523.

jasimancas, 2019-06-04:
Updating to CC2531_20190523.zip

Koenkk, 2019-06-05:
@jasimancas did you give the coordinator time to build up the  routes? Because this part is exactly the same as the max stability firmware.

jasimancas, 2019-06-05:
> @jasimancas did you give the coordinator time to build up the routes? Because this part is exactly the same as the max stability firmware.

19:00 - Update
19:10 - 22:00 All OK
22:00 - 23:30: aleatory buttons dont work, light dont turn on when I push them.
23:30: turn off nuc, change USB Coordinator, turn on
23:40 - 08:45: all OK

Koenkk, 2019-06-05:
@jasimancas at 23:30, what error did you see in the log when controlling the bulbs?

jasimancas, 2019-06-05:
The same one, this afternoon I erase all the log and I connect again the USB with the new version, and I attach the LOG if there are errors.

tricoos, 2019-06-09:
As @Koenkk removed the build linked above and created a new firmware build yesterday at https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/CC2531_20190608.zip I'm now using that one to see if it resolves the issue for me.

jasimancas, 2019-06-09:
Flashing new version and debug log active

jasimancas, 2019-06-10:
Again problem...
[image]

Koenkk, 2019-06-10:
@tricoos @jasimanacas can you try `20190610` https://github.com/Koenkk/Z-Stack-firmware/blob/dev/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190610.zip

[49 more replies]

closed 2019-06-26 by Koenkk