Koenkk/zigbee2mqtt #1538: Cannot start add-on

TimMakes, 2019-05-17:
When I want to start the add-on this is shown in the log. What can I do now?

[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-05-17T13:25:21: PM2 log: Launching in no daemon mode
2019-05-17T13:25:21: PM2 log: App [npm:0] starting in -fork mode-
2019-05-17T13:25:21: PM2 log: App [npm:0] online
> zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
> node index.js
  zigbee2mqtt:info 5/17/2019, 1:25:25 PM Logging to directory: '/share/zigbee2mqtt/log/2019-05-17.13-25-25'
  zigbee2mqtt:info 5/17/2019, 1:25:26 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 5/17/2019, 1:25:26 PM Starting zigbee-shepherd
  zigbee2mqtt:info 5/17/2019, 1:25:27 PM zigbee-shepherd started
  zigbee2mqtt:info 5/17/2019, 1:25:27 PM Coordinator firmware version: '20190223'
  zigbee2mqtt:info 5/17/2019, 1:25:27 PM Currently 0 devices are joined:
  zigbee2mqtt:info 5/17/2019, 1:25:27 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 5/17/2019, 1:25:27 PM Connecting to MQTT server at 127.0.0.1
/zigbee2mqtt-1.3.1/node_modules/q/q.js:155
                throw e;
                ^
Error: Missing protocol
    at Object.connect (/zigbee2mqtt-1.3.1/node_modules/mqtt/lib/connect/index.js:64:13)
    at MQTT.connect (/zigbee2mqtt-1.3.1/lib/mqtt.js:39:28)
    at Controller.onZigbeeStarted (/zigbee2mqtt-1.3.1/lib/controller.js:113:19)
    at zigbee.start (/zigbee2mqtt-1.3.1/lib/controller.js:173:26)
    at shepherd.start (/zigbee2mqtt-1.3.1/lib/zigbee.js:78:17)
    at /zigbee2mqtt-1.3.1/node_modules/q/q.js:2055:17
    at runSingle (/zigbee2mqtt-1.3.1/node_modules/q/q.js:137:13)
    at flush (/zigbee2mqtt-1.3.1/node_modules/q/q.js:125:13)
    at process._tickCallback (internal/process/next_tick.js:61:11)
npm
 ERR!
 code ELIFECYCLE
npm ERR! errno 1
npm ERR!
 zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR!
 This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-17T11_25_27_919Z-debug.log
2019-05-17T13:25:32: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]

Koenkk, 2019-05-17:
You probably need to add `mqtt://` before `server` in your configuration (http://www.zigbee2mqtt.io/configuration/configuration.html)

Koenkk, 2019-07-16:
Closing due to inactivity of author.

closed 2019-07-16 by Koenkk