Koenkk/zigbee2mqtt #1539: zigbee reconnection behavior after loss of power

Stefan300381, 2019-05-17:
Hi, this is more a question than an issue. Any forum where I should have posted it?

I have some end devices that connect to a Router. Now it happened that this router was out of power for an undefined amount of time (maybe several days). Once it got power it communicated again to the coordinator. But the end devices didn't connect back, even after some hours. Is this expected behavior? What helped was to press the button on the end device (here xiaomi aqara contact sensors). I still have one aqara body sensor, which doesn't generate any traffic.

so my questions are:

1. the router periodically sends this:
>> MQTT publish: topic 'zigbee2mqtt/Router-01', payload '{"state":false,"linkquality":70}'

What does state mean?

2. Is there any other way (but pressing the physical button ON the end device) to make it re-connect to the Router?

3. What is the default re-connection process? Like once every 5 min for 24hours and then followed by once ever 60min for the next 24hours and then never again?

Thanks in advance

Stefan

Stefan300381, 2019-05-17:
Additionally: 

I noticed that after a restart of zigbee2mqtt it was again required to push the button on each end device. And I still saw the "state false" messages from the router. Then I had the idea to enable permit join on the Coordinator. Voila! state true from the Router and devices are connecting and messaging as expected.

sascha777, 2019-05-17:
I can confirm that after a restart of a router it is required to restart zigbee2mqtt with join permitted to get the router back online. Else it gives the "state false" message and is shown "offline" in the network map.

Stefan300381, 2019-05-17:
awesome, thanks!

Stefan300381, 2019-05-20:
and now I can confirm that if you DO NOT press the button on the end device, it will never come back (at least it didn't after 3 days). so I assume, once it thinks it won't be able to connect again to the router (or probably coordinator as well) it will go into unlimited standby

Stefan300381, 2019-05-22:
it's driving me crazy. next power outage and I enabled pairing (OK, router can connect succesfully). now one of my 4 window/door sensors doesn't communicate. if I press the local button on the contact, it's sending his payload, but no further automatic teledata :-(

9600, 2019-06-18:
I think I've just run into this. To summarise:

* Started with CC2531ZNP-Prod_20190109 and zigbee2mqtt v1.4.0.
* Everything was running fine for a few weeks, with 3x Salus SP600 + 1x Philips Hue motion sensor + 2x Philips Hue bulbs
* Added three more Salus SP600 and then the CC2531 started to crash after a short while of running...
* Shut everything down
* Upgraded a few days later to CC2531_DEFAULT_20190608
* Had to re-pair all the SP600s... And it took numerous attempts with each

Could the SP600 plugs be losing their keying info? This is incredibly annoying behaviour and so I'm keen to avoid it, even if it means throwing 6x SP600 in the bin and buying something else. Also the Hue bulbs haven't come back online and I've not attempted re-pairing yet.

closed 2019-08-24 by stale[bot]