Koenkk/zigbee2mqtt #1540: Cannot pair Devices

DerMattes, 2019-05-17:
# Bug Report
## What happened
Wanted to pair WXKG11LM and MCCGQ11LM, but didn't work:
` zigbee2mqtt:warn 5/17/2019, 9:30:49 PM Device with modelID 'undefined' is not supported.
zigbee2mqtt:warn 5/17/2019, 9:30:49 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

## What did you expect to happen
Correct pairing like with similar devices in the past, already have several MCCGQ11LM and one WXKG11LM working.

## How to reproduce it (minimal and precise)
Enable paring, press and hold pairing button on device.

## Debug Info
zigbee2mqtt version: HASS.IO Addon 1.3.1
CC253X firmware version: 20190425, upgraded from 20181024 but did not change anything.

Koenkk, 2019-05-17:
Should be fixed in zigbee2mqtt 1.4, once updated to 1.4 just press the reset button on the device once and they will be recognized.

DerMattes, 2019-05-21:
> Should be fixed in zigbee2mqtt 1.4, once updated to 1.4 just press the reset button on the device once and they will be recognized.

Sadly not the case. Just updated the Hass.io addon to 1.4 but still get the same message and no new device. It does not change if I long press (like joins with past new devices) or just click the reset button.

Koenkk, 2019-05-22:
Can you provide the log when pressing the reset button?

DerMattes, 2019-05-24:
`  zigbee2mqtt:info 5/24/2019, 9:01:58 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 5/24/2019, 9:01:58 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"info","permit_join":true}'
  zigbee2mqtt:warn 5/24/2019, 9:02:37 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/24/2019, 9:02:37 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Koenkk, 2019-05-26:
Could you post the `log_level: debug` log? (http://www.zigbee2mqtt.io/configuration/configuration.html)

DerMattes, 2019-05-27:
`zigbee2mqtt:info 5/27/2019, 8:32:35 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"debug","permit_join":false}'
  zigbee2mqtt:debug 5/27/2019, 8:32:36 PM Received MQTT message on 'zigbee2mqtt/bridge/config/permit_join' with data 'true'
  zigbee2mqtt:info 5/27/2019, 8:32:36 PM Zigbee: allowing new devices to join.
  zigbee2mqtt:info 5/27/2019, 8:32:36 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.4.0","commit":"unknown","coordinator":20190425,"log_level":"debug","permit_join":true}'
  zigbee2mqtt:warn 5/27/2019, 8:32:58 PM Failed to configure 0x000d6ffffe23e62b (0x000d6ffffe23e62b) ('Error: Timed out after 10000 ms') (attempt #1)
  zigbee2mqtt:warn 5/27/2019, 8:32:58 PM This can be ignored if the device is working properly
  zigbee2mqtt:debug 5/27/2019, 8:32:58 PM Configuring 0x000b57fffe2f8da5 (0x000b57fffe2f8da5) ...
  zigbee2mqtt:debug 5/27/2019, 8:33:00 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d0001e79b8a"' of device 'undefined' (0x00158d0001e79b8a) of endpoint 1
  zigbee2mqtt:warn 5/27/2019, 8:33:00 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/27/2019, 8:33:00 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 5/27/2019, 8:33:01 PM Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'Plug 01' (0x7cb03eaa0a014ff5) of endpoint 3
  zigbee2mqtt:info 5/27/2019, 8:33:01 PM MQTT publish: topic 'zigbee2mqtt/0x7cb03eaa0a014ff5', payload '{"state":"OFF","linkquality":26}'
  zigbee2mqtt:debug 5/27/2019, 8:33:01 PM Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":0}}' of device 'Plug 01' (0x7cb03eaa0a014ff5) of endpoint 3`

This is from switching on the allow-new-evices feature until after the rest button was pressed. After this there no other messages until devices updated their status.

Koenkk, 2019-05-28:
I don't see the expected messages when clicking the reset button on the Xiaomi devices.

I think it's the best to know:
- Stop zigbee2mqtt
- Remove all entries of non working devices from `data/database.db`
- Start zigbee2mqtt
- Re-pair

DerMattes, 2019-05-31:
Done, but now I get the followong message:
`zigbee2mqtt:error 5/31/2019, 12:44:05 PM Cannot get the Node Descriptor of the Device: 0x00158d0001e79b8a (Error: Timed out after 10000 ms)`
I also tried to pair on a repeater, that does not work either. After two attempts, the message changed:
`zigbee2mqtt:info 5/31/2019, 12:45:30 PM Device incoming...
  zigbee2mqtt:info 5/31/2019, 12:45:30 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 5/31/2019, 12:45:30 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/31/2019, 12:45:30 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Koenkk, 2019-06-03:
@DerMattes did you follow http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#xiaomi?

DerMattes, 2019-06-19:
@Koenkk yes, I followed the instructions but I have several Xiaomi devices paired and working already. But with one of the later patches pairing stopped working for me. I'm a little bit frustrated as I have 3 more dor sensors and one switch useless on my desk and 4 more temp sensors coming in as well.

My existing installation works, so I do not want to start over.

DerMattes, 2019-07-27:
After updating to v1.5.1.1, reflashing the firmware, switchign to the source routing firmware and intensive troubleshooting due to update problems, at least my existing 31 devices are working again.

Then I tried to pair two new Aquara door/window sensors and a temp/hum./etc. sensor.

`zigbee2mqtt:info 7/27/2019, 12:37:19 PM New device 'undefined' with address 0x00158d0001e79b8a connected!
  zigbee2mqtt:info 7/27/2019, 12:37:19 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0001e79b8a","meta":{}}'
  zigbee2mqtt:warn 7/27/2019, 12:37:19 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/27/2019, 12:37:19 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 7/27/2019, 12:37:31 PM Successfully reenabled joining`

So I'm still not able to pair new devices. I'm trying to pair in range of the CC2531 and outside but in range of a repeater, no success.
What can I do?

Could this be a problem with (factory-) updated firmware of the Aquara devices?!?

Koenkk, 2019-07-27:
Are you sure that some time after that it paired successfully? It can take some minutes before the model ID arrives, otherwise remove all entries of `0x00158d0001e79b8a` from `data/database.db` and try again. Also make sure that you use the correct pairing procedure: https://www.zigbee2mqtt.io/devices/MCCGQ11LM.html#pairing

DerMattes, 2019-07-27:
I removed the entries from the database (there were 2) and tried to repair.

I got zhe same 'undefined' message as before, but after that there were some messages posted more than once:
`zigbee2mqtt:warn 7/27/2019, 8:57:02 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/27/2019, 8:57:02 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/27/2019, 8:57:20 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/27/2019, 8:57:20 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 7/27/2019, 8:58:40 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 7/27/2019, 8:58:40 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 7/27/2019, 8:58:53 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 7/27/2019, 8:58:53 PM Zigbee: disabling joining new devices.
  zigbee2mqtt:info 7/27/2019, 8:58:53 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190619,"log_level":"info","permit_join":false}'
  zigbee2mqtt:info 7/27/2019, 8:58:53 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.5.1","commit":"unknown","coordinator":20190619,"log_level":"info","permit_join":false}'
  zigbee2mqtt:info 7/27/2019, 8:59:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c906b2', payload '{"humidity":46.81,"linkquality":0,"temperature":31.08,"pressure":964.5,"battery":100,"voltage":3065}'
  zigbee2mqtt:info 7/27/2019, 8:59:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c906b2', payload '{"humidity":46.63,"linkquality":0,"temperature":31.08,"pressure":964.5,"battery":100,"voltage":3065}'
  zigbee2mqtt:info 7/27/2019, 8:59:10 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c906b2', payload '{"humidity":46.63,"linkquality":0,"temperature":31.08,"pressure":964,"battery":100,"voltage":3065}'
  zigbee2mqtt:info 7/27/2019, 8:59:23 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002bf9a1b', payload '{"temperature":22.89,"linkquality":0,"humidity":73.64,"pressure":965,"battery":97,"voltage":2995}'
  zigbee2mqtt:info 7/27/2019, 8:59:45 PM MQTT publish: topic 'zigbee2mqtt/0x00158d0002c906d1', payload '{"temperature":27.34,"linkquality":0,"humidity":57.51,"pressure":964.3,"battery":86,"voltage":2975}'`

I didn't noticed such copies before, is this because of the source routing firmware?

Koenkk, 2019-07-29:
Looks that the device is paired now, otherwise you won't see the temperature, humidity, etc.

DerMattes, 2019-07-29:
> Looks that the device is paired now, otherwise you won't see the temperature, humidity, etc.

No, that is a different device. As I said, I have already 30+ devices paired and working with zigbee2mqtt.
I just noticed that a couple messages are posted more than once.

I am still unable to pair new devices, with the pairing instructions of the documentation.
I also tried slightly other procedures like I successfully paired devices of this type in the past, but these were unsuccessful as well.

So since zigbee2mqtt 1.3.x I can not pair new devices!
Any help is much appreciated!

Koenkk, 2019-07-29:
Do you have enough routers in your network to support all those devices?

DerMattes, 2019-07-29:
Until 10 minutes ago I had 3 Osram Smart+ Plugs and 2 Ikea light bulbs as routers, the rest are Aquara sensors.
I had a 4th Osram Smart+ Plug unused, so I tried to pair it. It worked like a charm and paired immediately.

But still no luck with the additional Aquara sensors, they still do not pair.

Koenkk, 2019-07-30:
Do you see anything when you try to pair them? Did you try pairing it close to the coordinator?

DerMattes, 2019-07-30:
The [WSDCGQ11M](https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html) blinks 4 times long, then a short pause and then 3 times fast. As far as I remember pairing the other 11 of these devices I already have in use, the fast blinks should be long ones to indicate a successful pairing.
Besides this I only get the 'undefined' message in the log, switching to debug loggin brought no additional infos.

I tried pairing at the coordinator and at different routers, no change in behaviour. As I now use the source routing firmware, pairing at the coordinator would not work anymore, right?

Will 6 routers be enough for like 36 battery powered devices?

Koenkk, 2019-07-31:
@DerMattes source routing firmware shouldn't matter, 6 routers may be tricky, you could try adding another CC2531 router.

DerMattes, 2019-07-31:
@Koenkk thanks so far, why a CC2531 and not another powered zigbee device? Stability or number of devices supported?

Koenkk, 2019-08-01:
More devices tend to connect to this. (especially Xiaomi as the modelId has a xiaomi prefix).

closed 2019-09-30 by Koenkk