Koenkk/zigbee2mqtt #1541: Xiaomi devices don't pair to routers

sascha777, 2019-05-17:
## What happened
I updated to v1.4.0 and also flashed my coordinator with the new unique firmware 20190515. In my network I have 2 CC2531 Routers running the 20180927 FW (also tested 20190204) and an Innr SP 120 plug and an OSRAM power plug+ that also running as routers.
Since 1.4.0 pairing Xiaomi device really works like a charm! Even the smoke detectors, that were a pain to pair, connects immediately. Very cool!!!
But when pairing my end devices to a router (same for all of the above) Xiaomi device hardly don't pair at all. I tried 3 hours and were able to get 3 devices up and running. Stopped then for the remaining 25 devices

## What did you expect to happen
Easy and successful pairing to routers as it works with the coordinator

## How to reproduce it (minimal and precise)
Pair a Xiaomi device to a router within a distance, where the end device does not communicate with the coordinator directly

## Debug Info
zigbee2mqtt version: 1.4.0
CC253X firmware version: 20190515 (coordinator) and diverse FW for routers

chris-jennings, 2019-05-17:
I think this is related to [1536](https://github.com/Koenkk/zigbee2mqtt/issues/1536)

Koenkk, 2019-05-18:
@ptvoinfo what is the status of pairing through CC253x routers?

ptvoinfo, 2019-05-18:
@Koenkk The router pass pairing requests to a coordinator. No problems. Just tested on my home network.

sascha777, 2019-05-18:
Thanks for digging into this!
The router pairing issue also occurs when using the Innr SP 120 and the Osram SmartPlug+. So I assume the issue is more due to the coordinator or zigbee2mqtt.

Koenkk, 2019-05-18:
@sascha777 pairing through what router?

sascha777, 2019-05-19:
Pairing Xiaomi end devices (smoke detectors are by far the hardest) to any router. I tried CC2531, the Innr Plug and an Osram plug as router. 
Pairing these routers themselves to the coordinator works like a charm. Also pairing the Xiaomi smoke detectors directly to the coordinator works great. 
I just can’t pair them through the routers to have them assigned as a direct child of the router instead of a direct child of the coordinator (which would exceed the max amount of devices of the coordinator).

Koenkk, 2019-05-19:
@sascha777 I don't think there is much to influence from the zigbee2mqtt side. It really depends on the device/routers itself.

You can try to pair the device directly to the coordinator, and then move it to the desired location. The device should then connect to a router (and won't be a direct child of the coordinator anymore).

orientalsniper, 2019-05-21:
I'm also having problem with routers, my setup is just a few xiaomi door/motion sensors with hue bulbs.

I have a CC2531 as coordinator and a CC2530 as router, both flashed with latest firmware, I can pair and receive signal if the sensors are close to the coordinator, but if I get farther away, the signal quality gets poorer.

I can't get the Hue bulbs or the CC2530 to route the signal back to the coordinator.

Koenkk, 2019-05-22:
@orientalsniper In my experience it can take some time (1 or 2 days) before the Xiaomi end devices switch to a new router.

roygbiv856, 2019-05-22:
@orientalsniper Where did you download the latest router firmware from?

dr-apple, 2019-05-26:
Ich have the same problem ;-(

dennyreiter, 2019-05-26:
I've been having a heck of a time with this also.  I've used a bunch of Osram Smart+ plugs as routers but even if I get the Xiaomi sensors to pair, they disappear within a day or so.  Sometimes I can get them back by pushing the button, but today I can't get them to pair for anything :( The two sensors that are connected directly to the coordinator seem to stay solid, though.

Koenkk, 2019-05-27:
@dennyreiter there are known issue with OSRAM devices kicking off devices.

orientalsniper, 2019-05-27:
> 
> 
> @orientalsniper Where did you download the latest router firmware from?

All the firmware can be found here:
https://github.com/Koenkk/Z-Stack-firmware

I gave up routing the Xiaomi Sensors that are farther away and just used the Xiaomi Gateway with Home Assistant.

alex-savin, 2019-07-09:
@Koenkk is there any way to fix the issue with Osram devices, or better to create a second network for them?

PS: I have the same issue, that Xiaomi devices are being kicked from routers/coordinator

Koenkk, 2019-07-09:
@alex-savin you could create a second network by setting up another zigbee2mqtt instance.

closed 2019-09-14 by stale[bot]