Koenkk/zigbee2mqtt #1542: zigbee2mqtt@1.3.1~start: Failed to exec start script

D3ltorohd, 2019-05-18:
Hello,
 this is my first install of the script. I installed it like it descripted in the wiki. For the install it seems to be all good. When i will start the script with npm start the following error comes out..

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.9.0
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.3.1~prestart: zigbee2mqtt@1.3.1
6 info lifecycle zigbee2mqtt@1.3.1~start: zigbee2mqtt@1.3.1
7 verbose lifecycle zigbee2mqtt@1.3.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.3.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.3.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.3.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.3.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.3.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.3.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.3.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.9.0-8-amd64
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.9.0
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.3.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.3.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]
`

So what i can do, or what misstake i have done that the script will not start ?

This is my config so far::

`# Required: Home Assistant integration (MQTT discovery)
homeassistant: false
# Required: allow new devices to join.
# WARNING: Disable this after all devices have been paired!
permit_join: true
# Required: MQTT settings
mqtt:
# Required: MQTT base topic for zigbee2mqtt MQTT messages
base_topic: zigbee2mqtt
# Required: MQTT server URL
server: 'mqtt://localhost:1883'
# Optional: MQTT client ID
client_id: 'Zigbee'
# Optional: Include device information to mqtt messages (default: false)
include_device_information: true
# Required: serial settings
serial:
# Required: location of CC2531 USB sniffer
port: /dev/ttyACM0
disbale_led: false
`

Greet's

JLFN, 2019-05-18:
"disbale_led: false"  How is the spelling?

D3ltorohd, 2019-05-18:
I think that can be closed. I have found my problem, something with the config went wrong. I copied it again from the docu and now it works. !!

closed 2019-05-18 by D3ltorohd