Koenkk/zigbee2mqtt #1545: Trying to support new device but shows up as unknown even when defined in database

Naturalien, 2019-05-19:
Hello,

currently I am trying to integrate my Zigbee compatible light from B-Leuchten. But I basically fail at the start because it shows up as unknown in my Zigbee2MQTT. If I look up my database.db to correctly link the unique ID with a modelID I found this rather well defined entry for the light, which sets the modelID to 020B0B. So I didn't change a thing because it looks all legit to me. Actually it shows up two times but zigbee-Sheperd is kinda weird and seems to do that all the time. I added it to the devices.js and still it shows up as unknown for some reason.

`{"id":6,"type":"Router","ieeeAddr":"0x00158d0001dcefc2","nwkAddr":39970,"manufId":4151,"manufName":"SEASTAR INTELLIGENCE","powerSource":"DC Source","modelId":"020B0B","epList":[1,2],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"swBuildId":"1000-0013","status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":49246,"epId":2,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20170206","_id":"8nJIAvZlgSo0t43a"}`

I added it to the devices.js as:

`    {
        zigbeeModel: ['20B0B'],
        model: '20B0B',
        vendor: 'SEASTAR INTELLIGENCE',
        description: 'ZigBee B-Leuchten LED Lamp',
        supports: 'on/off, brightness',
        fromZigbee: [],
        toZigbee: [],
    },`

ZigbeeDebug:

`npm start

> zigbee2mqtt@1.4.0 start /opt/zigbee2mqtt
> node index.js

                Zigbee2mqtt requires node version 10, you are running v8.16.0!

  zigbee2mqtt:info 2019-5-19 22:29:02 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-05-19.22-29-02'
  zigbee2mqtt:debug 2019-5-19 22:29:02 Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","coordBackupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 2019-5-19 22:29:02 Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 2019-5-19 22:29:03 Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 2019-5-19 22:29:03 Starting zigbee2mqtt version 1.4.0 (commit #6b75465)
  zigbee2mqtt:info 2019-5-19 22:29:03 Starting zigbee-shepherd
  zigbee2mqtt:info 2019-5-19 22:29:04 zigbee-shepherd started
  zigbee2mqtt:info 2019-5-19 22:29:04 Coordinator firmware version: '20181024'
  zigbee2mqtt:debug 2019-5-19 22:29:04 zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0019368e48","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20181024},"startTime":1558297744,"joinTimeLeft":0}
  zigbee2mqtt:info 2019-5-19 22:29:04 Currently 3 devices are joined:
  zigbee2mqtt:info 2019-5-19 22:29:04 0x00158d00023acdc9 (0x00158d00023acdc9): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
  zigbee2mqtt:info 2019-5-19 22:29:04 0x00158d0002252e18 (0x00158d0002252e18): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
  zigbee2mqtt:info 2019-5-19 22:29:04 0x00158d0001dcefc2 (0x00158d0001dcefc2): unkown - undefined unknown (Router)`

Koenkk, 2019-05-20:
Could you post your complete `database.db`?

Naturalien, 2019-05-20:
> Could you post your complete `database.db`?

Sure! Here it is:

`root@HIVE_HUB:/opt/zigbee2mqtt/data# cat database.db
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d0002252e18","nwkAddr":28331,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"4VsczVRQQPdJySR6"}
{"id":6,"type":"Router","ieeeAddr":"0x00158d0001dcefc2","nwkAddr":39970,"manufId":4151,"manufName":"SEASTAR INTELLIGENCE","powerSource":"DC Source","modelId":"020B0B","epList":[1,2],"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"swBuildId":"1000-0013","status":"offline","joinTime":null,"endpoints":{"1":{"profId":49246,"epId":1,"devId":4096,"inClusterList":[4096],"outClusterList":[4096],"clusters":{"lightLink":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":49246,"epId":2,"devId":528,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[0,25],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":1},"attrs":{}},"genGroups":{"dir":{"value":1},"attrs":{}},"genScenes":{"dir":{"value":1},"attrs":{}},"genOnOff":{"dir":{"value":1},"attrs":{}},"genLevelCtrl":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":1},"attrs":{}}}}},"dateCode":"20170206","_id":"8nJIAvZlgSo0t43a"}
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d00023acdc9","nwkAddr":4961,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sens","epList":[1,2,3],"status":"offline","joinTime":null,"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,18,25,65535],"outClusterList":[0,3,4,5,18,25,65535],"clusters":{"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":3},"attrs":{}},"manuSpecificCluster":{"dir":{"value":3},"attrs":{}}}},"2":{"profId":260,"epId":2,"devId":24322,"inClusterList":[3,18],"outClusterList":[3,4,5,18],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genMultistateInput":{"dir":{"value":3},"attrs":{}}}},"3":{"profId":260,"epId":3,"devId":24323,"inClusterList":[3,12],"outClusterList":[3,4,5,12],"clusters":{"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genAnalogInput":{"dir":{"value":3},"attrs":{}}}}},"_id":"YcjNpK1R10Svs1SV"}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0019368e48","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,11],"status":"online","joinTime":1558297744,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"ruLpyLZsNIcfx4CX"}
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0019368e48","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1558298616,"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}}},"_id":"ruLpyLZsNIcfx4CX"}
[... cut ...] "epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{"ssIasZone":{"dir":{"value":2},"cmdRsps":{"enrollReq":{"exec":"_exec_"},"statusChangeNotification":{"exec":"_exec_"}}}}}},"_id":"ruLpyLZsNIcfx4CX"}`

Koenkk, 2019-05-20:
The model ID you defined in `database.db` is **0**20B0B, while in `devices.js` it's 20B0B.

Naturalien, 2019-06-02:
Thanks. That resolved that isse. But I am still struggling to integrate it into my network as it doesn't respond to PINGs and it doesn't show the functions. And if I define it as a simple on/off/brightness lamp as I have seen in the configs it doesn't respond to my commands and doesn't turn on or off.

`5/20/2019, 8:45:00 PM - info: Starting zigbee2mqtt version 1.4.0 (commit #6b75465)
5/20/2019, 8:45:00 PM - info: Starting zigbee-shepherd
5/20/2019, 8:45:02 PM - info: zigbee-shepherd started
5/20/2019, 8:45:02 PM - info: Coordinator firmware version: '20181024'
5/20/2019, 8:45:02 PM - debug: zigbee-shepherd info: {"enabled":true,"net":{"state":"Coordinator","channel":"11","panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"ieeeAddr":"0x00124b0019368e48","nwkAddr":0},"firmware":{"transportrev":2,"product":0,"version":"2.6.3","revision":20181024},"startTime":1558377902,"joinTimeLeft":0}
5/20/2019, 8:45:02 PM - info: Currently 3 devices are joined:
5/20/2019, 8:45:02 PM - info: 0x00158d00023acdc9 (0x00158d00023acdc9): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
5/20/2019, 8:45:02 PM - info: 0x00158d0002252e18 (0x00158d0002252e18): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
5/20/2019, 8:45:02 PM - info: 0x00158d0001dcefc2 (0x00158d0001dcefc2): 020B0B - SEASTAR INTELLIGENCE ZigBee B-Leuchten LED Lamp (Router)
5/20/2019, 8:45:02 PM - warn: `permit_join` set to  `true` in configuration.yaml.
5/20/2019, 8:45:02 PM - warn: Allowing new devices to join.
5/20/2019, 8:45:02 PM - warn: Set `permit_join` to `false` once you joined all devices.
5/20/2019, 8:45:02 PM - info: Zigbee: allowing new devices to join.
5/20/2019, 8:45:02 PM - info: Connecting to MQTT server at mqtt://localhost
5/20/2019, 8:45:02 PM - info: zigbee-shepherd ready
5/20/2019, 8:45:02 PM - info: Connected to MQTT server
5/20/2019, 8:45:02 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
5/20/2019, 8:45:02 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d00023acdc9', payload '{"temperature":20.94,"linkquality":15,"humidity":62.89,"battery":37,"voltage":2975}'
5/20/2019, 8:45:03 PM - info: MQTT publish: topic 'zigbee2mqtt/0x00158d0002252e18', payload '{"temperature":7.93,"linkquality":36,"humidity":60.13,"battery":42,"voltage":2985}'
5/20/2019, 8:45:03 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023acdc9/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d00023acdc9","json_attributes_topic":"zigbee2mqtt/0x00158d00023acdc9","name":"0x00158d00023acdc9_temperature","unique_id":"0x00158d00023acdc9_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023acdc9","name":"0x00158d00023acdc9","sw_version":"Zigbee2mqtt 1.4.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
5/20/2019, 8:45:03 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023acdc9/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/0x00158d00023acdc9","json_attributes_topic":"zigbee2mqtt/0x00158d00023acdc9","name":"0x00158d00023acdc9_humidity","unique_id":"0x00158d00023acdc9_humidity_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023acdc9","name":"0x00158d00023acdc9","sw_version":"Zigbee2mqtt 1.4.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
5/20/2019, 8:45:03 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023acdc9/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x00158d00023acdc9","json_attributes_topic":"zigbee2mqtt/0x00158d00023acdc9","name":"0x00158d00023acdc9_battery","unique_id":"0x00158d00023acdc9_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023acdc9","name":"0x00158d00023acdc9","sw_version":"Zigbee2mqtt 1.4.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
5/20/2019, 8:45:03 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d00023acdc9/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d00023acdc9","json_attributes_topic":"zigbee2mqtt/0x00158d00023acdc9","name":"0x00158d00023acdc9_linkquality","unique_id":"0x00158d00023acdc9_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d00023acdc9","name":"0x00158d00023acdc9","sw_version":"Zigbee2mqtt 1.4.0","model":"MiJia temperature & humidity sensor  (WSDCGQ01LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
5/20/2019, 8:45:03 PM - info: MQTT publish: topic 'homeassistant/sensor/0x00158d0002252e18/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x00158d0002252e18","json_attributes_topic":"zigbee2mqtt/0x00158d0002252e18","name":"0x00158d0002252e18_temperature","unique_id":"0x00158d0002252e18_temperature_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x00158d0002252e18","name":"0x00158d0002252e18","sw_version":"Zigbee [... cut ...]
5/20/2019, 8:47:45 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d0001dcefc2/set' with data '{"state": "OFF"}'
5/20/2019, 8:47:46 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d0001dcefc2/set' with data '{"state": "ON"}'
5/20/2019, 8:47:59 PM - debug: Received MQTT message on 'zigbee2mqtt/0x00158d0001dcefc2/set' with data '{"state": "ON"}'
5/20/2019, 8:48:02 PM - debug: Ping 0x00158d0001dcefc2 (basic)
5/20/2019, 8:48:10 PM - error: Zigbee publish to device '0x00158d0001dcefc2', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms`

Koenkk, 2019-06-03:
Very strange that is gives an error on the on/off commands. Is the light in range of the coordinator?

Naturalien, 2019-06-05:
> Very strange that is gives an error on the on/off commands. Is the light in range of the coordinator?

It is actually right next to it for testing purposes. So the stick is at most 1 meter from the antenna of the lamp.
Sadly I don't have a lot of time at my hands at the time so further testing will have to wait a little.

closed 2019-08-11 by stale[bot]