Koenkk/zigbee2mqtt #1548: No serial output of CC2530 via USBTTL

phiten, 2019-05-20:
# Bug Report
## What happened

I'm currently working on the zigbee integration for tasmota. I flashed a CC2530 with the coordinator from here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin using a wemos d1 mini and the cclib (Yes I removed the second last line from the .hex). The flashing was completed and I wanted to check the traffic which the coordinator sends out via rx and tx. So I connected my cc2530 via a USBTTL adapter to my serial console and tried to connect via putty but there was absolutely no output in the console. 
I've used the following pinout for the USBTTL connection:
TX = P0_3 // P0_2
RX = P0_2 // P0_3
GND = GND
3v3 = VCC
P20 to GND

Some more informations from cc_info.py: https://pastebin.com/Pr3xJKc3

## What did you expect to happen

Any output to the console via USBTTL.

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: /
CC253X firmware version: CC2530_20190425

Koenkk, 2019-05-20:
You won't get output by default, you need to initialise the CC253x, closing as this is not a zigbee2mqtt issue. (but feel free to continue discussion).

closed 2019-05-20 by Koenkk

phiten, 2019-05-20:
Hi @Koenkk thanks for your quick response. May you be able to point me to the right place to look up this initialization procedure? Either some docs or even just the code you use in zigbee2mqtt? I will be able to figure out the rest I need then. If you have some arduino code for the initialization already I would appreciate if you would send it to me.

Koenkk, 2019-05-20:
The init procedure can be found here: https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/controller.js#L258.

phiten, 2019-05-20:
Thank you I will check it out.

JoanMCD, 2020-01-01:
Hi, I had also some problems to get the CC2530 --> ESP8266 --> ZigBee2MQTT to get communicate. The missed info I found late - the 

advanced:
  rtscts: false

and the possibility of
serial:
  port: 'tcp://IP:PORT'

in the configuration.yaml

On this way I wish and it would be interessing to thest the right flashed FW and the corect RX TX wireing by terminal.    

So, here my log of the first bytes through the ESP with a Xiaomi Temp/hum Sensor.
Xiaomi  ___/---- > CC2530 --> ESP8266 --> ZigBee2MQTT

--> 0x0<\r><\n>
<\n>
--> 0x0 0x0 0x0 0x0 0x0<\n>
<-- 0xFE 0xE 0x61 0x2 0x2 0x0 0x2 0x6 0x3 0x90 0x15 0x34 0x1 0x2 0x0 0x0 0x0 0x0 0xDA<\n>
--> 0x0 0x21 0x2 0x23 0x0 0x0 0x0 0x0<\n>
<-- 0xFE 0x3 0x61 0x8 0x0 0x1 0x55 0x3E<\n>
--> 0x3 0x21 0x8 0x0 0xF 0x0 0x25 0x0<\n>
<-- 0xFE 0x3 0x61 0x8 0x0 0x1 0x55 0x3E<\n>
--> 0x3 0x21 0x8 0x0 0xF 0x0 0x25 0x0<\n>
<-- 0xFE 0x6 0x61 0x8 0x0 0x4 0x0 0x8 0x0 0x0 0x63<\n>
--> 0x3 0x21 0x8 0x84 0x0 0x0 0xAE 0x0<\n>
<-- 0xFE 0x3 0x61 0x8 0x0 0x1 0x0 0x6B<\n>
--> 0x3 0x21 0x8 0x63 0x0 0x0<\n>
<-- 0xFE 0x13 0x66 0x4 0x0 0x62 0x10 0x1 0x3 0x5 0x7 0x9 0xB 0xD 0xF 0x0 0x2 0x4 0x6 0x8 0xA 0xC 0xD 0x0<\n>
--> 0x1 0x26 0x4 0x62 0x41 0x0 0x49 0x0<\n>
<-- 0xFE 0x4 0x61 0x8 0x0 0x2 0x62 0x1A 0x17<\n>
--> 0x3 0x21 0x8 0x83 0x0 0x0 0xA9 0x0<\n>
<-- 0xFE 0xA 0x61 0x8 0x0 0x8 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0x6B<\n>
--> 0x3 0x21 0x8 0x2D 0x0<\n>
<-- 0xFE 0xE 0x67 0x0 0x0 0x7C 0x12 0xD1 0x1B 0x0 0x4B 0x12 0x0 0x0 0x0 0x7 0x0 0x0 0x93<\n>
--> 0x0 0x27 0x0 0x27 0x0 0x0 0x7<\r><\n>

JoanMCD, 2020-01-01:
Hi, I had also some problems to get the CC2530 --> ESP8266 --> ZigBee2MQTT to get communicate. The missed info I found late - the 

```
advanced:
  rtscts: false
```

and the possibility of
```
serial:
  port: 'tcp://IP:PORT'
```

in the configuration.yaml . . .great :-) !!

On this way I wish and it would be interessing to thest the right flashed FW and the corect RX TX wireing by terminal.    

So, here my log of the first bytes through the ESP with a Xiaomi Temp/hum Sensor.
Xiaomi  ___/---- > CC2530 --> ESP8266 --> ZigBee2MQTT

--> from ZigBee2MQTT
<-- to ZigBee2MQTT

```
--> 0x00<\r><\n>
<\n>
--> 0x00 0x00 0x00 0x00 0x00<\n>
<-- 0xFE 0x0E 0x61 0x02 0x02 0x00 0x02 0x06 0x03 0x90 0x15 0x34 0x01 0x02 0x00 0x00 0x00 0x00 0xDA<\n>
--> 0x00 0x21 0x02 0x23 0x00 0x00 0x00 0x00<\n>
<-- 0xFE 0x03 0x61 0x08 0x00 0x01 0x55 0x3E<\n>
--> 0x03 0x21 0x08 0x00 0x0F 0x00 0x25 0x00<\n>
<-- 0xFE 0x03 0x61 0x08 0x00 0x01 0x55 0x3E<\n>
--> 0x03 0x21 0x08 0x00 0x0F 0x00 0x25 0x00<\n>
<-- 0xFE 0x06 0x61 0x08 0x00 0x04 0x00 0x08 0x00 0x00 0x63<\n>
--> 0x03 0x21 0x08 0x84 0x00 0x00 0xAE 0x00<\n>
<-- 0xFE 0x03 0x61 0x08 0x00 0x01 0x00 0x6B<\n>
--> 0x03 0x21 0x08 0x63 0x00 0x00<\n>
<-- 0xFE 0x13 0x66 0x04 0x00 0x62 0x10 0x01 0x03 0x05 0x07 0x09 0x0B 0x0D 0x0F 0x00 0x02 0x04 0x06 0x08 0x0A 0x0C 0x0D 0x00<\n>
--> 0x01 0x26 0x04 0x62 0x41 0x00 0x49 0x00<\n>
<-- 0xFE 0x04 0x61 0x08 0x00 0x02 0x62 0x1A 0x17<\n>
--> 0x03 0x21 0x08 0x83 0x00 0x00 0xA9 0x00<\n>
<-- 0xFE 0x0A 0x61 0x08 0x00 0x08 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0x6B<\n>
--> 0x03 0x21 0x08 0x2D 0x00<\n>
<-- 0xFE 0x0E 0x67 0x00 0x00 0x7C 0x12 0xD1 0x1B 0x00 0x4B 0x12 0x00 0x00 0x00 0x07 0x09 0x00 0x9A<\n>
--> 0x00 0x27 0x00 0x27 0x00 0x00 0x07 0x00 0x00<\n>
<-- 0xFE 0x01 0x65 0x05 0x00 0x61 0xFE 0x10 0x45 0x85 0x00 0x00 0x00 0x00 0x00 0x0A 0x0C 0x6E 0x0B 0x08 0x06 0x05 0x04 0x03 0x02 0x01 0xBC<\n>
--> 0x04 0x25 0x05 0x00 0x00<\n>
<-- 0xFE 0x18 0x65 0x50 0x00 0x00 0x09 0x62 0x1A 0x00 0x00 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0xDD 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x0B 0x57<\n>
--> 0x00 0x25 0x50 0x75 0x00 0x00 0x00 0x24 0x00 0x00<\n>
<-- 0xFE 0x01 0x65 0x36 0x00 0x52<\n>
<-- 0xFE 0x01 0x45 0xCB 0xFE 0x71 0xFE 0x03 0x45 0xB6 0x00 0x00 0x00 0xF0<\r><\n>

```

zigbee2mqtt:info  2020-00-00 13:06:46: MQTT publish: topic 'zigbee2mqtt/0x00158d0001576cc9', payload '{"battery":86,"voltage":2975,"temperature":24.35,"humidity":46.66,"linkquality":173}'

results in : 

<-- 0xFE 0x1C 0x44 0x81 0x00 0x00 0x02 0x04 0x41 0xF0 0x01 0x01 0x00 0xAD 0x00 0xC9 0xAB 0x16 0x00 0x00 0x08 0x18 0x9A 0x0A 0x00 0x00 0x29 0x83 0x09 0x41 0xF0 0x1D 0x38 0xFE 0x1C 0x44 0x81 0x00 0x00 0x05 0x04 0x41 0xF0 0x01 0x01 0x00 0xAD 0x00 0xD5 0xAB 0x16 0x00 0x00 0x08 0x18 0x9B 0x0A 0x00 0x00 0x21 0x3A 0x12 0x41 0xF0 0x1D 0x88<\r><\n>

BR