Koenkk/zigbee2mqtt #1549: CC2530 router issues - not joining

goldriver2, 2019-05-20:
I have build a CC2530 router as instruction givin at https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html and used the CC2531_20190425\CC2531ZNP-Prod.hex file. Once programmed, I made sure that the _verify against hex file_ was OK

I plugged the device to power outlet and waited, waited, waited for about 60 minutes and still, the zigbe2mqtt log does not show any signs of the router.

To make sure I didn't break anything while assembling the device, I replugged the debugger on the cc2530 and did a check againt hex file to make sur the cc2530 was still working with the correct hex file.

I than once again plugged the router to the power outlet and let it sit for another hour and still no appearance of the router in the log file.

Any clue on what could be wrong ?

using zigbee2mqtt version 1.1.1 (commit #bdfabeb)

mabi, 2019-05-20:
try this:
СС2530 can be re-paired if you power on/power off it three times (power on, wait 2 seconds, power off, repeat this cycle three times).
(source http://ptvo.info)

goldriver2, 2019-05-20:
ok tried the power on/off three times about an hour ago and still, I don't see the router appearing in the log

goldriver2, 2019-05-20:
Ok tried the on/off 3 times about an hour ago and still, the router does not show in the log

goldriver2, 2019-05-20:
Ok got it working, looks like the ling on the documentation page of zigbee2mqtt is not pointing to the right .hex file, looking on  http://ptvo.info, I found out that the hex to use is : router-cc2530-std.hex from http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/

not CC2531ZNP-Prod.hex

closed 2019-05-20 by goldriver2

Koenkk, 2019-05-20:
@goldriver2 which documentation page contained the wrong link?

goldriver2, 2019-05-20:
https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

when you fallow the link in this paragraph:

_Flashing
Now the CC2530 can be flashed using the same instructions as **Flashing the CC2531**. The router firmwares can be found here._

Koenkk, 2019-05-21:
@goldriver2 yes, but as mentioned you need to follow *The router firmwares can be found here* which links to https://github.com/Koenkk/Z-Stack-firmware/tree/master/router

goldriver2, 2019-05-21:
Sorry I missed the last link for the firmware location