Koenkk/zigbee2mqtt #1551: Trying to support a new device but I get an error.  HASSIO

proton999, 2019-05-20:
First of all I hope I can ask overhere for help.
I'm trying to add a nice vintage, zigbee certified, led light from the manufacturer GMY.
www.gmysmart.com but i'm getting an error. I'm using a 2531CC stick with the latest firmware 20190425 on a HASSIO system with software level 92.1. The zigbee2mqtt addon is version 1.3.1.
The modelID is 'CCT box'

Below the logs, configs and error log. Could someone check what i'm doing wrong.

**HASSIO z2m log:**
zigbee2mqtt:warn 5/20/2019, 7:13:23 PM Message without device!
  zigbee2mqtt:warn 5/20/2019, 7:13:25 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 5/20/2019, 7:13:25 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 5/20/2019, 7:13:26 PM Device incoming...
  zigbee2mqtt:info 5/20/2019, 7:13:26 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 5/20/2019, 7:13:26 PM Device with modelID 'CCT box' is not supported.
  zigbee2mqtt:warn 5/20/2019, 7:13:26 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn 5/20/2019, 7:13:26 PM Device with modelID 'CCT box' is not supported.

**database:**
{"id":5,"type":"Router","ieeeAddr":"0x00158d00013ea369","nwkAddr":31557,"manufId":4151,"epList":[1],"status":"online","joinTime":1558372007,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[0,3,4,5,6,8,25,768,4096,4111],"clusters":{}}},"_id":"dfK1W26bTZuJ8ErI"}
{"id":5,"type":"Router","ieeeAddr":"0x00158d00013ea369","nwkAddr":31557,"manufId":4151,"epList":[1],"status":"online","joinTime":1558372007,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[0,3,4,5,6,8,25,768,4096,4111],"clusters":{"4111":{"dir":{"value":3},"attrs":{}},"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":3},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{}},"genOnOff":{"dir":{"value":3},"attrs":{}},"genLevelCtrl":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":3},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"dfK1W26bTZuJ8ErI"}
{"$$deleted":true,"_id":"dfK1W26bTZuJ8ErI"}
{"id":5,"type":"Router","ieeeAddr":"0x00158d00013ea369","nwkAddr":54013,"manufId":4151,"epList":[1],"status":"online","joinTime":1558372404,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[0,3,4,5,6,8,25,768,4096,4111],"clusters":{}}},"_id":"ezauSfP3ZUs0QRu9"}
{"id":5,"type":"Router","ieeeAddr":"0x00158d00013ea369","nwkAddr":54013,"manufId":4151,"manufName":"ZB/GMY-CCT","powerSource":"Mains (single phase)","modelId":"CCT box","epList":[1],"status":"online","joinTime":1558372404,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[0,3,4,5,6,8,25,768,4096,4111],"clusters":{}}},"_id":"ezauSfP3ZUs0QRu9"}
{"id":5,"type":"Router","ieeeAddr":"0x00158d00013ea369","nwkAddr":54013,"manufId":4151,"manufName":"ZB/GMY-CCT","powerSource":"Mains (single phase)","modelId":"CCT box","epList":[1],"status":"online","joinTime":1558372404,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[0,3,4,5,6,8,25,768,4096,4111],"clusters":{"4111":{"dir":{"value":3},"attrs":{}},"genBasic":{"dir":{"value":3},"attrs":{}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":3},"attrs":{}},"genScenes":{"dir":{"value":3},"attrs":{}},"genOnOff":{"dir":{"value":3},"attrs":{}},"genLevelCtrl":{"dir":{"value":3},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"lightingColorCtrl":{"dir":{"value":3},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"ezauSfP3ZUs0QRu9"}

**configuration.yaml**
devices:
  '0x000b57fffee82d28':
    friendly_name: Ikea-Lamp1
    retain: false
  '0x00158d00021646c8':
    friendly_name: DoorswitchXD1
    retain: false
  '0x00158d0002b7586d':
    friendly_name: PIR1
    retain: false
  '0x00158d00013ea369':
    friendly_name: '0x00158d00013ea369'
    retain: false

**hassio addon config:**
{
  "data_path": "/share/zigbee2mqtt",
  "homeassistant": true,
  "permit_join": false,
  "mqtt_base_topic": "zigbee2mqtt",
  "mqtt_server": "mqtt://homeassistant:1883",
  "serial_port": "/dev/ttyACM0",
  "devices": [],
  "network_key": [],
  "mqtt_user": "XXXXX",
  "mqtt_pass": "XXXXX",
  "zigbee_shepherd_devices": true
}

**added to devices.js**
// GMY
    {
        zigbeeModel: ['CCT box'],
        model: 'CCT box',
        vendor: 'GMY Smart Bulb',
        description: 'GMY Smart bulb, 470lm, vintage dimmable, 2700-6500k',
        extend: generic.light_onoff_brightness_colortemp,
    },

**errorlog addon:**
'{"version":"1.3.1","commit":"unknown","coordinator_firmware":20190425,"log_level":"info","permit_join":false}'
/zigbee2mqtt-1.3.1/lib/extension/deviceReceive.js:109
                if (c.cid === cid) {
                      ^
TypeError: Cannot read property 'cid' of undefined
    at mappedDevice.fromZigbee.filter (/zigbee2mqtt-1.3.1/lib/extension/deviceReceive.js:109:23)
    at Array.filter (<anonymous>)
    at DeviceReceive.onZigbeeMessage (/zigbee2mqtt-1.3.1/lib/extension/deviceReceive.js:105:52)
    at extensions.filter.forEach (/zigbee2mqtt-1.3.1/lib/controller.js:145:31)
    at Array.forEach (<anonymous>)
    at Controller.onZigbeeMessage (/zigbee2mqtt-1.3.1/lib/controller.js:145:14)
    at Zigbee.onMessage (/zigbee2mqtt-1.3.1/lib/zigbee.js:224:18)
    at ZShepherd.emit (events.js:182:13)
    at ZShepherd.<anonymous> (/zigbee2mqtt-1.3.1/node_modules/zigbee-shepherd/lib/shepherd.js:170:14)
    at ZShepherd.emit (events.js:182:13)
npm ERR! code
 ELIFECYCLE
npm 
ERR! errno 1
npm ERR! zigbee2mqtt@1.3.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-05-20T18_15_19_712Z-debug.log
2019-05-20T20:15:22: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2019-05-20T20:15:22: PM2 log: App [npm:0] starting in -fork mode-
2019-05-20T20:15:22: PM2 log: App [npm:0] online
> zigbee2mqtt@1.3.1 start /zigbee2mqtt-1.3.1
> node index.js
  zigbee2mqtt:info 5/20/2019, 8:15:26 PM Logging to directory: '/share/zigbee2mqtt/log/2019-05-20.20-15-26'
  zigbee2mqtt:info 5/20/2019, 8:15:27 PM Starting zigbee2mqtt version 1.3.1 (commit #unknown)
  zigbee2mqtt:info 5/20/2019, 8:15:27 PM Starting zigbee-shepherd
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM zigbee-shepherd started
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM Coordinator firmware version: '20190425'
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM Currently 4 devices are joined:
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM DoorswitchXD1 (0x00158d00021646c8): MCCGQ01LM - Xiaomi MiJia door & window contact sensor (EndDevice)
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM PIR1 (0x00158d0002b7586d): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM Ikea-Lamp1 (0x000b57fffee82d28): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
  zigbee2mqtt:info 5/20/2019, 8:15:28 PM 0x00158d00013ea369 (0x00158d00013ea369): CCT box - GMY Smart Bulb GMY Smart b

Koenkk, 2019-05-20:
Can you post the complete devices.js?

proton999, 2019-05-20:
Ofcourse, here it is.

'use strict';

const debug = require('debug')('zigbee-shepherd-converters:devices');
const fz = require('./converters/fromZigbee');
const tz = require('./converters/toZigbee');

const repInterval = {
    MAX: 62000,
    HOUR: 3600,
    MINUTE: 60,
};

const coordinatorGroup = 99;

const defaultIgnoreConverters = [
    fz.ignore_genGroups_devChange, fz.ignore_genIdentify_change, fz.ignore_genScenes_change,
    fz.ignore_diagnostic_change, fz.ignore_lightLink_change, fz.ignore_basic_change,
];

const generic = {
    light_onoff_brightness: {
        supports: 'on/off, brightness',
        fromZigbee: [
            fz.ignore_light_brightness_change, fz.state_change, fz.state, fz.brightness_report,
        ].concat(defaultIgnoreConverters),
        toZigbee: [tz.light_onoff_brightness, tz.ignore_transition, tz.light_alert],
    },
    light_onoff_brightness_colortemp: {
        supports: 'on/off, brightness, color temperature',
        fromZigbee: [
            fz.ignore_light_brightness_change, fz.color_colortemp, fz.state_change, fz.state,
            fz.brightness_report, fz.color_colortemp_report,
        ].concat(defaultIgnoreConverters),
        toZigbee: [tz.light_onoff_brightness, tz.light_colortemp, tz.ignore_transition, tz.light_alert],
    },
    light_onoff_brightness_colorxy: {
        supports: 'on/off, brightness, color xy',
        fromZigbee: [
            fz.ignore_light_brightness_change, fz.color_colortemp, fz.state_change, fz.state,
            fz.brightness_report, fz.color_colortemp_report,
        ].concat(defaultIgnoreConverters),
        toZigbee: [tz.light_onoff_brightness, tz.light_color, tz.ignore_transition, tz.light_alert],
    },
    light_onoff_brightness_colortemp_colorxy: {
        supports: 'on/off, brightness, color temperature, color xy',
        fromZigbee: [
            fz.ignore_light_brightness_change, fz.color_colortemp, fz.state_change, fz.state,
            fz.brightness_report, fz.color_colortemp_report,
        ].concat(defaultIgnoreConverters),
        toZigbee: [
            tz.light_onoff_brightness, tz.light_color_colortemp, tz.ignore_transition,
            tz.light_alert,
        ],
    },
};

const gledopto = {
    light_onoff_brightness: {
        supports: generic.light_onoff_brightness.supports,
        fromZigbee: generic.light_onoff_brightness.fromZigbee,
        toZigbee: [tz.gledopto_light_onoff_brightness, tz.ignore_transition, tz.light_alert],
    },
    light_onoff_brightness_colortemp: {
        supports: generic.light_onoff_brightness_colortemp.supports,
        fromZigbee: generic.light_onoff_brightness_colortemp.fromZigbee,
        toZigbee: [
            tz.gledopto_light_onoff_brightness, tz.gledopto_light_colortemp, tz.ignore_transition,
            tz.light_alert,
[... 3896 lines cut ...]
        model: 'CCT box',
        vendor: 'GMY Smart Bulb',
        description: 'GMY Smart bulb, 470lm, vintage dimmable, 2700-6500k',
        extend: generic.light_onoff_brightness_colortemp,
    },
    
    
    // Gira
    {
        zigbeeModel: [' Remote'],
        model: '2430-100',
        vendor: 'Gira',
        description: 'ZigBee Light Link wall transmitter',
        supports: 'action',
        fromZigbee: [
            fz.GIRA2430_scene_click, fz.GIRA2430_on_click, fz.GIRA2430_off_click, fz.GIRA2430_down_hold,
            fz.GIRA2430_up_hold, fz.GIRA2430_stop,
        ],
        toZigbee: [],
    },
];

module.exports = devices.map((device) =>
    device.extend ? Object.assign({}, device.extend, device) : device
);

Koenkk, 2019-05-21:
At the location of your `devices.js`, do you also have `converters/fromZigbee.js` and `converters/toZigbee.js`? https://github.com/Koenkk/zigbee-shepherd-converters/tree/master/converters

proton999, 2019-05-22:
I copied the suggested files too the path where the device.js is and now it pops up in the Z2M addon log and it reacts to commands which I publish with MQTT Explorer. But it is not popping up in Home Assistant as an entity.   Tips are welcome.

Koenkk, 2019-05-22:
that will be done once it's integrated, can you provide a link (url) for this device so I can do this?

proton999, 2019-05-22:
Koenkk, thanks for all your support, highly appreciated.  Below a link where I bought the lamps.

https://www.amazon.de/dp/B07KG5KF5R/ref=pe_3044161_185740101_TE_item

Koenkk, 2019-05-22:
Great, will be supported in the edge version in a few hours!

closed 2019-05-22 by Koenkk

arnonel, 2019-10-11:
Any ideas how to put this device into pairing mode? 

p.s.
I have HS2SK-E-UK

Nosstrom, 2020-10-05:
> 
> 
> Great, will be supported in the edge version in a few hours!

Hello 👍    is it possible to support the GL-C-006? from gledopto its dimmer , i have tried everything and i cant do it

{
    zigbeeModel: ['GL-C-006', 'GL-C-009'],
    model: 'GL-C-006/GL-C-009',
    vendor: 'Gledopto',
    description: 'Zigbee LED controller WW/CW Dimmer',
    extend: gledopto.light_onoff_brightness_colortemp,
    ep: (device) => {
        if (device.epList.toString() === '11,12,13') {
            return {'': 12};
        } else if (device.epList.toString() === '10,11,13' || device.epList.toString() === '11,13') {
            return {'': 11};
        } else {
            return {};
        }
    },
},

Koenkk, 2020-10-06:
@Nosstrom please share the `data/database.db` entry of your device.

Nosstrom, 2020-10-06:
> 
> 
> @Nosstrom please share the `data/database.db` entry of your device.

{"id":12,"type":"Router","ieeeAddr":"0x00124b001e13d5","nwkAddr":1744,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-006S","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-006S","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0"}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":544,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1601990366735}

{"id":13,"type":"Router","ieeeAddr":"0x00124b001ed01c","nwkAddr":46826,"manufId":0,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-C-006S","epList":[11,13],"endpoints":{"11":{"profId":49246,"epId":11,"devId":544,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"GL-C-006S","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":1,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0"}}},"binds":[],"meta":{}},"13":{"profId":49246,"epId":13,"devId":544,"inClusterList":[4096],"outClusterList":[4096],"clusters":{},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20131206","swBuildId":"2.0.0","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1601990366741}

this is the devices  which are paired but not supported.

Koenkk, 2020-10-06:
Added, Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Nosstrom, 2020-10-06:
> 
> 
> Added, Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

thank you so much for the ultra fast reply.I apreciate it! when will be this supported to the non edge version?

Koenkk, 2020-10-06:
On the next release (1 november)

Nosstrom, 2020-10-06:
Perfect!!! No problem, the only thing that matters is planning to be supported even is the next month.

> 
> 
> On the next release (1 november)

Perfect!!! No problem, the only thing that matters is that will be supported even next month.Thank you for this!!!
So i will use the edge untill then. Is there anything else i can do or just enable edge and wait?

Koenkk, 2020-10-06:
Just switch to edge

Nosstrom, 2020-10-06:
> Just switch to edge

i made it work ..works but doesnt get state., if i switch it from controller....i used installation method with "zigbee_shepherd_devices: true" into conf.yaml and copy and paste the devices.js file of the latest release into the share/zigbee2mqtt folder after pairing and and 2 restart worked.

Koenkk, 2020-10-07:
@Nosstrom use the report feature: https://www.zigbee2mqtt.io/information/report.html

sibero80, 2020-10-18:
Hello all, I'm about to pull the trigger on the "same" model, but is listed as 53170162 instead of 53170161.
In your eperience, do a lot of things change on iterations of the same model? could this break support?
If anyone is interested, they are heavily discounted right now. https://www.amazon.com/Commercial-Electric-Matte-Recessed-Downlight/dp/B01LE3757M