Koenkk/zigbee2mqtt #1553: SecuriFi Peanut Smart Plug (PP-WHT-US) unkown - undefined unknown (Router)

ChrisSantiago82, 2019-05-21:
Hello,

I'm new to the zigbee2mqtt so please be patient with me.

Since a week I'm trying to pair a SecuriFi Peanut Smart Plug.
I have done everything how it says in the documentation and more than one time.
Including adding the ModelId (http://www.zigbee2mqtt.io/getting_started/pairing_devices.html)

But nothing works. I can start the pairing. I'm getting the Database entries with the missing ModelId . I'm adding the missing ModelId and restarting the zigbee2mqtt.
And every time I'm getting the unkown - undefined unknown (Router) in the log.

Can anyone help me or push me in the right direction?

Log:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3204475/log.txt)

Database.db: (changed it to txt to be able to upload here)
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/3204480/database.txt)

Koenkk, 2019-05-22:
Can you try with `modelId` instead of `ModelId` (lower `M`)

ChrisSantiago82, 2019-05-22:
Yes, you are correct. That is the solution. And I'm feeling stupid to have worked on it for days.
How can we adapt the documentation so that it doesn't happen the same way to the next person?
Because here it's not a lower M:
https://www.zigbee2mqtt.io/getting_started/pairing_devices.html

Koenkk, 2019-05-22:
thanks, fixed!

closed 2019-05-22 by Koenkk

ChrisSantiago82, 2019-05-22:
I thank you for everything